intel/compute-runtime #529: Support for building just L0 Sysman support without IGC dependency

eero-t, 2022-03-29:
Because IGC is preventing building of latest compute-runtime versions: https://github.com/intel/intel-graphics-compiler/issues/224

Alternatively, compute-runtime could detect at build time whether installed IGC version supports EU fusion, and use it only when IGC supports it.  That way compute-runtime could still be built with older IGC versions that do not break compute-runtime build.

jandres742, 2022-03-29:
@eero-t thanks for the suggestion. At this moment, we are working at L0 spec to define how SYSMAN and Core APIs should better interact, following the comments on https://github.com/oneapi-src/level-zero/issues/36. WE could take this suggestion under consideration and see how it fits in the new scheme being discussed.

eero-t, 2022-05-12:
FYI: IGC build issue got resolved, but I think there's still value in separating compute and Sysman functionality as I do not think there's much overlap between them.  Being able to build Sysman functionality separately without needing also to build all the extra compute dependencies (IGC and its deps) would be great.

tkatila, 2024-02-21:
Are there any updates to this?

eero-t, 2025-04-11:
> FYI: IGC build issue got resolved, but I think there's still value in separating compute and Sysman functionality as I do not think there's much overlap between them. Being able to build Sysman [...]

@JablonskiMateusz Having Sysman as a separate binary would be also great for containers that need just to identify what devices are present, and get metrics from them (for Kubernetes devices drivers, metric monitors etc).  IGC + LLVM deps more than double the container sizes.