intel/compute-runtime #531: Build error: >type ‘struct drm_i915_gem_create_ext’ violates the C++ One Definition Rule

ran-dall, 2022-04-13:
I observed this build problem a few versions ago, and I was just able to forcibly turn off LTO to make it build; however, upon further inspection of merging the fix into the ebuild, Gentoo upstream suggested the root cause of the problem to be more serious...

```
>../compute-runtime-22.14.22890/third_party/uapi/drm/i915_drm.h:2951: error: >type ‘struct drm_i915_gem_create_ext’ violates the C++ One Definition Rule >[-Werror=odr]
> 2951 | struct drm_i915_gem_create_ext {
```

Ref 👉 https://bugs.gentoo.org/838241

JablonskiMateusz, 2022-04-26:
Hi @ran-dall. Thanks for reporting the issue. I've added this to our queue.

ran-dall, 2022-05-03:
@ArchangeGabriel Thanks for confirming.

closed 2022-05-03 by ran-dall