intel/compute-runtime #533: libigdgmm12 (= 22.1.2)

sagitt, 2022-04-23:
can't download this lib in last release:

wget https://github.com/intel/compute-runtime/releases/download/22.16.22992/intel-gmmlib_22.1.2_amd64.deb
--2022-04-23 14:07:04--  https://github.com/intel/compute-runtime/releases/download/22.16.22992/intel-gmmlib_22.1.2_amd64.deb
Risoluzione di github.com (github.com)... 140.82.121.3
Connessione a github.com (github.com)|140.82.121.3|:443... connesso.
Richiesta HTTP inviata, in attesa di risposta... 404 Not Found
2022-04-23 14:07:04 ERRORE 404: Not Found.

ArturHarasimiuk, 2022-04-25:
@sagitt - thanks for feedback. I've corrected urls in release notes

eero-t, 2022-05-12:
@sagitt Can you verify and close this?

sagitt, 2022-05-12:
> @sagitt Can you verify and close this?

Works.
Thanks

closed 2022-05-12 by sagitt