intel/compute-runtime #536: %% string breaks printf

linehill, 2022-04-27:
Link to OpenCL reproducer: https://github.com/pocl/pocl/blob/master/tests/spirv/printf-kernels.cl#L55.

Instead of printing the expected output: 
```
hello
%
hello % world
%s
```
It print something like this instead (output may vary):
```
hello
%%\n,%\n,%\n,%\n,%\n,%\n,%\n,%\n,%\n,%\n,%\n
```
Test environment:
OS: Ubuntu 20.04.4 LTS
Intel Compute Runtime version: 22.16.22992
Device: Intel(R) HD Graphics 530 [0x1912]

JablonskiMateusz, 2022-10-24:
Hi @linehill 
It should be fixed within https://github.com/intel/compute-runtime/commit/20bb97ece3feb40270bd787c752a200ba9092db8 
Could you confirm that it works fine when using this change or recent release?

linehill, 2022-10-25:
Confirmed to be fixed in release 22.42.24548.

closed 2022-10-25 by linehill