intel/compute-runtime #540: HDR Tone Mapping in Plex Docker Broken

nickp85, 2022-05-16:
After a recent Intel driver update, the hardware transcoding capability for HDR Tone Mapping in the Plex docker container no longer works. Transcoding 4K HDR content to 1080p with HDR tone mapping enabled forces software transcoding. Disabling this setting leads to hardware transcode again.

We have eliminated Plex and the docker container as the cause and can confirm that this started after an Intel driver update inside the container.

Can you please look into this and resolve?

CPU I have is 9900K.

Thank you

jandres742, 2022-06-10:
hi @nickp85 . do you have data on which driver was working previously, and with which one is now not working?

pwilma, 2022-06-13:
Isn't it a duplicate of https://github.com/intel/compute-runtime/issues/488?

AdamCetnerowski, 2022-11-24:
Closing as duplicate

closed 2022-11-24 by AdamCetnerowski