intel/compute-runtime #543: Fail sum on last release

sagitt, 2022-05-28:
sha256sum -c ww21.sum
intel-igc-core_1.0.11279_amd64.deb: OK
**intel-igc-opencl_1.0.11279_amd64.deb: NON RIUSCITO**
intel-level-zero-gpu-dbgsym_1.3.23269_amd64.ddeb: OK
intel-level-zero-gpu_1.3.23269_amd64.deb: OK
intel-opencl-icd-dbgsym_22.21.23269_amd64.ddeb: OK
intel-opencl-icd_22.21.23269_amd64.deb: OK
libigdgmm12_22.1.3_amd64.deb: OK
sha256sum: ATTENZIONE: 1 codice di controllo calcolato NON corrisponde

[22.21.23269](https://github.com/intel/compute-runtime/releases/tag/22.21.23269)

craigkc, 2022-05-29:
Same issue here.  From an English OS, including proof .deb is corrupt:
`
$ sha256sum -c ww21.sum
intel-igc-core_1.0.11279_amd64.deb: OK
intel-igc-opencl_1.0.11279_amd64.deb: FAILED
intel-level-zero-gpu-dbgsym_1.3.23269_amd64.ddeb: OK
intel-level-zero-gpu_1.3.23269_amd64.deb: OK
intel-opencl-icd-dbgsym_22.21.23269_amd64.ddeb: OK
intel-opencl-icd_22.21.23269_amd64.deb: OK
libigdgmm12_22.1.3_amd64.deb: OK
sha256sum: WARNING: 1 computed checksum did NOT match
`

It's not just a sum wrong in .sum file, the .deb file is legitimately corrupt as shown with this output:
`
$ sudo dpkg -i intel-igc-opencl_1.0.11279_amd64.deb
Unpacking intel-igc-opencl (1.0.11279) over (1.0.10409) ...
dpkg-deb (subprocess): decompressing archive member: internal gzip read error: '<fd:4>: invalid distance code'
dpkg-deb: error: <decompress> subprocess returned error exit status 2
dpkg: error processing archive intel-igc-opencl_1.0.11279_amd64.deb (--install):
 cannot copy extracted data for './usr/local/lib/libopencl-clang.so.11' to '/usr/local/lib/libopencl-clang.so.11.dpkg-new': unexpected end of file or stream
`

This is the (broken) file that is being distributed:
`
$ sha256sum intel-igc-opencl_1.0.11279_amd64.deb
67b9608115995b4b91725d145e7f3f79dbdfb4f175d6a561da052a8ce1206c9c  intel-igc-opencl_1.0.11279_amd64.deb
`
`
$ ls -al
-rw-rw-r-- 1 redacted redacted  56027994 May 24 11:41 intel-igc-opencl_1.0.11279_amd64.deb
`

powderluv, 2022-05-30:
https://github.com/intel/intel-graphics-compiler/issues/244

powderluv, 2022-05-30:
New SHA is 
```
 sha256sum intel-igc-opencl_1.0.11279_amd64.deb
0d6fddea8dd00c710973c31a92a6a3d3d7adc358aaceacfa6c73d2df1a3c1708  intel-igc-opencl_1.0.11279_amd64.deb
```

ArturHarasimiuk, 2022-05-30:
ww21.sum was updated with corrected values.
Sorry for inconvenience.

closed 2022-05-30 by ArturHarasimiuk