intel/compute-runtime #545: IGSC detection

servesh, 2022-06-06:
Adding IGSC (Intel(R) Graphics System Controller Firmware Update Library (IGSC FU)) (https://github.com/intel/igsc) sources to the same directory level as compute runtime, results in CMake picking up config from the sources instead of install location which results in this error.
```
CMake Error at /tmp/intel-gpu-umd/driver/igsc/lib/cmake/igscConfig.cmake:3 (include):
  include could not find requested file:

    /tmp/intel-gpu-umd/driver/igsc/lib/cmake/igscTargets.cmake
Call Stack (most recent call first):
  level_zero/CMakeLists.txt:82 (find_package)
```
It also seems the IGSC include directory is hardcoded in level_zero/CMakeLists.txt making it non-configurable.

How can one specify the location of the IGSC install directory?

jandres742, 2022-06-06:
@servesh it seems we haven't documented this properly and there's something missing in igsc.  NEO assumes dependencies to be in the same workspace as compute-runtime repo. Please see if following steps help:

```sh
# create workspace
mkdir ~/compute-runtime-workspace
cd ~/compute-runtime-workspace

# Instructions from https://github.com/intel/igsc
git clone https://github.com/intel/igsc.git
cd igsc
cmake -DSYSLOG:BOOL=OFF -DCMAKE_BUILD_TYPE=Release -G Ninja -S . -B build
meson setup build
meson configure -Dsyslog=true build
ninja -v -C build/

# this is the step that seems to be missing in igsc repo
find ../ -name "igscTargets.cmake" -type f | head -n 1 | xargs -IFILE cp FILE ./lib/cmake/

# download/install other dependencies

# now compute-runtime
cd ../
git clone https://github.com/intel/compute-runtime.git
cd compute-runtime
mkdir build
cd build
cmake ../

# and you should see the igsc found

-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
...
-- igsc Library headers directory: /home/user/compute-runtime-workspace/igsc/include
-- igsc version:
-- Prebuilt kernels are linked to Level Zero.
-- default igsc version:
-- default igsc version:
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/compute-runtime-workspace/compute-runtime/build
```

kgibala, 2025-11-21:
@anvesh-intel thanks for the fix.

@servesh  I am closing the issue as the changes are available in https://github.com/intel/compute-runtime/commit/864ba10e8a541eff5af289118bccc792f612482b