intel/compute-runtime #546: [Question] Shared and Host Buffers can offer the same overall performance on Intel Integrated Graphics?

jjfumero, 2022-06-10:
_Moved from here: https://github.com/oneapi-src/level-zero/issues/92_

I am interested in analyzing the overall performance (end-to-end applications) when using different types of buffer allocation. I wrote this blog-entry for reference:

https://jjfumero.github.io/posts/2022/05/overall-performance-of-unified-shared-memory-level-zero/

What I saw was that running an application with host buffers offers the same performance as running with shared memory buffers. My understanding is that, when running applications using shared memory buffers, the GPU driver can migrate the buffers from the host to the device, while host memory will be accessed from the device every time a data item is required. I have two scenarios: a) memory-bound and b) compute-bound. I was surprised to see that, when running the memory-bound case, the overall performance was very similar when allocating buffers using host memory only, and shared memory only. Is this performance expected when running on Intel Integrated graphics? Any pointers to validate this will be appreciated. 

If you want to reproduce all numbers, the whole application is available here: https://github.com/jjfumero/codeBlogArticles/tree/master/may2022/sharedMemoryEffect 

Setup and compute-runtime versions I used:
- Intel Core i9-10885H CPU running with Fedora 34 and the compute-runtime 21.38.21026.
- i7-12700K running Ubuntu 22.04 LTS. The compute-runtime installed is the 22.20.23198.

jandres742, 2022-06-10:
thanks @jjfumero . So you are correct in your understanding of how shared-memory works. However, if the shared-allocation is actually migrated or not depends on your target platform.

From your data above, you are using an i9, which is an integrated GPU. In this case, memory allocated for host and device reside in the same system memory. Consequently, there's no need to migrate a shared-allocation back and forth, since the host and device are using the same.

If you had a discrete GPU, the memory in the host would be separate from the memory in device. In that case, data may actually be migrated, and you would see a performance difference, either because the data was migrated, or because it was accessed through PCI.

Let us know if this clarifies it.

jjfumero, 2022-06-13:
Thank you @jandres742, that clarifies my question. That was also my guess when I looked at the numbers. 

So my take here is that,  when running on shared memory GPUs, there is no benefit in using device buffers (unless those buffers can be accessed across multiple GPUs directly). Host or shared memory buffers will perform much better. In fact, in these target platforms (shared memory GPUs), host and shared memory buffer types are, in practice, the same. 
Please, feel free correct me if I misunderstood.

jandres742, 2022-06-13:
@jjfumero . Correct.  There are some points to consider here:

- If you write your code creating allocations as you need (not as they behave in a given platform), then your code will need less maintenance in the future, when trying different platforms. 
- You could use https://spec.oneapi.io/level-zero/latest/core/api.html?highlight=integrated#_CPPv4N25ze_device_property_flag_t34ZE_DEVICE_PROPERTY_FLAG_INTEGRATEDE to check whether you are using integrated or discrete devices so you can take the proper path in your code.

jjfumero, 2022-06-14:
Thank you @jandres742. It is clearer now.

eero-t, 2022-06-14:
@jjfumero If everything's answered, could you close this?  Thanks!

closed 2022-06-14 by jjfumero