intel/compute-runtime #547: Zello-sysman tool does not support testing all L0 OneAPI spec metric types

eero-t, 2022-06-14:
https://github.com/intel/compute-runtime/blob/master/level_zero/tools/test/black_box_tests/zello_sysman.cpp#L1150

Is lacking options for testing metrics support for:
* PSUs: https://spec.oneapi.io/level-zero/latest/sysman/api.html#psu
* Fans: https://spec.oneapi.io/level-zero/latest/sysman/api.html#fan

Which have been supported by the public level-zero headers + loader already for months.

eero-t, 2022-06-15:
@jandres742 I would think how backend handles unimplemented frontend APIs to also be obviously of interest. E.g. to verify that those are also done according to spec / provide correct return value (see: https://github.com/intel/compute-runtime/issues/510).

While more extensive support for APIs could wait until they are properly supported by backend, I think test tool could have basic support for all APIs that just calls enumeration functions and lists how many items are available (if any).

eero-t, 2022-11-25:
PSU and Fan tests are still missing (no options listed for them by zello_sysman).

saik-intel, 2024-04-24:
@eero-t  zello_sysman is just BB test tool to validate the L0 driver internally and XPUManager is official tool for E2E validation. Do you want to add the hooks for zello_sysman too ? please confirm

eero-t, 2024-04-24:
> @eero-t zello_sysman is just BB test tool to validate the L0 driver internally

@saik-intel If it is indeed intended for BlackBox testing, then it makes even more sense to add all L0 metrics to it, regardless of whether backend has already support for them or not.   That way it can be used to test that backend does work as expected also when unsupported metrics are requested (through latest frontend)!

eero-t, 2025-11-12:
> We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

This is rather lazy request as checking this would have taken only couple of minutes, and would have shown that PSU metrics are still missing:
```
zello_sysman -h | grep -i -e fan -e psu
  -a,   --fan                                                                       selectively run fan black box test
```

Above was from latest version from September, but Git head shows same:
```
compute-runtime$ git pull
...
compute-runtime$ grep -i psu ./level_zero/tools/test/black_box_tests/zello_sysman.cpp
compute-runtime$ grep -i fan ./level_zero/tools/test/black_box_tests/zello_sysman.cpp
                 "\n  -a,   --fan                                                                       selectively run fan black box test"
...
```