intel/compute-runtime #548: clinfo not working with Intel Iris Xe Graphics

cvelloth, 2022-06-21:
Processor	11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz   1.80 GHz
Installed RAM	16.0 GB (15.7 GB usable)
System type	64-bit operating system, x64-based processor
Pen and touch	Touch support with 10 touch points

clinfo on wsl gives number of platforms 0

cvelloth, 2022-06-21:
root@cvelloth-mobl:/home/cvelloth/neo# strace -s 255 -f clinfo
execve("/usr/bin/clinfo", ["clinfo"], 0x7fffe664d5b8 /* 19 vars */) = 0
brk(NULL)                               = 0x7fffb90a0000
arch_prctl(0x3001 /* ARCH_??? */, 0x7fffc122dad0) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=81765, ...}) = 0
mmap(NULL, 81765, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fb9d038c000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libOpenCL.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\200A\0\0\0\0\0\0@\0\0\0\0\0\0\0@\241\0\0\0\0\0\0\0\0\0\0@\08\0\10\0@\0\34\0\33\0\1\0\0\0\5\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\354\206\0\0\0\0\0\0\354\206\0\0\0\0\0\0\0\0 \0\0\0\0\0\1\0\0\0\6\0\0\0\250\224\0\0\0\0\0\0\250\224 \0\0\0\0\0\250\224 \0\0\0\0\0d\v\0\0\0\0\0\0\250\v\0\0\0\0\0\0\0\0 \0\0\0\0\0\2\0\0\0\6\0\0\0\320\234\0\0\0\0\0\0\320\234 \0\0\0\0\0\320\234 \0\0\0\0\0\0\2\0\0\0\0\0\0\0\2\0\0\0\0\0\0\10\0\0\0\0\0\0\0\4\0\0\0\4\0\0\0\0\2\0\0\0\0\0\0\0\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=43072, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb9d0380000
mmap(NULL, 2138192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb9d0170000
mprotect(0x7fb9d0179000, 2097152, PROT_NONE) = 0
mmap(0x7fb9d0379000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7fb9d0379000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 \22\0\0\0\0\0\0@\0\0\0\0\0\0\0\340A\0\0\0\0\0\0\0\0\0\0@\08\0\v\0@\0\37\0\36\0\1\0\0\0\4\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0H\16\0\0\0\0\0\0H\16\0\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\5\0\0\0\0\20\0\0\0\0\0\0\0\20\0\0\0\0\0\0\0\20\0\0\0\0\0\0\211\21\0\0\0\0\0\0\211\21\0\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0000\0\0\0\0\0\0\0000\0\0\0\0\0\0\0000\0\0\0\0\0\0\300\6\0\0\0\0\0\0\300\6\0\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\6\0\0\0p=\0\0\0\0\0\0pM\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18848, ...}) = 0
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb9d0160000
mmap(0x7fb9d0161000, 8192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7fb9d0161000
mmap(0x7fb9d0163000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7fb9d0163000
mmap(0x7fb9d0164000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7fb9d0164000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\300A\2\0\0\0\0\0@\0\0\0\0\0\0\0\30\347\36\0\0\0\0\0\0\0\0\0@\08\0\16\0@\0D\0C\0\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0\20\3\0\0\0\0\0\0\20\3\0\0\0\0\0\0\10\0\0\0\0\0\0\0\3\0\0\0\4\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\34\0\0\0\0\0\0\0\34\0\0\0\0\0\0\0\20\0\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\350\24\2\0\0\0\0\0\350\24\2\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\5\0\0\0\0 \2\0\0\0\0\0\0 \2\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0\20\3\0\0\0\0\0\0\20\3\0\0\0\0\0\0\10\0\0\0\0\0\0\0\3\0\0\0\4\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\34\0\0\0\0\0\0\0\34\0\0\0\0\0\0\0\20\0\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\350\24\2\0\0\0\0\0\350\24\2\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\5\0\0\0\0 \2\0\0\0\0\0\0 \2\0\0\0\0\0\0 \2\0\0\0\0\0$v\27\0\0\0\0\0$v\27\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0\240\31\0\0\0\0\0\0\240\31\0\0\0\0\0\0\240\31\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\30x\346\264ur\f|Q\226\236i\253-'o\256m\35\356\4\0\0\0\20\0\0\0\1\0\0\0GNU\0\0\0\0\0\3\0\0\0\2\0\0\0\0\0\0\0", 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029592, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0\20\3\0\0\0\0\0\0\20\3\0\0\0\0\0\0\10\0\0\0\0\0\0\0\3\0\0\0\4\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\240\346\33\0\0\0\0\0\34\0\0\0\0\0\0\0\34\0\0\0\0\0\0\0\20\0\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\350\24\2\0\0\0\0\0\350\24\2\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\5\0\0\0\0 \2\0\0\0\0\0\0 \2\0\0\0\0\0\0 \2\0\0\0\0\0$v\27\0\0\0\0\0$v\27\0\0\0\0\0\0\20\0\0\0\0\0\0\1\0\0\0\4\0\0\0\0\240\31\0\0\0\0\0\0\240\31\0\0\0\0\0\0\240\31\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\30x\346\264ur\f|Q\226\236i\253-'o\256m\35\356\4\0\0\0\20\0\0\0\1\0\0\0GNU\0\0\0\0\0\3\0\0\0\2\0\0\0\0\0\0\0", 68, 880) = 68
mmap(NULL, 2037344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb9cff60000
mmap(0x7fb9cff82000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7fb9cff82000
mmap(0x7fb9d00fa000, 319488, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19a000) = 0x7fb9d00fa000
[... cut ...]
openat(AT_FDCWD, "/usr/lib/wsl/lib/libdxcore.so", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
close(3)                                = 0
fstat(1, {st_mode=S_IFCHR|0660, st_rdev=makedev(0x4, 0x1), ...}) = 0
ioctl(1, TCGETS, {B38400 opost isig icanon echo ...}) = 0
write(1, "Number of platforms                               0\n", 52Number of platforms                               0
) = 52
exit_group(0)                           = ?
+++ exited with 0 +++
root@cvelloth-mobl:/home/cvelloth/neo#

JablonskiMateusz, 2022-06-22:
Hi @cvelloth 
As per readme: `To allow NEO access to GPU device make sure user has permissions to files /dev/dri/renderD*.` 
In attached log I see you don't have such files:
```
openat(AT_FDCWD, "/dev/dri/by-path", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/dev/dri/renderD128", O_RDWR) = -1 ENOENT (No such file or directory)
fstat(-1, 0x7fffc122d120) = -1 EBADF (Bad file descriptor)
openat(AT_FDCWD, "/dev/dri/renderD129", O_RDWR) = -1 ENOENT (No such file or directory)
```

Please ensure that i915 module is correctly initialized.

cvelloth, 2022-06-23:
> Hi @cvelloth As per readme: `To allow NEO access to GPU device make sure user has permissions to files /dev/dri/renderD*.` In attached log I see you don't have such files:
>
> ```
> openat(AT_FDCWD, [...]

Hi @JablonskiMateusz ,

Missed to add that i am working in wsl with this. The issue is on the wsl version of ubuntu 20.04. The graphics driver is correctly installed and is updated to the latest version.
How can i check the i915 drivers in this case ?

Thanks,
Chandana