intel/compute-runtime #557: ocloc disasm error

houjingyi233, 2022-08-16:
On Ubuntu 20.04.4 LTS:

$ apt-get install intel-opencl-icd
Reading package lists... Done
Building dependency tree       
Reading state information... Done
intel-opencl-icd is already the newest version (22.23.23405+i755~u20.04).

$ cat test.cl
__kernel void f0(){
}

$ ocloc -file test.cl -device skl
Build succeeded.

$ ocloc disasm -file test_Gen9core.bin -device skl
Warning : Path to dump folder not specificed - using ./dump as default.
Path to patch list not provided - using defaults, skipping patchokens as undefined.
Trying to disassemble f0.krn
Error while disassembling with IGA!
Status msg: version mismatch

Why it showed version mismatch? Did I do something wrong? Thank you!

KrystianChmielewski, 2022-08-18:
Hey!
You can to download required packages from https://github.com/intel/compute-runtime/releases. I tried it with 22.23.23405 and faced no problem. Please try it and let me know if you are still facing this issue.

AdamCetnerowski, 2022-09-29:
Looks like issue is resolved. If there are new problems, feel free to open a new issue.

closed 2022-09-29 by AdamCetnerowski