intel/compute-runtime #558: DG2 support in prebuilt weekly releases?

oscarbg, 2022-08-26:
Hi,
just wanted to ask the status of Neo DG2 support? 
also if current prebuilt releases like https://github.com/intel/compute-runtime/releases/tag/22.33.23949 support DG2 "out of the box"? 
I guess not as release notes only show DG1 support.. 
if not, hope in the next few releases, support is added, now that DG2 is more "broadly" available on US and tech sites like Phoronix start getting it and using it (https://www.phoronix.com/review/intel-arc-graphics-linux)..
also "standard users" use it on Linux (https://twitter.com/Loeschzwerg_3DC/status/1562470315518672897)
thanks..

AdamCetnerowski, 2022-08-29:
DG2/Alchemist is supported in the releases. We will update the Release Notes / FAQ once we've switched over testing to an opensource kernel version that we can easily point to.

oscarbg, 2022-08-30:
nice.. thanks!

closed 2022-08-30 by oscarbg