intel/compute-runtime #561: Possible regression on single thread SPIRV kernel (#411)

qedawkins, 2022-09-07:
Hi,
the issue I'm having is the same as in #411. The previous issue was dispatching a similar kernel to the one below with a single thread
```
__kernel void copy(__global int* input, __global int* output) {
       output[0] = input[0];
} 
```
And in #411 it was fixed by updating the Intel OpenCL Drivers to `21.13.19438`. I'm now seeing the same problem on version `22.35.24055` (https://github.com/intel/compute-runtime/releases/tag/22.35.24055)
```
Device   : Intel(R) UHD Graphics 770 [0x4680]
Type     : GPU
Vendor ID: 8086
#Queue Groups: 1
size of pointer:8
size of pointer:8
Group X: 1
Group Y: 1
Group Z: 1
Discrepancy: 0 != 120021

Copy validation FAILED
```

The code I am using is based on the example in #411 and can be found here: https://gist.github.com/qedawkins/25568f6f06d86f0c820e90094451ac3e

### SPIRV Dump
Here is the disassembled SPIRV that I am using
```
; SPIR-V
; Version: 1.0
; Generator: Khronos LLVM/SPIR-V Translator; 14
; Bound: 12
; Schema: 0
               OpCapability Addresses
               OpCapability Kernel
          %1 = OpExtInstImport "OpenCL.std"
               OpMemoryModel Physical32 OpenCL
               OpEntryPoint Kernel %6 "copy"
         %11 = OpString "kernel_arg_type.copy.int*,int*,"
               OpSource OpenCL_C 102000
               OpName %a "a"
               OpName %b "b"
               OpName %entry "entry"
               OpDecorate %a FuncParamAttr NoCapture
               OpDecorate %b FuncParamAttr NoCapture
       %uint = OpTypeInt 32 0
       %void = OpTypeVoid
%_ptr_CrossWorkgroup_uint = OpTypePointer CrossWorkgroup %uint
          %5 = OpTypeFunction %void %_ptr_CrossWorkgroup_uint %_ptr_CrossWorkgroup_uint
          %6 = OpFunction %void None %5
          %a = OpFunctionParameter %_ptr_CrossWorkgroup_uint
          %b = OpFunctionParameter %_ptr_CrossWorkgroup_uint
      %entry = OpLabel
         %10 = OpLoad %uint %a Aligned 4
               OpStore %b %10 Aligned 4
               OpReturn
               OpFunctionEnd
```

I also find that if I change from Physical32 to Physical64 addressing the example works.
```
Device   : Intel(R) UHD Graphics 770 [0x4680]
Type     : GPU
Vendor ID: 8086
#Queue Groups: 1
size of pointer:8
size of pointer:8
Group X: 1
Group Y: 1
Group Z: 1
Validated: 120021 == 120021

Copy validation PASSED
```

Please let me know if there is anything else I can provide to reproduce the error. Thanks for any feedback!

bashbaug, 2022-09-24:
Hi, can you say how you are compiling your SPIR-V file?  Our pointer size is 64-bits, so you'll need to generate a SPIR-V file with Physical64 and not Physical32 addressing.  If the Physical32 file is working it is only by chance :smile: 

If you're compiling with Clang and the SPIR-V LLVM Translator you should be able to generate a 64-bit SPIR-V file by specifying `-target spir64` rather than `-target spir`.  Example: https://godbolt.org/z/6PszfbYqK

qedawkins, 2022-09-24:
I see, so the working cases we had before with Physical32 were only working by chance. There is no error when compiling with `-target spir64` so this issue can be closed then. Thanks for the response!

closed 2022-09-29 by AdamCetnerowski