intel/compute-runtime #562: Typos in zello_sysman.cpp

eero-t, 2022-09-15:
```
$ git grep avalialbe
level_zero/tools/test/black_box_tests/zello_sysman.cpp:        return "NOT SUPPORTED MODE Engine avalialbe";
level_zero/tools/test/black_box_tests/zello_sysman.cpp:        return "NOT SUPPORTED MODE Engine avalialbe";
level_zero/tools/test/black_box_tests/zello_sysman.cpp:        return "NOT SUPPORTED MODE Engine avalialbe";
```
Bugs:
* avalialbe -> available
* None of these error messages are for engine modes, they are for:
  * engine flag types
  * temperature sensor types
  * engine group types

saik-intel, 2022-09-29:
FIXED WITH PR # https://github.com/intel-innersource/drivers.gpu.compute.runtime/pull/14834#pullrequestreview-1125014492

JablonskiMateusz, 2022-10-04:
change is merged https://github.com/intel/compute-runtime/commit/183d82b02161b43b9f6172e8feda38522def416e

closed 2022-10-04 by JablonskiMateusz

eero-t, 2022-10-04:
Looks good, thanks!