intel/compute-runtime #563: The file 'LICENSE.md' is not a license

jeremiah, 2022-09-16:
It appears to be just a licensing header based on SPDX format. Is the all the source code under the MIT License? If the whole project is MIT licensed it might be best to change 'LICENSE.md' to a LICENSE file like https://github.com/intel/compute-runtime/blob/master/third_party/DirectX-Headers/LICENSE and put the MIT license with copyright in that top-level License file like the DirectX-Headers dir does.

JablonskiMateusz, 2022-09-21:
Hi @jeremiah, is there anything wrong with having `LICENSE.md` file? GitHub docs says that it is popular approach: https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository#determining-the-location-of-your-license

dvrogozh, 2022-09-21:
I think the question is not about a file name, but about a file contents. Which is:
```
Copyright (C) 2021 Intel Corporation

SPDX-License-Identifier: MIT
```

Instead of:
```
Copyright (c) 2021 Intel Corporation

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
```

This does make sense to me since Github API fails to correctly distinguish license type for compute-runtime. You can see that in the right pane of top level page - it just suggests to "View license" vs. listing it as "MIT license" as it does for other projects. Compare with MSDK for example:
* https://github.com/Intel-Media-SDK/MediaSDK - see right pane
* https://github.com/Intel-Media-SDK/MediaSDK/blob/master/LICENSE - and that's how license rendered when you look inside

JablonskiMateusz, 2022-10-13:
@jeremiah I provided fix in https://github.com/intel/compute-runtime/commit/33f5915062ae32d17c689873edf9b8431e474742

closed 2022-10-27 by JablonskiMateusz