intel/compute-runtime #564: [OpenCL] Deb-package's update-alternatives/ICD loader delpoyment is brutal

pjaaskel, 2022-09-20:
It seems the Debian package installs Intel-provided ICD loader using the update-alternatives mechanism. The annoying part is that the Ubuntu-shipped ocl-icd-loader doesn't seem to provide a target for update-alternatives, which makes it rather annoying to revert back to the system ICD loader. Could the package simply use the default ICD loader if it's installed?

JablonskiMateusz, 2022-10-04:
Hi @pjaaskel . Could you elaborate more what do you expect from our driver package?

pjaaskel, 2022-10-04:
Hi @JablonskiMateusz. It would be nice if it used the already packaged ocl-icd-loader instead of force-installing its own ICD loader. It could also install the .icds to the standard /etc/OpenCL/vendors location so it acts nicely next to other OpenCL implementations.

JablonskiMateusz, 2022-10-06:
Did you follow instructions from https://dgpu-docs.intel.com ?

pjaaskel, 2022-10-06:
I *think* the problem is that libOpenCL does not use the update-alternatives in the Debian/Ubuntu packages for ocl-icd-dev so I don't have the choice to set it as a priority.

I'm not familiar what get installed from which package in the oneapi/compute-runtime stack), but this seems to happen to after the oneapi installations/upgrades.

Some shell dumps:
```
 ls /etc/alternatives/*opencl* -l
lrwxrwxrwx 1 root root 21 loka    4 16:17 /etc/alternatives/opencl-intel-runtime -> /opt/intel/oneapi/lib
lrwxrwxrwx 1 root root 37 loka    4 16:17 /etc/alternatives/opencl-intel-runtime-icd -> /opt/intel/oneapi/lib/etc/intel64.icd
lrwxrwxrwx 1 root root 34 loka    4 16:17 /etc/alternatives/opencl-libOpenCL.so -> /opt/intel/oneapi/lib/libOpenCL.so
lrwxrwxrwx 1 root root 36 loka    4 16:17 /etc/alternatives/opencl-libOpenCL.so.1 -> /opt/intel/oneapi/lib/libOpenCL.so.1
lrwxrwxrwx 1 root root 38 loka    4 16:17 /etc/alternatives/opencl-libOpenCL.so.1.2 -> /opt/intel/oneapi/lib/libOpenCL.so.1.2
...
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ ls -la /usr/lib/x86_64-linux-gnu/libOpenCL.so*
lrwxrwxrwx 1 root root    37 loka    4 16:17 /usr/lib/x86_64-linux-gnu/libOpenCL.so -> /etc/alternatives/opencl-libOpenCL.so
lrwxrwxrwx 1 root root    39 loka    4 16:17 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 -> /etc/alternatives/opencl-libOpenCL.so.1
-rw-r--r-- 1 root root 69288 syys   30  2021 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0
lrwxrwxrwx 1 root root    41 loka    4 16:17 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.2 -> /etc/alternatives/opencl-libOpenCL.so.1.2
...
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ OCL_ICD_VENDORS=/etc/OpenCL/vendors/intel64.icd LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0 clinfo|grep CPU
  Device Type                                     CPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  Success (1)
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ OCL_ICD_VENDORS=/etc/OpenCL/vendors/intel64.icd LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libOpenCL.so.1 clinfo|grep CPU
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ dpkg -S /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0
ocl-icd-libopencl1:amd64: /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0
...
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ sudo apt remove ocl-icd-libopencl1 
...
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ sudo apt install ocl-icd-libopencl1 ocl-icd-opencl-dev clinfo
...
s -la /usr/lib/x86_64-linux-gnu/libOpenCL.so*
lrwxrwxrwx 1 root root    18 syys   30  2021 /usr/lib/x86_64-linux-gnu/libOpenCL.so -> libOpenCL.so.1.0.0
lrwxrwxrwx 1 root root    16 loka    6 18:04 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1 -> libOpenCL.so.1.2
-rw-r--r-- 1 root root 69288 syys   30  2021 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.0.0
lrwxrwxrwx 1 root root    41 loka    4 16:17 /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.2 -> /etc/alternatives/opencl-libOpenCL.so.1.2
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ dpkg -S /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.2
dpkg-query: no path found matching pattern /usr/lib/x86_64-linux-gnu/libOpenCL.so.1.2
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ cd /etc/alternatives/
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ ls -la opencl-libOpenCL.so.1.2
lrwxrwxrwx 1 root root 38 loka    4 16:17 opencl-libOpenCL.so.1.2 -> /opt/intel/oneapi/lib/libOpenCL.so.1.2
...
pjaaskel@pjaaskel-mobl1:/etc/alternatives$ OCL_ICD_VENDORS=/etc/OpenCL/vendors/intel64.icd clinfo|grep CPU
  Device Type                                     CPU
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_CPU)  Success (1)

```

pjaaskel, 2022-10-07:
I need to understand why it could be a distro problem if I see the oneapi package overwriting the ICD loader's libOpenCL.so from ocl-icd with update-alternatives symlinks. What type of bug I should report and where? 

To provide an 'opencl-icd' means that the package installs an .icd file which is basically a pointer to an OpenCL driver (platform implementation). The correct/default location for these .icd files is /etc/OpenCL/vendors. So, if the package provides an OpenCL driver and acts nicely with ICD loaders, this is a correct virtual package to provide.

Contrast this with an "icd loader", which is the dispatcher that loads these ICDs the different drivers provide and presents them as OpenCL platform alternatives to the OpenCL API clients. So indeed each OpenCL driver should provide one or more .icd files (opencl-icd) but not overwrite the ICD dispatcher (ocl-icd is now defacto standard in Debian/Ubuntu), like the compute-runtime package seems to do.

```
sudo apt show ocl-icd-libopencl1
Package: ocl-icd-libopencl1
Version: 2.2.14-3
Priority: extra
Section: universe/libs
Source: ocl-icd
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian OpenCL Maintainers <pkg-opencl-devel@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 143 kB
Provides: libopencl-1.1-1, libopencl-1.2-1, libopencl-2.0-1, libopencl-2.1-1, libopencl-2.2-1, libopencl-3.0-1, libopencl1
Depends: libc6 (>= 2.34)
Suggests: opencl-icd  <----------- Suggests, not provides.  
Conflicts: amd-app, libopencl1, nvidia-libopencl1-dev
Replaces: amd-app, libopencl1, nvidia-libopencl1-dev
Homepage: https://github.com/OCL-dev/ocl-icd
Task: kubuntu-desktop, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop, ubuntu-budgie-desktop-raspi
Download-Size: 39,1 kB
APT-Manual-Installed: yes
APT-Sources: http://fi.archive.ubuntu.com/ubuntu jammy/universe amd64 Packages
Description: Generic OpenCL ICD Loader
 OpenCL (Open Computing Language) is a multivendor open standard for
 general-purpose parallel programming of heterogeneous systems that include
 CPUs, GPUs and other processors.
 .
 This package contains an installable client driver loader (ICD Loader)
 library that can be used to load any (free or non-free) installable client
 driver (ICD) for OpenCL. It acts as a demultiplexer so several ICD can
 be installed and used together.
```

tjaalton, 2022-10-07:
sounds like your debs should use dpkg-divert instead

eero-t, 2022-10-07:
@pjaaskel Timo maintains many of the GPU related packages in Debian and Ubuntu distributions.  I asked his opinion on what he thinks the "compute-runtime" packages should do, and he answered that question, not what you should do...  Sorry for the confusion.

PS. although I work for Intel, I'm not "compute-runtime" maintainer either, just another of its users.

pjaaskel, 2022-10-07:
Alright, thanks for the clarification :)

kgibala, 2025-11-12:
Hi @pjaaskel,

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!