intel/compute-runtime #565: Publish the newest releases on conda-forge (for Python)

ogrisel, 2022-09-21:
This would make it easy to use Intel GPUs via OpenCL in Python via `dpnp` / `dpctl` /`numba-dpex`.

The last few release have had failed builds and nobody fixed them:

- https://github.com/conda-forge/intel-compute-runtime-feedstock/pulls

I think it would be great if at least one maintainer of `intel/compute-runtime` would request to be registered as a [recipe maintainer](https://github.com/conda-forge/intel-compute-runtime-feedstock/blob/eab90fb94b538d0a81a9b90d375dc41e3be60a27/recipe/meta.yaml#L42-L44) on `conda-forge/intel-compute-runtime-feedstock` in addition to @isuruf to adjust the build recipe on the conda-forge CI when needed.

ogrisel, 2022-09-21:
Related issue for `intel-level-zero` on conda-forge: https://github.com/oneapi-src/level-zero/issues/98

AdamCetnerowski, 2022-11-24:
There are no plans for proactive enabling and distro adoption except for Ubuntu. We are open to accepting contributions (pull requests) that unblock support.

closed 2022-11-24 by AdamCetnerowski

ogrisel, 2023-05-04:
I think not providing a way to install all the runtime dependencies for packages such as `dpctl`, `dpnp` and `numba_dpex` to users without sudo rights is a very strong impediment for adoption of the Intel xPU technology stack in the Python datascience ecosystem.

Publishing such packages on conda channels such as `intel` would be an improvement but it would even be better to directly contribute to maintain packages on the main `conda-forge` channels or on pypi.org to ease dependency management of third party library developers that would like to ship software on this stack.

> There are no plans for proactive enabling and distro adoption except for Ubuntu.

BTW: #588 was also closed and was specifically about installation/packaging problems on Ubuntu.

> We are open to accepting contributions (pull requests) that unblock support.

I am already maintaining way too many open source packages myself, I do not have the bandwidth to handle the packaging of a system level libraries I only interact with via higher level components such as the aforementioned packages.