intel/compute-runtime #567: zeCommandListAppendMemoryFill fails to set uint16_t value when starting from none-zero offset

pengtu, 2022-09-27:
This problem is found in (https://github.com/CHIP-SPV/chip-spv/issues/132) test.  Allocate a device buffer and fill it with 16 bit values.  If the fill starts from the beginning of the device buffer, it works properly.  If starting from offset 1 (2 bytes from the buffer start), the value is not filled.

Here is a simple cutdown test case:

#include <cassert>
#include <climits>
#include <cstdio>
#include <cstdlib>
#include <cstring>
#include <level_zero/ze_api.h>
#include <iostream>
#include <vector>
#include <limits>

#define check(ans)                                                             \
  { do_check((ans), __FILE__, __LINE__); }
void do_check(ze_result_t code, const char *file, int line) {
  if (code != ZE_RESULT_SUCCESS) {
    fprintf(stderr, "Failed: %d at %s %d\n", code, file, line);
    exit(1);
  }
}

int main()
{
  // Initialize driver
  check(zeInit(ZE_INIT_FLAG_GPU_ONLY));

  // Retrieve driver
  uint32_t driverCount = 0;
  check(zeDriverGet(&driverCount, nullptr));

  ze_driver_handle_t driverHandle;
  check(zeDriverGet(&driverCount, &driverHandle));

  ze_context_desc_t contextDesc = {};
  ze_context_handle_t context;
  check(zeContextCreate(driverHandle, &contextDesc, &context));

  // Retrieve device
  uint32_t deviceCount = 0;
  check(zeDeviceGet(driverHandle, &deviceCount, nullptr));

  ze_device_handle_t device;
  deviceCount = 1;
  check(zeDeviceGet(driverHandle, &deviceCount, &device));

  // Print some properties
  ze_device_properties_t deviceProperties = {};
  check(zeDeviceGetProperties(device, &deviceProperties));

  // Create command queue
  uint32_t numQueueGroups = 0;
  check(zeDeviceGetCommandQueueGroupProperties(device, &numQueueGroups, nullptr));
  if (numQueueGroups == 0)
  {
    return 1;
  }
  std::vector<ze_command_queue_group_properties_t> queueProperties(numQueueGroups);
  check(zeDeviceGetCommandQueueGroupProperties(device, &numQueueGroups,
                                               queueProperties.data()));

  ze_command_queue_handle_t command_queue;
  ze_command_queue_desc_t cmdQueueDesc = {};

  for (uint32_t i = 0; i < numQueueGroups; i++)
  {
    if (queueProperties[i].flags & ZE_COMMAND_QUEUE_GROUP_PROPERTY_FLAG_COMPUTE)
    {
      cmdQueueDesc.ordinal = i;
    }
  }
  cmdQueueDesc.index = 0;
  cmdQueueDesc.mode = ZE_COMMAND_QUEUE_MODE_ASYNCHRONOUS;
  check(zeCommandQueueCreate(context, device, &cmdQueueDesc, &command_queue));

  // Create command list
  ze_command_list_handle_t command_list, command_list_1;
  ze_command_list_desc_t cmdListDesc = {};
  cmdListDesc.commandQueueGroupOrdinal = cmdQueueDesc.ordinal;
  check(zeCommandListCreate(context, device, &cmdListDesc, &command_list));

  size_t numElements = 10;
  size_t Nbytes = numElements * sizeof(uint16_t);
  uint16_t memsetval = 16;
  uint16_t *A_d;

  ze_device_mem_alloc_desc_t device_desc;
  device_desc.ordinal = 0;

  check(zeMemAllocDevice(context, &device_desc, Nbytes, 0, device, (void **)(&A_d)));
  uint16_t *A_h = reinterpret_cast<uint16_t *>(malloc(Nbytes));

  int offset = 1;
  check(zeCommandListAppendMemoryFill(command_list, A_d + offset, &memsetval, sizeof(uint16_t),
                                      (numElements - offset) * sizeof(uint16_t), nullptr, 0, nullptr));
  check(zeCommandListAppendBarrier(command_list, nullptr, 0, nullptr));
  check(zeCommandListClose(command_list));

  check(zeCommandQueueExecuteCommandLists(command_queue, 1, &command_list, nullptr));
  check(zeCommandQueueSynchronize(command_queue, std::numeric_limits<uint64_t>::max()));

  check(zeCommandListCreate(context, device, &cmdListDesc, &command_list_1));
  check(zeCommandListAppendMemoryCopy(command_list_1, A_h, A_d, Nbytes, nullptr, 0, nullptr));
  check(zeCommandListAppendBarrier(command_list_1, nullptr, 0, nullptr));
  check(zeCommandListClose(command_list_1));
  check(zeCommandQueueExecuteCommandLists(command_queue, 1, &command_list_1, nullptr));
  check(zeCommandQueueSynchronize(command_queue, std::numeric_limits<uint64_t>::max()));

  for (size_t i = offset; i < numElements; i++)
  {
    printf("A_h[%zu] = %d\n", i, A_h[i]);
    fflush(stdout);
    assert(A_h[i] == memsetval);
  }
}

The expected value is (memsetvalue = 16).  If we remove the assert in the final checking loop, the 9 values from A_h[] are:

A_h[1] = 0
A_h[2] = 0
A_h[3] = 0
A_h[4] = 0
A_h[5] = 0
A_h[6] = 0
A_h[7] = 0
A_h[8] = 16
A_h[9] = 0

Strangely only the A_h[8] gets filled correctly.

If we change the 'offset' from 1 to 0, then all 10 values are filled properly.

pengtu, 2022-09-27:
[](https://github.com/CHIP-SPV/chip-spv/issues/132)

pvelesko, 2023-04-27:
@pengtu updates for this?

JablonskiMateusz, 2023-05-11:
@pvelesko  the fix is merged https://github.com/intel/compute-runtime/commit/a68534f464f8abf4d87e4e8a71e6b9717a782309

closed 2024-02-01 by pengtu