intel/compute-runtime #569: Remove vaGetLibFunc use-cases

evelikov, 2022-10-02:
The `vaGetLibFunc` function is a wrapper around `dlsym()`.

As we consider that driver-internal function is fetched (aka `DdiMedia_ExtGetSurfaceHandle`) it makes more sense to directly use `dlsym()` be that by `dlopen()-ing` the driver or using the `RTLD_DEFAULT` aka `dlsym(RTLD_DEFAULT, ...)`.

JablonskiMateusz, 2022-11-04:
Hi @evelikov, according to VA sharing documentation `vaGetLibFunc` is a function to get pointers to other functions. Could you explain why you think we should use `dlsym` instead?

evelikov, 2022-11-04:
libva is a vendor agnostic frontend library, while the API is a vendor specific hack. As such should not exist in the first place.

Mind you, the compute-runtime driver is free to use a behind-the-scenes side channel for communication, but that channel should not be exposed to normal users. In other words it should not be an API exposed via libva.

Since it's a dummy wrapper, replacing it with dlopen/dlsym should just work. Saying should, since I don't have the hardware to actually test it myself.

Happy to make a PR though, assuming someone is willing to test for me?

JablonskiMateusz, 2022-11-08:
@evelikov could you get feedback from VA-API spec architects?

closed 2023-01-12 by AdamCetnerowski

evelikov, 2023-01-17:
Been distracted with more severe things lately. Will circle back once I have the chance.