intel/compute-runtime #570: Missing gpu from clinfo when installing 2 RAM modules

v-lopez, 2022-10-06:
I have an intel 1260p.

With one 16GB RAM module, I can see the gpu device and use it.
If I install a second module, the computer works as expected, except that clinfo displays no devices.

This is the output of clinfo and lspci with one module: 

[lspci 1ram.txt](https://github.com/intel/compute-runtime/files/9726825/lspci.1ram.txt)

[clinfo 1ram.txt](https://github.com/intel/compute-runtime/files/9726823/clinfo.1ram.txt)

And this is with 2 modules:
[clinfo 2ram.txt](https://github.com/intel/compute-runtime/files/9726837/clinfo.2ram.txt)
[lspci 2ram.txt](https://github.com/intel/compute-runtime/files/9726838/lspci.2ram.txt)

JablonskiMateusz, 2022-10-07:
could you attach logs from strace?

v-lopez, 2022-10-07:
Sure:
``` 
execve("/usr/bin/clinfo", ["clinfo"], 0x7ffc500ef860 /* 32 vars */) = 0
brk(NULL)                               = 0x561fd323d000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffe7ddd1bf0) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=79557, ...}) = 0
mmap(NULL, 79557, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8284271000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libOpenCL.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\200A\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=43072, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f828426f000
mmap(NULL, 2138192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8284064000
mprotect(0x7f828406d000, 2097152, PROT_NONE) = 0
mmap(0x7f828426d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7f828426d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 \22\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18816, ...}) = 0
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f828405e000
mmap(0x7f828405f000, 8192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f828405f000
mmap(0x7f8284061000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f8284061000
mmap(0x7f8284062000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f8284062000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360A\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\237\333t\347\262\27\320l\223\27*\202C\370T\177"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029560, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\237\333t\347\262\27\320l\223\27*\202C\370T\177"..., 68, 880) = 68
mmap(NULL, 2037344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8283e6c000
mmap(0x7f8283e8e000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f8283e8e000
mmap(0x7f8284006000, 319488, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19a000) = 0x7f8284006000
mmap(0x7f8284054000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7f8284054000
mmap(0x7f828405a000, 13920, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f828405a000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8283e69000
arch_prctl(ARCH_SET_FS, 0x7f8283e69740) = 0
mprotect(0x7f8284054000, 16384, PROT_READ) = 0
mprotect(0x7f8284062000, 4096, PROT_READ) = 0
mprotect(0x7f828426d000, 4096, PROT_READ) = 0
mprotect(0x561fd18b3000, 4096, PROT_READ) = 0
mprotect(0x7f82842b2000, 4096, PROT_READ) = 0
munmap(0x7f8284271000, 79557)           = 0
openat(AT_FDCWD, "/etc/OpenCL/vendors", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x2), ...}) = 0
brk(NULL)                               = 0x561fd323d000
brk(0x561fd325e000)                     = 0x561fd325e000
write(1, "Number of platforms             "..., 52Number of platforms                               0
) = 52
exit_group(0)                           = ?
+++ exited with 0 +++
```

v-lopez, 2022-10-07:
Sorry I usually run it inside a docker, here is the proper strace:
```
execve("/usr/bin/clinfo", ["clinfo"], 0x7fffde571790 /* 9 vars */) = 0
brk(NULL)                               = 0x563828b28000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffdd5cce050) = -1 EINVAL (Invalid argument)
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=217991, ...}) = 0
mmap(NULL, 217991, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f8b2081c000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libOpenCL.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\200A\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=43072, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8b2081a000
mmap(NULL, 2138192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8b2060f000
mprotect(0x7f8b20618000, 2097152, PROT_NONE) = 0
mmap(0x7f8b20818000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7f8b20818000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0 \22\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=18816, ...}) = 0
mmap(NULL, 20752, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8b20609000
mmap(0x7f8b2060a000, 8192, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f8b2060a000
mmap(0x7f8b2060c000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f8b2060c000
mmap(0x7f8b2060d000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3000) = 0x7f8b2060d000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\360A\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\237\333t\347\262\27\320l\223\27*\202C\370T\177"..., 68, 880) = 68
fstat(3, {st_mode=S_IFREG|0755, st_size=2029560, ...}) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0\20\0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0", 32, 848) = 32
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0\237\333t\347\262\27\320l\223\27*\202C\370T\177"..., 68, 880) = 68
mmap(NULL, 2037344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f8b20417000
mmap(0x7f8b20439000, 1540096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f8b20439000
mmap(0x7f8b205b1000, 319488, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x19a000) = 0x7f8b205b1000
mmap(0x7f8b205ff000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1e7000) = 0x7f8b205ff000
mmap(0x7f8b20605000, 13920, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f8b20605000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f8b20414000
arch_prctl(ARCH_SET_FS, 0x7f8b20414740) = 0
mprotect(0x7f8b205ff000, 16384, PROT_READ) = 0
mprotect(0x7f8b2060d000, 4096, PROT_READ) = 0
mprotect(0x7f8b20818000, 4096, PROT_READ) = 0
mprotect(0x5638277ce000, 4096, PROT_READ) = 0
mprotect(0x7f8b2087f000, 4096, PROT_READ) = 0
munmap(0x7f8b2081c000, 217991)          = 0
openat(AT_FDCWD, "/etc/OpenCL/vendors", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 3
fstat(3, {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
brk(NULL)                               = 0x563828b28000
brk(0x563828b49000)                     = 0x563828b49000
getdents64(3, /* 3 entries */, 32768)   = 80
getdents64(3, /* 0 entries */, 32768)   = 0
lseek(3, 0, SEEK_SET)                   = 0
getdents64(3, /* 3 entries */, 32768)   = 80
openat(AT_FDCWD, "/etc/OpenCL/vendors/intel.icd", O_RDONLY) = 4
fstat(4, {st_mode=S_IFREG|0644, st_size=52, ...}) = 0
fstat(4, {st_mode=S_IFREG|0644, st_size=52, ...}) = 0
lseek(4, 0, SEEK_SET)                   = 0
read(4, "/usr/lib/x86_64-linux-gnu/intel-"..., 52) = 52
lseek(4, 52, SEEK_SET)                  = 52
close(4)                                = 0
openat(AT_FDCWD, "/usr/lib/x86_64-linux-gnu/intel-opencl/libigdrcl.so", O_RDONLY|O_CLOEXEC) = 4
read(4, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\350\4\0\0\0\0\0"..., 832) = 832
fstat(4, {st_mode=S_IFREG|0644, st_size=10126008, ...}) = 0
mmap(NULL, 10762480, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 4, 0) = 0x7f8b1f9d0000
mprotect(0x7f8b1fa1d000, 8028160, PROT_NONE) = 0
mmap(0x7f8b1fa1d000, 6860800, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 4, 0x4d000) = 0x7f8b1fa1d000
[... 842 lines cut ...]
ioctl(4, DRM_IOCTL_GEM_CLOSE, 0x7ffdd5ccdc80) = 0
ioctl(4, DRM_IOCTL_I915_GEM_WAIT or DRM_IOCTL_RADEON_GEM_OP, 0x7ffdd5ccdca0) = 0
ioctl(4, DRM_IOCTL_GEM_CLOSE, 0x7ffdd5ccdc80) = 0
munmap(0x7f8b1e58d000, 8654848)         = 0
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_DESTROY, 0x7ffdd5ccdcd0) = 0
ioctl(4, DRM_IOCTL_I915_GEM_WAIT or DRM_IOCTL_RADEON_GEM_OP, 0x7ffdd5ccdca0) = 0
ioctl(4, DRM_IOCTL_GEM_CLOSE, 0x7ffdd5ccdc80) = 0
ioctl(4, DRM_IOCTL_I915_GEM_WAIT or DRM_IOCTL_RADEON_GEM_OP, 0x7ffdd5ccdca0) = 0
[... cut ...]
ioctl(4, DRM_IOCTL_I915_GEM_CONTEXT_DESTROY, 0x7ffdd5ccdcd0) = 0
futex(0x563828d20bc0, FUTEX_WAKE_PRIVATE, 2147483647) = 1
futex(0x7f8b1f5ce9d0, FUTEX_WAIT, 780, NULL) = -1 EAGAIN (Resource temporarily unavailable)
ioctl(4, DRM_IOCTL_GEM_CLOSE, 0x7ffdd5ccdd70) = 0
ioctl(4, DRM_IOCTL_I915_GEM_WAIT or DRM_IOCTL_RADEON_GEM_OP, 0x7ffdd5ccdd80) = 0
ioctl(4, DRM_IOCTL_GEM_CLOSE, 0x7ffdd5ccdd60) = 0
close(4)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```

v-lopez, 2022-10-07:
I see this on dmesg as well:
```
Your graphics device 46a6 is not properly supported by the driver in this
               kernel version. To force driver probe anyway, use i915.force_probe=46a6
               module parameter or CONFIG_DRM_I915_FORCE_PROBE=46a6 configuration option,
               or (recommended) check for kernel updates.
```

How can adding  a RAM module cause this?

v-lopez, 2022-10-07:
Adding i915.force_probe=46a6 fix the issue, but I am curious as to why the id changes.

JablonskiMateusz, 2022-10-12:
It doesn't seem to be an issue with OCL runtime. Could you please close this issue and file an issue against i915 kernel?

v-lopez, 2022-10-17:
Ok, thanks.

closed 2022-10-17 by v-lopez