intel/compute-runtime #572: Running Ubuntu 20.04 Docker image on Ubuntu 22.04

SebastianGrans, 2022-10-13:
I was trying to run a Docker image based on ubuntu 20.04 on a machine running 22.04 which resulted in the GPU not being detected.
Is it possible to make this work? 

The machine was running the latest release (22.39.24347), but I also tried running release 20.13.16352 which is the latest supported by 20.04, but that didn't work either. 

```bash
dockerissue$ docker run --rm --device /dev/dri minimal-example20.04 clinfo
Number of platforms                               0
dockerissue$ docker run --rm --device /dev/dri minimal-example22.04 clinfo  
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  ...
```

**Dockerfile:**
```Dockerfile
FROM ubuntu:22.04 # or ubuntu:20.04
RUN apt-get update && apt-get upgrade --assume-yes
RUN apt-get install intel-opencl-icd clinfo --assume-yes
```

JablonskiMateusz, 2022-10-13:
could you attach logs from strace when running clinfo?

SebastianGrans, 2022-10-17:
Pinging @JablonskiMateusz :)

eero-t, 2022-10-24:
According to strace, "clinfo" loads the Intel driver and noticed the GPU device (I'm assuming `renderD128` is the correct one).

> I was trying to run a Docker image based on ubuntu 20.04 on a machine running 22.04 which resulted in the GPU not being detected. Is it possible to make this work?
>
> The machine was running the [...]

Except for the kernel, what drivers you have installed on the host, does not affect HW support in your containers.

You did not mention which HW you have.  It's possible that Intel HW user-space drivers in your Ubuntu 20.04 container just are too old for your HW.

If Ubuntu 20.04 HWE (HW enabling) packages are not enough (Ubuntu HWE stack may include just kernel and Mesa 3D drivers), you can install latest Intel (user-space) driver packages to your containers as documented here: https://dgpu-docs.intel.com/installation-guides/index.html

PS. Note that in case of Intel dGPU (discrete GPU) support, because all of dGPU support is not yet in upstream, there are / can be dGPU related uAPI differences.  Therefore your host kernel i915 GPU driver needs to match the user-space drivers in your containers (and on your host if you want things to work also on host) i.e. be from same repo.

(Because of this potential API disconnect and how it limits your kernel selection, I would recommend using Ubuntu's own (HWE) drivers *when* those are new enough to support your HW, and using Intel repository packages only until full driver support for given HW is in upstream / Ubuntu.)

JablonskiMateusz, 2022-11-21:
Hi @SebastianGrans 
could you provide which driver version do you have on ubuntu 20 image and what GPU do you have (device id would be specific enough). As @eero-t  said, I think this gpu may not be supported yet in the driver on your ubuntu 20 image

kgibala, 2025-11-12:
Hi @SebastianGrans,

We’d like to know if this issue is still affecting you. If so, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

kgibala, 2025-12-17:
Hi @SebastianGrans,

We're closing this issue due to inactivity. We haven't received the requested information or updates for an extended period.

If you encounter this or a similar problem in the future, please feel free to create a new issue. When submitting, please follow our [Issue Submission Guide](https://github.com/intel/compute-runtime/blob/master/ISSUE_SUBMISSION_GUIDE.md) to help us understand and address your concern efficiently.