intel/compute-runtime #573: UUID mismatch between Vulkan and OpenCL

chrismile, 2022-10-15:
When using Vulkan-OpenCL interoperability, one needs to match a Vulkan device to an OpenCL device (or vice versa). On NVIDIA, cl_khr_device_uuid yields the same device UUID as VkPhysicalDeviceIDProperties::driverUUID. However, for the Intel Mesa driver and the Intel Compute Runtime, there seems to be a mismatch in device UUIDs. For a Intel HD Graphics 530 GPU, I get the Vulkan device UUID 7450d249-b957-0fa9-6c7b-d2e31620dbf5, but the OpenCL device UUID 86800000-1b19-0000-0000-000000000000.

The OpenCL spec states: "Device UUIDs must be **immutable** for a given device across processes, **driver APIs**, driver versions, and system reboots." (https://registry.khronos.org/OpenCL/specs/3.0-unified/html/OpenCL_Ext.html). Thus, I believe this is probably a bug in the Intel Compute Runtime implementation of cl_khr_device_uuid, right?

Below, you can find the output of vulkaninfo and clinfo. I used the latest drivers as available on Manjaro.

[vulkaninfo_i530.txt](https://github.com/intel/compute-runtime/files/9792766/vulkaninfo_i530.txt)
[clinfo_i530.txt](https://github.com/intel/compute-runtime/files/9792767/clinfo_i530.txt)

nyanmisaka, 2022-10-21:
Intel has never implemented Vulkan-OpenCL interop.

chrismile, 2022-10-21:
> Intel has never implemented Vulkan-OpenCL interop.

In my application I use fallback code for going the OpenCL -> CPU -> Vulkan route when the necessary extensions are not yet supported by the OpenCL implementation. However, in this case it is still necessary to somehow match devices. For example, on systems with an NVIDIA dGPU and an Intel iGPU, the NVIDIA dGPU may still show up when enumerating the OpenCL devices even though the GPU is disabled through NVIDIA Prime. Also, according to https://github.com/intel/compute-runtime/issues/166, there is active work being done on implementing cl_khr_gl_sharing. The same problem as for Vulkan would also happen with cl_khr_gl_sharing on the OpenCL side and GL_DRIVER_UUID_EXT on the OpenGL side if the Intel Compute runtime does not have a standard-conformant implementation of cl_khr_device_uuid where the device UUID is actually immutable across driver APIs. The OpenCL spec stating that the device UUID must be immutable across driver APIs is also completely independent of the existence of Vulkan or OpenGL interop extensions, by the way.

nyanmisaka, 2022-10-21:
Then if the UUID is not reliable, you have to use deviceID and vendorID to filter devices. This is just a stopgap because filtering can become ambiguous when multiple identical GPUs are present on a system.

chrismile, 2022-10-21:
> Then if the UUID is not reliable, you have to use deviceID and vendorID to filter devices. This is just a stopgap because filtering can become ambiguous when multiple identical GPUs are present on a system. The Vulkan code [in ffmpeg](https://github.com/FFmpeg/FFmpeg/blob/4cfffdd5519516ee7b6b440920d0aededc4bc06f/libavutil/hwcontext_vulkan.c#L781-L908) is doing what I said.

Yes, that's a good workaround, thanks for the link! In the long run, it would of course still be good if the problem with cl_khr_device_uuid could be fixed in Intel Compute Runtime.

nyanmisaka, 2022-10-21:
Definitely. They still need to work with the Mesa ANV team to ensure UUIDs are conformant to the OpenCL 3.0 specs.

kallaballa, 2022-10-28:
> Also, according to #166, there is active work being done on implementing cl_khr_gl_sharing.
I am actually using it on my Tigerlake and currently I am working on merging with upstream.
I implemented some demos to showcase it: https://github.com/kallaballa/GCV and there are more to come that i haven't published yet. :)

JablonskiMateusz, 2023-01-13:
Change is merged on our side: https://github.com/intel/compute-runtime/commit/ad579be87c1ba258f2143f0c9e63daf8ff0dd420

nyanmisaka, 2023-01-13:
As a side note the Mesa side patch is in pending:

https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/20408

JablonskiMateusz, 2023-03-22:
fix in https://github.com/intel/compute-runtime/releases/tag/23.05.25593.11