intel/compute-runtime #575: zeCommandQueueSynchronize() synchronizes other command queues as well

Sarbojit2019, 2022-10-18:
As per the spec, **zeCommandQueueSynchronize()** will synchronize the command queue passed to it but it is synchronizing unrelated queues as well. I have created a small reproducer of the issue reported CHIP-SPV/chip-spv#207, reproducer does as below

1> Create 2 command queues (say, q1 & q2)
2>  In q1, add a blocking event
3> Create a host thread and sleep for 10 secs, once completed signal blocking event pushed in step2
4> From host thread call zeCommandQueueSynchronize() for **q2**

**Issue:** Execution gets blocked on step 4 until task in q1 gets completed. 
**Expectation:** Step4 should be completed quickly as q2 is empty and task in q1 is not related to q2.

```
#include <level_zero/ze_api.h>
#include <stdio.h>
#include <iostream>
#include <vector>
#include <limits>
#include <cassert>
#include <unistd.h>
#include <thread>

#define check(status) { ze_result_t code = status;\
if (code != ZE_RESULT_SUCCESS){\
    fprintf(stderr,"Failed with error %d\n", code);\
    exit(1);}\
}

void callback(ze_event_handle_t signalBlockedEvent, ze_event_handle_t waitEvent) {
  printf("Going to sleep for 10 secs\n");
  sleep(10);
  printf("Just woker-up, going to signal the event\n");
  check(zeEventHostSignal(signalBlockedEvent));
  check(zeEventHostSynchronize(waitEvent, std::numeric_limits<uint64_t>::max()));
  printf("Callback is completed\n");
}

int main() {
  // init
  check(zeInit(ZE_INIT_FLAG_GPU_ONLY));

  // get driver 
  uint32_t driverCount = 0;
  check(zeDriverGet(&driverCount, nullptr));
  if (driverCount > 1) {
      printf("Warning!! More than one driver found\n");
      driverCount = 1;
  }
  ze_driver_handle_t driverHandle;
  check(zeDriverGet(&driverCount, &driverHandle));

  // get device
  uint32_t deviceCount = 0;
  check(zeDeviceGet(driverHandle, &deviceCount, nullptr));
  if (deviceCount > 1) {
      printf("Warning!! more than one device is found\n");
      deviceCount = 1;
  }
  ze_device_handle_t deviceHandle;
  check(zeDeviceGet(driverHandle, &deviceCount, &deviceHandle));

  // create context
  ze_context_desc_t ctxDesc;
  ze_context_handle_t ctxHandle;
  check(zeContextCreate(driverHandle, &ctxDesc, &ctxHandle));

  // Create Events
  ze_event_pool_desc_t EventPoolDesc = {ZE_STRUCTURE_TYPE_EVENT_POOL_DESC, nullptr, ZE_EVENT_POOL_FLAG_HOST_VISIBLE, 1};
  ze_event_pool_handle_t eventPoolHandle;
  check(zeEventPoolCreate(ctxHandle, &EventPoolDesc, 0, nullptr, &eventPoolHandle));

  ze_event_desc_t EventDesc = {ZE_STRUCTURE_TYPE_EVENT_DESC,nullptr,0,ZE_EVENT_SCOPE_FLAG_HOST,ZE_EVENT_SCOPE_FLAG_HOST};
  int eventListSize = 15;
  std::vector<ze_event_handle_t> eventList(eventListSize);
  for (int i=0; i< eventListSize; ++i) {
    check(zeEventCreate(eventPoolHandle, &EventDesc, &eventList[i]));
  }
  int nextEventIndex = 1; // index=0 is reserved

  // Create queues
  uint32_t numQueueGrp = 0;
  check(zeDeviceGetCommandQueueGroupProperties(deviceHandle, &numQueueGrp, nullptr));
  if (numQueueGrp == 0) {
      printf("Error!! number of queue group is zero\n");
      return 0;
  }
  std::vector<ze_command_queue_group_properties_t> queueProperties(numQueueGrp);
  check(zeDeviceGetCommandQueueGroupProperties(deviceHandle, &numQueueGrp,
                                              queueProperties.data()));
  ze_command_queue_desc_t cmdQueueDesc = {};
  for(int i=0; i<numQueueGrp; ++i) {
      if (queueProperties[i].flags & ZE_COMMAND_QUEUE_GROUP_PROPERTY_FLAG_COMPUTE) {
          cmdQueueDesc.ordinal = i;
      }
  }
  cmdQueueDesc.index = 0;
  cmdQueueDesc.mode = ZE_COMMAND_QUEUE_MODE_ASYNCHRONOUS;

  ze_command_queue_handle_t command_queue1, command_queue2;
  check(zeCommandQueueCreate(ctxHandle, deviceHandle, &cmdQueueDesc, &command_queue1));
  check(zeCommandQueueCreate(ctxHandle, deviceHandle, &cmdQueueDesc, &command_queue2));

  // mimic callback
  ze_command_list_handle_t command_lists_callback[3];
  ze_command_list_desc_t cmdListDesc = {ZE_STRUCTURE_TYPE_COMMAND_LIST_DESC, 0, 0, 0};
  cmdListDesc.commandQueueGroupOrdinal = cmdQueueDesc.ordinal;

  check(zeCommandListCreate(ctxHandle, deviceHandle, &cmdListDesc, &command_lists_callback[1]));
  auto blockedEvent = eventList[0];
  check(zeEventHostReset(blockedEvent));
  check(zeEventHostReset(eventList[++nextEventIndex]));
  check(zeCommandListAppendBarrier(command_lists_callback[1], eventList[nextEventIndex], 1, &blockedEvent));
  check(zeEventHostReset(eventList[++nextEventIndex]));
  check(zeCommandListAppendBarrier(command_lists_callback[1], eventList[nextEventIndex], 0, nullptr));
  check(zeCommandListClose(command_lists_callback[1]));
  check(zeCommandQueueExecuteCommandLists(command_queue1, 1, &command_lists_callback[1], nullptr));

  check(zeCommandListCreate(ctxHandle, deviceHandle, &cmdListDesc, &command_lists_callback[2]));
  ze_event_handle_t waitEvent = eventList[++nextEventIndex];
  check(zeEventHostReset(waitEvent));
  check(zeCommandListAppendSignalEvent(command_lists_callback[2], waitEvent));
  check(zeEventHostReset(eventList[++nextEventIndex]));
  check(zeCommandListAppendBarrier(command_lists_callback[2], eventList[nextEventIndex], 0, nullptr));
  check(zeCommandListClose(command_lists_callback[2]));
  check(zeCommandQueueExecuteCommandLists(command_queue1, 1, &command_lists_callback[2], nullptr));

  // Created separate thread to signal blocked event
  std::thread th(callback, blockedEvent, waitEvent);
  
  //sync queue 2 which is empty hence sycn should finish immidiately but it waits for queue1 to finish the task
  ze_command_list_handle_t command_list[1];
  check(zeCommandListCreate(ctxHandle, deviceHandle, &cmdListDesc, &command_list[0]));
  check(zeEventHostReset(eventList[++nextEventIndex]));
  check(zeCommandListAppendBarrier(command_list[0], eventList[nextEventIndex], 0, nullptr));
  check(zeEventHostReset(eventList[++nextEventIndex]));
  check(zeCommandListAppendBarrier(command_list[0], eventList[nextEventIndex], 0, nullptr));
  check(zeCommandListClose(command_list[0]));
  check(zeCommandQueueExecuteCommandLists(command_queue2, 1, &command_list[0], nullptr));
  
  // Bug: Synchronization is called for queue2 but it is blocking host thread until task on queue1 is complete 
  //      If below line is commented out test make progress and ends as expected.
  check(zeCommandQueueSynchronize(command_queue2, std::numeric_limits<uint64_t>::max()));

  bool statusCheck = true;
  {
    ze_result_t status = zeEventQueryStatus(blockedEvent);
    if (status == ZE_RESULT_NOT_READY)
    {
      printf("Event is not signaled\n");
      statusCheck = false;
    } else {
      printf("Status is %d\n", status);
    }
  }

  // Wait until callback thread is completed
  th.join();

  if (statusCheck == false)
  {
    ze_result_t status = zeEventQueryStatus(blockedEvent);
    if (status == ZE_RESULT_NOT_READY)
    {
      printf("Event is not signaled\n");
    } else {
      printf("Status is %d\n", status);
    }
  }
  return 0;
}
```

zzdanowicz, 2022-10-18:
In your example, both command queues use the same ordinal/index pair. And they are connected to the same OS context. This works as expected as queues sharing the same OS context can be synchronized by the other queue. 

What I would suggest is to query the number of indices available for COMPUTE ordinal and then create queues on separate indexes - given HW provides them.

Sarbojit2019, 2022-10-19:
@zzdanowicz, thanks for suggestion, I will verify the behavior and update the ticket accordingly.

Sarbojit2019, 2022-10-27:
Closing it as suggestion given by @zzdanowicz solved my task. Thanks

closed 2022-10-27 by Sarbojit2019