intel/compute-runtime #576: dependency problems

hfp, 2022-10-24:
Version [22.42.24548](https://github.com/intel/compute-runtime/releases/tag/22.42.24548) is the first version which breaks installation under Debian 11. Previously, installing the provided `.deb` and `.ddeb` packages was just fine (presumably all built under Ubuntu). Now, `intel-igc-opencl` requires at least `libc6 v2.34` and `libstdc++6  v11`. I wonder if this is a necessity, or if IGC can be built such that the compute runtime continues serving the current Debian version. Another suggestion is to build the deb packages under Debian/stable in general (rather than Ubuntu). If libc 2.34 was introduced unexpectedly, it can be IMHO another reason to rely on Debian.

```text
dpkg: dependency problems prevent configuration of intel-igc-opencl:
 intel-igc-opencl depends on intel-igc-core (= 1.0.12260.1); however:
  Package intel-igc-core is not configured yet.
 intel-igc-opencl depends on libc6 (>= 2.34); however:
  Version of libc6:amd64 on system is 2.31-13+deb11u5.
 intel-igc-opencl depends on libstdc++6 (>= 11); however:
  Version of libstdc++6:amd64 on system is 10.2.1-6.
```

al42and, 2022-10-25:
Similar error on Ubuntu 20.04.5. The latest compatible version is 22.39.24347

eero-t, 2022-10-26:
Installation procedure tells for which Ubuntu version the binary packages are for.

If you want driver packages linked against older system libraries in older distribution versions, and working with kernels in those distributions, you can get them from the Intel package repository: https://dgpu-docs.intel.com/installation-guides/index.html

Note that you may also need a matching kernel GPU driver (DKMS) from the same repo (this is at least the case with media driver, I'm not sure about the compute one, as I'm not involved in compute driver development).

PS. Please change title to more specific, e.g. "Release packages for Ubuntu 22.04 depend on newer Glibc version than one in Ubuntu 20.04", or "Provide 20.04 driver packages in addition to ones for 22.04".

al42and, 2022-10-26:
> Installation procedure tells for which Ubuntu version the binary packages are for.

True, but the "installation procedure" was saying Ubuntu 22.04 since 22.25.23529: https://github.com/intel/compute-runtime/releases/tag/22.25.23529, yet the packages were working fine with Ubuntu 20.04 up until 22.40. So, the change is a bit surprising.

AdamCetnerowski, 2022-11-24:
There are no plans for proactive enabling and distro adoption except for Ubuntu. We are open to accepting contributions (pull requests) that unblock support.

closed 2022-11-24 by AdamCetnerowski