intel/compute-runtime #577: ocloc documentation + file naming bug

pvelesko, 2022-10-25:
I would like to use ocloc to generate a .cpp file

```
ocloc compile -file /home/pvelesko/reproducers/level0/time/KernelGPU.cl -device ats -cpp_file -output_no_suffix
```

This gives me 4 files:
```
-rw-r--r-- 1 pvelesko jlse   2172 Oct 25 06:19 KernelGPU.spv
-rw-r--r-- 1 pvelesko jlse   3172 Oct 25 06:19 KernelGPU.gen
-rw-r--r-- 1 pvelesko jlse  10437 Oct 25 06:19 KernelGPU.cpp
-rw-r--r-- 1 pvelesko jlse   5728 Oct 25 06:19 KernelGPU
```

1. When `-output_no_suffix` option is used, `KernelGPU` file was generated instead of `KernelGPU.bin` 

Inside `KernelGPU.cpp`:
```
#include <cstddef>
#include <cstdint>

size_t KernelGPUBinarySize_XE_HP_COREcore = 3172;
uint32_t KernelGPUBinary_XE_HP_COREcore[793] = {
    0x494e5443, 0x0000043b, 0x00000c05, 0x00000008, 0x00000001, 0x00000004, 0x00000000, 0x7f4c27d9, 
    0xc04e330e, 0x528a16b2, 0x00000010, 0x00000528, 0x00000640, 0x00000000, 0x00000020, 0x00000088, 
    0x000005b8, 0x43646461, 0x746e756f, 0x65766552, 0x00657372, 0x80030061, 0x7f054220, 0x00000000, 
    0x00000000, 0x80000065, 0x7f458220, 0x02000004, 0xffffffc0, 0x80039031, 0x02140000, 0xa7fa7f0c, 
    0x02100000, 0x80030061, 0x04050220, 0x00100004, 0x00000000, 0x80000966, 0x80018220, 0x01008000, 
    0x04c004c0, 0xe8000901, 0x00110001, 0xe8002001, 0x00110001, 0x80040970, 0x00018660, 0x350002a4, 
    0x00000000, 0x80240070, 0x00018660, 0x350002a4, 0x00000000, 0x80020072, 0x00040d50, 0x1d050005, 
    0x00050005, 0x80020972, 0x00840aa0, 0x1a020085, 0x00850085, 0x01050022, 0x0000c000, 0x000004c0, 
    0x000004c0, 0x22200041, 0x0311031e, 0xa2020049, 0x0311031a, 0x80000061, 0x02454560, 0x00000000, 
    0x00000000, 0x80001a52, 0x02240660, 0x06260104, 0x02040284, 0x80840970, 0x00018660, 0x35000224, 
    0xffffffff, 0xe2050061, 0x0011313c, 0x80001961, 0x30010220, 0x00000544, 0x00000000, 0x80240070, 
    0x00018660, 0x35000224, 0xffffffff, 0xe2050061, 0x0011303c, 0xe2030061, 0x001130a4, 0x80000061, 
    0x30014220, 0x00000000, 0x00000000, 0x00050070, 0x00010110, 0x11000004, 0x00000004, 0x8c001c61, 
    0x31010220, 0x00000544, 0x00000000, 0x80001c61, 0x30010220, 0x000003c4, 0x00000000, 0x80020072, 
    0x00040d50, 0x1d050005, 0x00050005, 0x80020972, 0x00840aa0, 0x1a020085, 0x00850085, 0x01850022, 
    0x0000c000, 0x00000290, 0x00000290, 0xe2030061, 0x001131a4, 0x80001161, 0x31014220, 0x00000000, 
    0x00000000, 0x00850070, 0x00010110, 0x11000004, 0x00000004, 0x8c800061, 0x02650660, 0x00000224, 
    0x00000000, 0x80001c61, 0x31010220, 0x000003c4, 0x00000000, 0xe2030061, 0x001130a4, 0x80000061, 
    0x30014220, 0x00000000, 0x00000000, 0x00050070, 0x00010110, 0x11000004, 0x00000004, 0x8c000061, 
    0x30014220, 0x00000000, 0xffffffff, 0x80000061, 0x02e50220, 0x00000564, 0x00000000, 0x81001f69, 
    0x06058660, 0x05000264, 0x00020002, 0x81000040, 0x02c50660, 0x06000264, 0x02000204, 0x80020072, 
    0x00040d50, 0x1d050005, 0x00050005, 0x80020972, 0x00840aa0, 0x1a020085, 0x00850085, 0x81004131, 
    0x050c0000, 0xa000060c, 0x00420000, 0xe8002101, 0x00110001, 0x81001161, 0x31010220, 0x00000564, 
    0x00000000, 0x80840070, 0x00018660, 0x350002c4, 0xffffffff, 0xe2030061, 0x0011308c, 0x81000061, 
    0x08050aa0, 0x00000604, 0x00000000, 0x81000061, 0x05650220, 0x00003100, 0x00000000, 0x91000061, 
    0x05650220, 0x000002e4, 0x00000000, 0x81000961, 0x31010220, 0x00000564, 0x00000000, 0x80a40070, 
    0x00018660, 0x350002c4, 0xffffffff, 0x80000061, 0x02e50220, 0x00000564, 0x00000000, 0x81000061, 
    0x05650220, 0x00003100, 0x00000000, 0x91001a61, 0x05650220, 0x000002e4, 0x00000000, 0x81000040, 
    0x07050660, 0x06000504, 0x000002a4, 0x81009231, 0x00000000, 0xa002080c, 0x00c2070c, 0x80003261, 
    0x30010220, 0x000003c4, 0x00000000, 0xe2030061, 0x001131a4, 0x80000061, 0x31010220, 0x000003e4, 
    0x00000000, 0x81800061, 0x30010220, 0x00000564, 0x00000000, 0x80001b61, 0x31010220, 0x000003c4, 
    0x00000000, 0x80020072, 0x00040d50, 0x1d050005, 0x00050005, 0x80020972, 0x00840aa0, 0x1a020085, 
    0x00850085, 0x11050028, 0x0000c000, 0x00000058, 0x00000058, 0xe2030061, 0x001130a4, 0x80001161, 
    0x30014220, 0x00000000, 0x00000000, 0x00050070, 0x00010110, 0x11000004, 0x00000004, 0x8c000061, 
    0x02650660, 0x000002c4, 0x00000000, 0x80001c61, 0x30010220, 0x000003c4, 0x00000000, 0x00050027, 
    0x00004000, 0x00000000, 0xfffffdd8, 0x00050025, 0x00004000, 0x00000000, 0x00000010, 0xe2030061, 
    0x001131a4, 0x80000061, 0x31014220, 0x00000000, 0x00000000, 0x00850070, 0x00010110, 0x11000004, 
    0x00000004, 0x8c800061, 0x31014220, 0x00000000, 0xffffffff, 0x80000061, 0x02e50220, 0x00000524, 
    0x00000000, 0x81800040, 0x02458660, 0x05000244, 0x00010001, 0x81800061, 0x30010220, 0x00000524, 
    0x00000000, 0x80041a70, 0x00010660, 0x56000244, 0x000002a4, 0x81800061, 0x05250220, 0x00003000, 
    0x00000000, 0x91800061, 0x05250220, 0x000002e4, 0x00000000, 0x81801961, 0x30010220, 0x00000524, 
    0x00000000, 0x80240070, 0x00010660, 0x56000244, 0x000002a4, 0x80000061, 0x02e50220, 0x00000524, 
    0x00000000, 0x81800061, 0x05250220, 0x00003000, 0x00000000, 0x91801a61, 0x05250220, 0x000002e4, 
    0x00000000, 0x81801961, 0x30010220, 0x00000524, 0x00000000, 0x80000061, 0x31010220, 0x000003c4, 
    0x00000000, 0x80020072, 0x00040d50, 0x1d050005, 0x00050005, 0x80020972, 0x00840aa0, 0x1a020085, 
    0x00850085, 0x01050027, 0x00004000, 0x00000000, 0xfffffbd0, 0x00050025, 0x00004000, 0x00000000, 
    0x00000010, 0x647f0061, 0x00100400, 0x80034331, 0x090c0000, 0xa000040c, 0x02780000, 0x80032361, 
[... 41 lines cut ...]
    0x00000000, 0x00000000, 0x00000000, 0x00000008, 0x00000000, 0x00000000, 0x00000000, 0x00000000, 
    0x00010000, 0x00000000, 0x00000000, 0x0000001b, 0x00000010, 0x00000004, 0x00000000, 0x0000001a, 
    0x00000048, 0x00000000, 0x0000000c, 0x00000008, 0x00000004, 0x00000008, 0x00000008, 0x6c675f5f, 
    0x6c61626f, 0x00000000, 0x454e4f4e, 0x00000000, 0x00645f41, 0x2a746e69, 0x0000383b, 0x454e4f4e, 
    0x00000000, 0x0000001a, 0x00000048, 0x00000001, 0x0000000c, 0x00000008, 0x00000004, 0x00000008, 
    0x00000008, 0x6c675f5f, 0x6c61626f, 0x00000000, 0x454e4f4e, 0x00000000, 0x00645f43, 0x2a746e69, 
    0x0000383b, 0x454e4f4e, 0x00000000, 0x0000001a, 0x0000004c, 0x00000002, 0x0000000c, 0x00000008, 
    0x00000008, 0x00000008, 0x00000008, 0x72705f5f, 0x74617669, 0x00000065, 0x454e4f4e, 0x00000000, 
    0x454c454e, 0x0000004d, 0x3b746e69, 0x00000034, 0x454e4f4e, 0x00000000, 0x0000001a, 0x0000004c, 
    0x00000003, 0x0000000c, 0x00000008, 0x00000008, 0x00000008, 0x00000008, 0x72705f5f, 0x74617669, 
    0x00000065, 0x454e4f4e, 0x00000000, 0x6e756f63, 0x00000074, 0x3b746e69, 0x00000034, 0x454e4f4e, 
    0x00000000};

#include "shared/source/built_ins/registry/built_ins_registry.h"

namespace NEO {
static RegisterEmbeddedResource registerKernelGPUBin(
    "XE_HP_COREcore_0_KernelGPU.builtin_kernel.bin",
    (const char *)KernelGPUBinary_XE_HP_COREcore,
    KernelGPUBinarySize_XE_HP_COREcore);
}
```

2. What is the purpose of the last part? On my system `#include "shared/source/built_ins/registry/built_ins_registry.h"` is not found. 
3. Not clear what `registerKernelGPUBin` does

HoppeMateusz, 2022-10-25:
Hello,

the extra code is leftover of logic when NEO used -cpp_file option to generate builtin kernels cpp files. It is not needed anymore and will be removed soon

thanks for reporting this issue.

pvelesko, 2022-10-25:
@HoppeMateusz are there any examples of how to integrate these cpp kernels into your l0 app?

HoppeMateusz, 2022-10-25:
i am not aware of examples, but you could use the binary like that:

`    ze_module_desc_t moduleDesc = {ZE_STRUCTURE_TYPE_MODULE_DESC};
    ze_module_build_log_handle_t buildlog;
    moduleDesc.format = ZE_MODULE_FORMAT_NATIVE;
    moduleDesc.pInputModule = (uint8_t*)KernelGPUBinary_XE_HP_COREcore;
    moduleDesc.inputSize = KernelGPUBinarySize_XE_HP_COREcore;
    moduleDesc.pBuildFlags = "";

    zeModuleCreate(context, device, &moduleDesc, &module, &buildlog);
`

last part ( points 2. & 3. ) needs to be removed to use the file in L0 application.

HoppeMateusz, 2022-10-26:
@pvelesko  - the change removing unneeded code  has been merged: 8980b2b817085a2b6cb6b3fa54d82f1442eb8c7b

can we close this issue ?

pvelesko, 2022-10-26:
@HoppeMateusz What about 

> When -output_no_suffix option is used, KernelGPU file was generated instead of KernelGPU.bin

HoppeMateusz, 2022-10-27:
can you point what is wrong or what would be expected filename with this  "-output_no_suffix" option?
is it missing  ".bin " ?

from first look - this is removing any suffixes added to the name.

pvelesko, 2022-10-27:
I would like to use ocloc to generate a .cpp file

ocloc compile -file /home/pvelesko/reproducers/level0/time/KernelGPU.cl -device ats -cpp_file -output_no_suffix

This gives me 4 files:

-rw-r--r-- 1 pvelesko jlse   2172 Oct 25 06:19 KernelGPU.spv
-rw-r--r-- 1 pvelesko jlse   3172 Oct 25 06:19 KernelGPU.gen
-rw-r--r-- 1 pvelesko jlse  10437 Oct 25 06:19 KernelGPU.cpp
-rw-r--r-- 1 pvelesko jlse   5728 Oct 25 06:19 KernelGPU

without this option it generates

-rw-r--r-- 1 pvelesko jlse   2172 Oct 25 06:19 KernelGPU_SomeSuffix.spv
-rw-r--r-- 1 pvelesko jlse   3172 Oct 25 06:19 KernelGPU_SomeSuffix.gen
-rw-r--r-- 1 pvelesko jlse  10437 Oct 25 06:19 KernelGPU_SomeSuffix.cpp
-rw-r--r-- 1 pvelesko jlse   5728 Oct 25 06:19 KernelGPU_SomeSuffix.bin

closed 2022-10-28 by pvelesko