intel/compute-runtime #579: Problem getting OpenCL to work on Haswell-ULT & Arc A750 - error 1001

WoolyBall, 2022-11-03:
Dear developers, 

I am faced with a bit bizarre problem getting my Intel Arc A750 ( and as well the Intel iGPU) to work in my Linux system. First of all lspci telling me that both adapters are there and also seem to be working

```
lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 09)
04:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A750] (rev 08)
```  

Also dmesg not throwing me reds about i915 driver - there are some messages related to that it might be better to activate resizeable bar and one error thats more like a warning about capture buffer may be not large enough ... but thats mostly it. Only other thing I see there are some complains about "Cannot find any crtc or sizes", but I consider that to be normal, because the computer is running headless.

```
[    6.388434] i915 0000:04:00.0: [drm] Failed to resize BAR2 to 8192M (-ENOSPC)
[    6.388439] i915 0000:04:00.0: BAR 2: assigned [mem 0xbe0000000-0xbefffffff 64bit pref]
[    6.388482] i915 0000:04:00.0: [drm] Local memory IO size: 0x0000000010000000
[    6.388485] i915 0000:04:00.0: [drm] Local memory available: 0x00000001fc000000
[    6.388487] i915 0000:04:00.0: [drm] Using a reduced BAR size of 256MiB. Consider enabling 'Resizable BAR' or similar, if available in the BIOS.
[    6.441147] i915 0000:04:00.0: [drm] Finished loading DMC firmware i915/dg2_dmc_ver2_07.bin (v2.7)
[    6.458550] i915 0000:04:00.0: [drm] GuC error state capture buffer maybe too small: 2097152 < 3737592 (min = 1245864)
[    6.467819] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    6.491428] i915 0000:04:00.0: [drm] Can't load HuC due to missing MEI modules
[    6.491435] i915 0000:04:00.0: [drm] HuC init failed with -5
[    6.492094] i915 0000:04:00.0: [drm] GuC firmware i915/dg2_guc_70.bin version 70.5.1
[    6.492098] i915 0000:04:00.0: [drm] HuC firmware i915/dg2_huc_gsc.bin version 7.10.3
[    6.532759] i915 0000:04:00.0: [drm] GuC submission enabled
[    6.532764] i915 0000:04:00.0: [drm] GuC SLPC enabled
[    6.533319] i915 0000:04:00.0: [drm] GuC RC: enabled
[    6.558089] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
[    6.612701] [drm] Initialized i915 1.6.0 20201103 for 0000:04:00.0 on minor 1
```

Generally the GPUs seem both to be detected by the system, at least I get
```
ls /dev/dri/by-path/ 
pci-0000:00:02.0-card  pci-0000:00:02.0-render  pci-0000:04:00.0-card  pci-0000:04:00.0-render
```

So far, so good - I hope at least. Now I made sure that I am in the video group, but to get sure I also did run "clinfo" with sudo to get sure I am not struggling with missing rights.

Running clinfo I am just left with
```
Number of platforms:				 1
  Platform Profile:				 FULL_PROFILE
  Platform Version:				 OpenCL 2.1 AMD-APP (3452.0)
  Platform Name:				 AMD Accelerated Parallel Processing
  Platform Vendor:				 Advanced Micro Devices, Inc.
  Platform Extensions:				 cl_khr_icd cl_amd_event_callback 

  Platform Name:				 AMD Accelerated Parallel Processing
Number of devices:				 0
```

So, we clearly see, that the Intel OpenCL platform does not show up - but an empty AMD platform. I need to note at this point that I use the system image normally to test multiple GPUs. Therefore in the image also drivers for Nvidia and AMD GPUs are present. Yet there are no GPUs of this type installed at the moment. 

Removing the AMD and the Nvidia OpenCL icd files from /etc/OpenCL/vendors and leaving only the Intel ICD there I am getting the error
```
ERROR: clGetPlatformIDs(-1001)
```

At first I wondered, if the Intel OpenCL driver is loaded at all, but running the same clinfo with "export LD_DEBUG=libs" telling me: yes, the right files are touched!

```
      4891:	initialize program: clinfo
      4891:	
      4891:	
      4891:	transferring control: clinfo
      4891:	
      4891:	find library=libigdrcl.so [0]; searching
      4891:	 search cache=/etc/ld.so.cache
      4891:	  trying file=/usr/local/lib/libigdrcl.so
      4891:	
      4891:	find library=libigdgmm.so.12 [0]; searching
      4891:	 search cache=/etc/ld.so.cache
      4891:	  trying file=/usr/local/lib/libigdgmm.so.12
      4891:	
      4891:	
      4891:	calling init: /usr/local/lib/libigdgmm.so.12
      4891:	
      4891:	
      4891:	calling init: /usr/local/lib/ibigdrcl.so
      4891:	
      4891:	
      4891:	calling fini: /usr/local/lib/libigdrcl.so [0]
      4891:	
      4891:	
      4891:	calling fini: /usr/local/lib/ibigdgmm.so.12 [0]
      4891:	
ERROR: clGetPlatformIDs(-1001)
```

So the question now is: what might cause this and how to fix this? The installed driver version is "22.43.24558".  
Note that carrying the exact same operation system image (!!!) installed on an USB stick to an other system equipped with the following GPUs  
```
lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation CoffeeLake-S GT2 [UHD Graphics 630]
07:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A380] (rev 05)
```
is working without any issues - here both the iGPU and the Arc A380 are displayed in clinfo without struggles and also can be used for compute operations without problems.

JablonskiMateusz, 2022-11-03:
Hi @WoolyBall  could you share logs from strace? 
You may get them using `strace -o strace.log clinfo`

WoolyBall, 2022-11-03:
Hey,

I need to report I narrowed things down to the Haswell iGPU. As soon as this is not present (e.g. disabling the iGPU in Bios, putting the A750 into an other system with no iGPU, newer Intel iGPU or iGPU of an other vendor), then it is working just fine with the same system on USB stick and same Arc A750.

So, the bug or in this case rather feature request is: allow this driver to skip over older Intel iGPUs that are not supported by the compute driver, so the newer discrete GPUs can still use it.

Kind regards.

JablonskiMateusz, 2022-11-03:
Our driver should skip older, not supported GPU and expose only supported devices. Therefore having more logs would be very helpful.

eero-t, 2022-11-21:
> I need to report I narrowed things down to the Haswell iGPU. As soon as this is not present (e.g. disabling the iGPU in Bios, putting the A750 into an other system with no iGPU, newer Intel iGPU or [...]

@WoolyBall What is the other system, did it use the same i915 kernel driver version, and did that also show dmesg warnings about resizable bar?

@JablonskiMateusz When I did some testing for Arc/DG2 on TGL month or two ago, Mesa 3D driver did not work with the GPU until resizable bar option from BIOS was set.  Before setting that, kernel reported only 256MB, instead of the full amount, so it would have been rather useless.

Is compute driver supposed to work (enough for clinfo) when resizable bar is NOT set?  And if yes, does it require some particular (new enough) kernel version for that?

JablonskiMateusz, 2022-11-21:
@eero-t basic app like clinfo doesn't submit anything to GPU, hence it should work fine as long as basic i915 ioctls return valid values, I believe that resizable bar option doesn't matter in this case

eero-t, 2022-11-21:
@JablonskiMateusz Would 1/4GB be enough to match OpenCL minimum requirements?

It looks like `clinfo` would be doing at least some GPU memory related operations:
```
strace -f clinfo 2>&1 | grep I915 | sed  -e 's/^.*ioctl([0-9]*, *//' -e 's/[ ,].*$//' | sort -u
DRM_IOCTL_I915_GEM_CONTEXT_CREATE_EXT
DRM_IOCTL_I915_GEM_CONTEXT_DESTROY
DRM_IOCTL_I915_GEM_CONTEXT_GETPARAM
DRM_IOCTL_I915_GEM_CONTEXT_SETPARAM
DRM_IOCTL_I915_GEM_CREATE_EXT
DRM_IOCTL_I915_GEM_MMAP_OFFSET
DRM_IOCTL_I915_GEM_USERPTR
DRM_IOCTL_I915_GEM_VM_CREATE
DRM_IOCTL_I915_GEM_VM_DESTROY
DRM_IOCTL_I915_GEM_WAIT
DRM_IOCTL_I915_GETPARAM
DRM_IOCTL_I915_QUERY
DRM_IOCTL_I915_REG_READ
```

@WoolyBall What's the strace output of your failing clinfo: `strace -f -e openat,ioctl clinfo 2>&1` ?

WoolyBall, 2022-11-22:
> @WoolyBall What is the other system, did it use the same i915 kernel driver version, and did that also show dmesg warnings about resizable bar?

Both systems have resizeable bar disabled and yes, the dmesg output is nearly the same on both systems. In fact both are mining rigs running the same image of mmpOS - based on Ubuntu kernels, but not running the precompiled drivers. 
On the older systems (I have some with Ivy Bridge and CoffeeLake) the card and iGPU get detected as render devices, but clinfo does not find the Intel platform. On the other systems with Kaby Lake or newer its working fine. As said, all systems use 1:1 identical os images. 

Here in the log you can find the strace output of both - I did put the A750 in the working system and for instance the A380 in the not working one.   

[strace_working.log](https://github.com/intel/compute-runtime/files/10069152/strace_working.log)
[strace_not_working.log](https://github.com/intel/compute-runtime/files/10069157/strace_not_working.log)

Do not mind the reports of Nvidia and AMD driver in there. As said, these are mining machines that have quite a GPU zoo in them. But the issue appears 100% same when removing Nvidia and AMD icd or when plugging of the other vendor cards of the system - it does not change the outcome.

eero-t, 2022-11-23:
Your strace shows clinfo using drivers for NVdia, AMD and Intel present:
* In non-working case there being 6 GPU devices of which 2 are Intel (card0/card1/renderD128/renderD129), and rest of Nvidia
* In working case there are 7 GPU devices of which 2 are Intel (card3/card6/renderD131/renderD134), 2 Nvidia and 3 AMD

It would have helped if you had disabled the other drivers (e.g. by temporarily removing their ICD files) before doing strace, as now it's really hard to find useful information among things all these drivers do.

In both cases neither "clinfo" nor OpenCL frontend seems to be from the distro, but maybe from Nvidia (based on `/opt/mmp/lib/cuda/libOpenCL.so.1` & `/opt/mmp/bin/clinfo` paths).

`i915` ioctls are done only in the working version, none are done in the non-working one.  I.e. non-working version does not even try to use the GPU.

So far the main difference  I've noticed in the straces, is this:
```
$ grep EACCES  strace_not_working.log 
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:08:00.0-render", O_RDWR) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:0e:00.0-render", O_RDWR) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:07:00.0-render", O_RDWR) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:02:00.0-render", O_RDWR) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:00:02.0-render", O_RDWR) = -1 EACCES (Permission denied)
openat(AT_FDCWD, "/dev/dri/by-path/pci-0000:0c:00.0-render", O_RDWR) = -1 EACCES (Permission denied)

$ grep EACCES  strace_working.log 
openat(AT_FDCWD, "/dev/shm/sem.hsakmt_semaphore", O_RDWR|O_NOFOLLOW) = -1 EACCES (Permission denied)
```

Does the user you're running as, have access rights to the GPU device files?

WoolyBall, 2022-11-23:
When removing the icds of Nvidia and Amd the outcome on the non working
systems is the above mentioned error -1001, while when they are in at least
clinfo output makes sense without finding the Intel platform.

This behavior is the same - both if I run clinfo as a user (checked it is
in the video group) as well as sudo or root. Outcome is the same here.
Therefore I assume the access rights are not making the difference here.

JablonskiMateusz, 2024-03-08:
@WoolyBall could you verify if the issue is visible on most recent release?

It should be fixed in f63dd1f4f2648f8becdbc0fbdf455379b5810ee5

closed 2024-09-20 by JablonskiMateusz