intel/compute-runtime #582: `ocloc disasm` doesn't accept `-device` argument

mattst88, 2022-11-14:
````sh
$ LD_LIBRARY_PATH=/usr/lib/llvm/14/lib64 ocloc compile -file icd-neo.txt-split/0x28308b77-1.cl -device dg2-g12 -output 0x28308b77-1.cl 
Build succeeded.
$ ocloc disasm -file 0x28308b77-1.cl_XE_HPG_COREdg2.bin -device dg2-g12 -dump dump               
Warning : missing or invalid -device parameter - results may be inacurate
Path to patch list not provided - using defaults, skipping patchokens as undefined.
Trying to disassemble main_function.krn
````

Why does `ocloc` print `Warning : missing or invalid -device parameter - results may be inacurate` despite the fact that I passed `-device dg2-g12` (which `ocloc compile` accepted!)?

mattst88, 2022-11-14:
Maybe it just doesn't accept `dg2-g12`? `tgllp` seems to work fine:

````sh
$ ocloc compile -file icd-neo.txt-split/0x28308b77-1.cl -device tgllp -output 0x28308b77-1.cl 
Build succeeded.
$ ocloc disasm -file 0x28308b77-1.cl_Gen12LPlp.bin -device tgllp -dump dump                  
Path to patch list not provided - using defaults, skipping patchokens as undefined.
Trying to disassemble main_function.krn
````

mattst88, 2022-11-23:
Any traction on this?

daariahinz, 2022-11-24:
Yes, fix has been added in https://github.com/intel/compute-runtime/commit/31deb4fd63c101080ce11dd0f58274156b45c969

JablonskiMateusz, 2022-11-28:
@mattst88 did it resolve your issue?

mattst88, 2022-12-02:
Sorry, I  haven't been able to test it yet. The patch doesn't apply cleanly to the latest tag.

mattst88, 2023-02-27:
I confirm this is fixed (tested 22.53.25242.13). Thanks!

closed 2023-02-27 by mattst88