intel/compute-runtime #587: aborting on os_interface/linux/drm_neo.cpp

Leoriem-code, 2022-12-02:
Hello, I am unable to use anything related to OpenCL because when I try to, I receive the following error:
```bash
$ clinfo
Abort was called at 347 line in file:
/var/tmp/portage/dev-libs/intel-compute-runtime-22.43.24558/work/compute-runtime-22.43.24558/shared/source/os_interface/linux/drm_neo.cpp
zsh: IOT instruction  clinfo
```

I did read #566 (I have a AlderLake CPU too) and followed [those](https://www.intel.com/content/www/us/en/developer/articles/technical/software-security-guidance/secure-coding/loading-microcode-os.html) instructions to update the microcode, but nothing changed.

If it can help, you can find the output of `strace clinfo` [here](https://pastebin.com/FiRuSFPh), and the one of `lscpu` [there](https://pastebin.com/0GXzwezX).

alalek, 2022-12-02:
`$ LD_DEBUG=libs clinfo` may help too.

---

Does it work under the `root` user?

Leoriem-code, 2022-12-02:
No it does not work with root, and [here](https://pastebin.com/uvgg9f0v) for `$ LD_DEBUG=libs clinfo`

alalek, 2022-12-02:
> /opt/intel/opencl-18.1.0.015/lib64/

and

> /usr/lib64/libigdgmm.so.12
> /usr/lib64/intel-opencl/libigdrcl.so

looks strange (directories should not mess - there are 3 variants probably).

Where is "/opt/intel/opencl-18.1.0.015" from?

How many runtimes are installed on your system?

Query:
```
$ grep -Rn '' /etc/OpenCL/vendors/
```

Also query the kernel version:

```
$ uname -a
```

---

Try this:

```
$ LD_PRELOAD_PATH=/opt/intel/opencl-18.1.0.015/lib64/ LD_DEBUG=libs clinfo
```

or vise-versa disable /opt (under `root` user)

```
# mv /opt/intel/opencl-18.1.0.015{,-disabled}
$ LD_DEBUG=libs clinfo
```

---

To avoid mess of the different packages on your system you could try to run from "docker" (if you have its installed):

```
docker run --it --device /dev/dri:/dev/dri --privileged --name test_opencl <your_distribution>
# ... install OpenCL runtime with clinfo from scratch ...
# clinfo
```

Leoriem-code, 2022-12-03:
Thanks for your responses, I recently upgraded my kernel version to `6.0.10` (when updating the microcode) but where was the problem from 6.10.6 and onwards.

`/opt/intel/opencl-18.1.0.015/lib64/` come from [`dev-util/intel-ocl-sdk-18.1.0.015`](http://packages.gentoo.org/packages/dev-util/intel-ocl-sdk), `/usr/lib64/libigdgmm.so.12` from [`media-libs/gmmlib-22.3.1`](https://packages.gentoo.org/packages/media-libs/gmmlib) and `/usr/lib64/intel-opencl/libigdrcl.so` from [`dev-libs/intel-compute-runtime-22.43.24558`](https://packages.gentoo.org/packages/dev-libs/intel-compute-runtime).

There are two runtimes installed:
```bash
$ grep -Rn '' /etc/OpenCL/vendors/
/etc/OpenCL/vendors/intel64.icd:1:/opt/intel/opencl-18.1.0.015/lib64/libintelocl.so
/etc/OpenCL/vendors/intel.icd:1:/usr/lib64/intel-opencl/libigdrcl.so
```

Finally, [`$ LD_PRELOAD_PATH=/opt/intel/opencl-18.1.0.015/lib64/ LD_DEBUG=libs clinfo`](https://pastebin.com/AwbCec0f) and [`$ LD_DEBUG=libs clinfo`](https://pastebin.com/v5K2AP9N)

Reading all those errors message, it seems the problem comes from `libigfxdbgxchg64`. Nothing is found when I use `# find / | rg -i libigfxdbgxchg64`, but I can't seem to find how to install it.

alalek, 2022-12-03:
> libigfxdbgxchg64

It should not be a problem - I don't have this file on my desktop with Alder Lake (and I see these `libs` messages with working `clinfo` on Ubuntu 22.04 or Fedora 36).

Leoriem-code, 2022-12-03:
Okay, so. Firstly, thank you @alalek, I managed to solve the problem.  
It came from the fact that I did not have `/lib/firmware/i915/adlp_guc_70.1.1.bin` built in the kernel, it caused the kernel to be unable to load the GuC firmware, which caused errors.  
Now I have another problem on my hands, but it is no longer something caused by intel-compute-runtime, so I will close this issue.

PS:  
I found the cause because when uninstalling `intel-compute-runtime` and using `mesa` instead, the error messages were differents. I just followed the trail, came back on the Gentoo wiki, and finally found the issue

closed 2022-12-03 by Leoriem-code