intel/compute-runtime #588: Issues with ubuntu 22.04 build

fcharras, 2022-12-06:
I'm trying to run SYCL-based software on gpu (using [dpctl](https://github.com/IntelPython/dpctl/) ) and it requires the compute runtime to be installed to detect gpu devices. 

The issue I have is that [using ubuntu build](https://packages.ubuntu.com/jammy/amd64/intel-opencl-icd/download) the gpu devices are not detected. The same version [downloaded from github](https://github.com/intel/compute-runtime/releases/tag/22.14.22890) works. But for end users it's much easier to use the build available in the official repo.

What differences in those two builds could explain this ? where would be the best place to report this to ?

JablonskiMateusz, 2022-12-15:
Hi @fcharras 
Please share more details about GPU you are using.

JablonskiMateusz, 2022-12-19:
The only difference that I assume may change anything is additional cmake flag used for builds on our github:
> packages were built with custom flag NEO_ENABLE_i915_PRELIM_DETECTION=1

but it shouldn't be the case for TGL.

Could you also info about kernel version and log from strace when using driver from ubuntu repository?

fcharras, 2022-12-19:
TY for looking at the issue, will provide the strace ASAP.

One thing I forgot to mention that could be useful for reproducibility is that I use a docker container from the latest, official ubuntu jammy image, created with the flag `--device=/dev/dri` for gpu passthrough. I didn't test directly on a ubuntu host. (my distro is an up to date arch linux-based endeavor os).

I've also provided quick reproduction steps in https://github.com/IntelPython/dpctl/issues/1010 and a more thorough installation guide from `.deb` packages provided here in [this README](https://github.com/soda-inria/sklearn-numba-dpex/blob/main/README.md#install-low-level-runtime-libraries-for-your-gpu-12).

fcharras, 2022-12-19:
Logs from strace (not familiar with strace, I used [ubuntu guide](https://wiki.ubuntu.com/Strace):

- [strace-dpctl_ubuntu_intel_runtime.log](https://github.com/intel/compute-runtime/files/10259875/strace-dpctl_ubuntu_intel_runtime.log) with ubuntu runtime (gpu is not detected)

- [strace-dpctl_upstream__intel_runtime.log](https://github.com/intel/compute-runtime/files/10259876/strace-dpctl_upstream__intel_runtime.log) with upstream runtime (gpu is detected)

Kernel version:

```
(my-dpex-env) root@cdd9d852219e:~# uname -srm
Linux 6.0.12-arch1-1 x86_64
```

JablonskiMateusz, 2022-12-21:
I see there are different dependencies loaded in both cases:
`[pid  5110] 14:12:38.847711 openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libigdgmm.so.12", O_RDONLY|O_CLOEXEC) = 6` for ubuntu runtime
`[pid  5186] 14:14:18.697559 openat(AT_FDCWD, "/usr/local/lib/libigdgmm.so.12", O_RDONLY|O_CLOEXEC) = 6` for github runtime

please verify your workspace where are they coming from

fcharras, 2023-01-07:
Can you reproduce the issue on your side ? given I run the instructions from a bare ubuntu image it should be easily reproducible.

From `dpkg`:

- ubuntu runtime:

```
root@a3265eaeac03:/# dpkg -S libigdgmm 
libigdgmm12:amd64: /usr/lib/x86_64-linux-gnu/libigdgmm.so.12
libigdgmm12:amd64: /usr/lib/x86_64-linux-gnu/libigdgmm.so.12.1.0
```

it's the [official build](https://packages.ubuntu.com/jammy/libigdgmm12)

- github runtime:

```
root@c65ca8d5e738:/# dpkg -S libigdgmm 
intel-gmmlib: /usr/local/lib/libigdgmm.so.12
intel-gmmlib: /usr/local/lib/libigdgmm.so.12.0.1423
```

it comes from the [set of deb packages that is provided in the install instructions in this repository](https://github.com/intel/compute-runtime/releases/tag/22.14.22890).

Note that the versions distributed by ubuntu and on github are the same (22.14.22890) so that the differences can only be in the build or install process.

But I don't think the issues comes from the dependencies. https://github.com/IntelPython/dpctl/issues/1010 which provides the quickest path  to reproduce the device detection issue (with a hack) shows that the root cause comes from the package `intel-opencl-icd`:

- ubuntu packages fail at detecting the device

- uninstalling **only** `intel-opencl-icd` and replacing only this package with the corresponding github build enables device detection (although mixing dependencies from ubuntu and github in such a way results in a broken dependency tree and a broken runtime overall in some other aspects)

JablonskiMateusz, 2023-01-24:
There is an issue with compiler front-end library. From what I see package from github requires opencl-clang 11
```
$ ldd /usr/local/lib/libigdfcl.so.1
        linux-vdso.so.1 (0x00007ffd918c8000)
        libopencl-clang.so.11 => /usr/local/lib/libopencl-clang.so.11 (0x00007f0966e00000)

```

while package from ubuntu repository requires opencl-clang 10
```
$ ldd /lib/x86_64-linux-gnu/libigdfcl.so.1
        linux-vdso.so.1 (0x00007ffd8d7e8000)
        libopencl-clang.so.10 => not found
```

Looks like build config issue on distro side.

closed 2023-01-24 by JablonskiMateusz

fcharras, 2023-01-30:
Looking back at it, I think my misfortune mostly comes from having read [this section of the README](https://github.com/intel/compute-runtime#via-system-package-manager) before any other documentation regarding the runtime, this repository being the only suggested source by the [intel/llvm](https://github.com/intel/llvm/blob/sycl/sycl/doc/GetStartedGuide.md) github repository. If distributing a fully working packages on ubuntu is not the priority her, I'd suggest at least editing the README, and replacing this section with [this official, up to date guide from intel that recommends using PPAs](https://dgpu-docs.intel.com/installation-guides/index.html).

Frankly `intel-opencl-icd` package isn't a dependency of any other package in ubuntu repositories so I think it would be even be better removed from there (so that users have no choice but finding out the best source) than distributed in this state.