intel/compute-runtime #593: Build error

linsson, 2022-12-10:
I am trying to compile compute-runtime (I tried both 22.43.24595.30 and 22.43.24558 version) but I always reveive an error  at "Generating ../../../../bin/built_ins/x64/spirv/fill_image3d.spv"

Here is the complete log:

-- The C compiler identification is GNU 8.5.0
-- The CXX compiler identification is GNU 8.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/local/gcc/8.5.0/gcc/bin/gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/local/gcc/8.5.0/gcc/bin/g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- branch dir list: /
-- Host processor: x86_64
-- Target processor: x86_64
-- Neo target processor: x86_64
-- WDK include paths: /usr/local/software/compute-runtime-22.43.24558/third_party/DirectX-Headers/include/wsl;/usr/local/software/compute-runtime-22.43.24558/third_party/DirectX-Headers/include/wsl/stubs;/usr/local/software/compute-runtime-22.43.24558/third_party/DirectX-Headers/include/directx;/usr/local/software/compute-runtime-22.43.24558/third_party/DirectX-Headers/include/dxguids;/usr/local/software/compute-runtime-22.43.24558/third_party/libdxg/include/
-- Driver model : drm_or_wddm
-- Performing Test COMPILER_SUPPORTS_LLD
-- Performing Test COMPILER_SUPPORTS_LLD - Failed
-- Performing Test COMPILER_SUPPORTS_GOLD
-- Performing Test COMPILER_SUPPORTS_GOLD - Success
-- Using linker: ld.gold
-- Release build configuration
-- Source Level Debugger headers dir: /usr/local/software/compute-runtime-22.43.24558/third_party/source_level_debugger
-- Aub Stream Headers dir: /usr/local/software/compute-runtime-22.43.24558/third_party/aub_stream/headers
-- Khronos OpenCL headers dir: /usr/local/software/compute-runtime-22.43.24558/third_party/opencl_headers
-- Khronos OpenGL headers dir: /usr/local/software/compute-runtime-22.43.24558/third_party/opengl_headers
-- AOT config headers dir: /usr/local/software/compute-runtime-22.43.24558/third_party/aot_config_headers
-- Third party dir: /usr/local/software/third_party
-- Found PkgConfig: /usr/local/gcc/8.5.0/pkg-config/0.29.2/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'igc-opencl'
--   Found igc-opencl, version 1.0.1
-- IGC include dirs: /usr/local/gcc/8.5.0/igc/1.0.12260.1/include/igc;/usr/local/gcc/8.5.0/igc/1.0.12260.1/include/igc/cif;/usr/local/gcc/8.5.0/igc/1.0.12260.1/include/igc/ocl_igc_shared/executable_format;/usr/local/gcc/8.5.0/igc/1.0.12260.1/include/igc/ocl_igc_shared/device_enqueue
-- VISA Dir: /usr/local/software/igc/visa
-- IGA Includes dir: /usr/local/software/igc/visa/iga/IGALibrary/api
-- Checking for module 'igdgmm'
--   Found igdgmm, version 12.2.0
-- GmmLib include dirs: /usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm;/usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm/GmmLib;/usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm/GmmLib/inc;/usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm/inc;/usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm/inc/common;/usr/local/gcc/8.5.0/intel-gmmlib/22.2.0/include/igdgmm/util
-- Checking for module 'libigdml'
--   No package 'libigdml' found
-- Metrics Library dir: /usr/local/software/compute-runtime-22.43.24558/third_party/metrics_library
-- Checking for module 'libigdmd'
--   No package 'libigdmd' found
-- Metrics Discovery dir: /usr/local/software/compute-runtime-22.43.24558/third_party/metrics_discovery
-- Checking for module 'libva>=1.0.0'
--   Found libva, version 1.2.0
-- Looking for vaGetLibFunc in va
-- Looking for vaGetLibFunc in va - found
-- Using libva 
-- LibVA include dirs: /usr/local/gcc/8.5.0//libva/2.2.0/include
-- AUB_STREAM_DIR = 
-- Engine node dir: /usr/local/software/compute-runtime-22.43.24558/third_party/aub_stream/headers
-- All supported platforms:  PVC DG2 XE_HP_SDV TGLLP DG1 RKL ADLS ADLP ADLN ICLLP LKF EHL SKL KBL GLK CFL BXT BDW
-- All tested platforms: 
-- Default supported platform: XE_HP_SDV
-- Default tested platform: 
-- All supported core families: GEN8;GEN9;GEN11;GEN12LP;XE_HP_CORE;XE_HPG_CORE;XE_HPC_CORE
-- All tested core families: 
-- Default tested family name: 
-- i915 prelim headers detection: FALSE
-- i915 includes dir: /usr/local/software/compute-runtime-22.43.24558/third_party/uapi
-- Performing Test COMPILER_SUPPORTS_INDIRECT_BRANCH_THUNK
-- Performing Test COMPILER_SUPPORTS_INDIRECT_BRANCH_THUNK - Success
-- Performing Test COMPILER_SUPPORTS_FUNCTION_RETURN_THUNK
-- Performing Test COMPILER_SUPPORTS_FUNCTION_RETURN_THUNK - Success
[... 551 lines cut ...]
make[2]: *** [shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/build.make:220: bin/built_ins/x64/spirv/fill_image3d.spv] Error 250
make[1]: *** [CMakeFiles/Makefile2:2935: shared/source/built_ins/kernels/CMakeFiles/builtins_spirv.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

Build command is:

mkdir build && cd build && \
cmake	-DCMAKE_INSTALL_PREFIX=${GLOBAL_PREFIX}/intel-graphics-compute-runtime/${INTEL_GRAPHICS_COMPUTE_RUNTIME_VERSION} \
		-DCMAKE_BUILD_TYPE=Release -DNEO_SKIP_UNIT_TESTS=1 .. && \
  make && make install

In this test the software stack is:
intel-gmmlib-22.2.0
intel/libva-2.2.0
intel/intel-graphics-compiler-igc-1.0.12260.1
SPIRV_LLVM_TRANSLATOR_VERSION=a31ffaeef77e23d500b3ea3d35e0c42ff5648ad9
SPIRV_TOOLS_VERSION="1.3.204.1"
VC_INTRINSICS_VERSION=0.6.0
LLVM_VERSION=11.1.0
SPIRV_HEADERS_VERSION="1.3.204.1"
SPIRV_TOOLS_VERSION="1.3.204.1"

Please, can anyone suggest me what am I missing?

Thank you.

JablonskiMateusz, 2022-12-14:
Hi @linsson  
could you please share output from 
```
/usr/local/software/compute-runtime-22.43.24558/build/bin/ocloc -spv_only -file /usr/local/software/compute-runtime-22.43.24558/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /usr/local/software/compute-runtime-22.43.24558/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
```
?

lolxdfly, 2022-12-16:
Hi,
I have the same issue!

@JablonskiMateusz 
The output is the following for me:
`/ocloc: error while loading shared libraries: libocloc.so: cannot open shared object file: No such file or directory`

Looks like ocloc can not find libocloc.so. The libocloc.so can be found in "build/bin/", but it seems like its not in LD_LIBRARY_PATH.

JablonskiMateusz, 2022-12-21:
LD_LIBRARY_PATH should be added to cmake targets automatically, please share verbosed make commands

linsson, 2022-12-22:
Hi, sorry for the delay:

```
/usr/local/software/compute-runtime-22.43.24595.30/build/bin/ocloc -q -spv_only -file /usr/local/software/compute-runtime-22.43.24595.30/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /usr/local/software/compute-runtime-22.43.24595.30/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
/usr/local/software/compute-runtime-22.43.24595.30/build/bin/ocloc: error while loading shared libraries: libocloc.so: cannot open shared object file: No such file or directory
```

linsson, 2022-12-22:
This is the result of 

`make VERBOSE=1
`

[mylog.txt](https://github.com/intel/compute-runtime/files/10289132/mylog.txt)

Actually I can see the full command is 

`LD_LIBRARY_PATH=/usr/local/software/compute-runtime-22.43.24595.30/build/bin /usr/local/software/compute-runtime-22.43.24595.30/build/bin/ocloc -q -spv_only -file /usr/local/software/compute-runtime-22.43.24595.30/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /usr/local/software/compute-runtime-22.43.24595.30/build/bin/built_ins/x64/spirv/ -output_no_suffix -options -cl-kernel-arg-info
`

linsson, 2022-12-22:
I just tried with:

`LD_LIBRARY_PATH=/usr/local/software/compute-runtime-22.43.24595.30/build/bin:${LD_LIBRARY_PATH} /usr/local/software/compute-runtime-22.43.24595.30/build/bin/ocloc -q -spv_only -file /usr/local/software/compute-runtime-22.43.24595.30/shared/source/built_ins/kernels/fill_image3d.builtin_kernel -out_dir /usr/local/software/compute-runtime-22.43.24595.30/build/bin/built_ins/x64/spirv/ -output_no_suffix -options "-cl-kernel-arg-info"
`

and it works fine. but I don't  know how to patch the source code to make it builds successfully.

JablonskiMateusz, 2023-01-02:
please ensure you have proper version of igc-opencl package 
I see you have 1.0.1 
> -- Checking for module 'igc-opencl'
-- Found igc-opencl, version 1.0.1

JablonskiMateusz, 2023-01-03:
you may install igc packages from igc release instead of building igc locally

linsson, 2023-01-10:
> you may install igc packages from igc release instead of building igc locally

Do you mean the deb package? I can't because I am not root.

linsson, 2023-01-10:
> please ensure you have proper version of igc-opencl package I see you have 1.0.1
> 
> > -- Checking for module 'igc-opencl'
> > -- Found igc-opencl, version 1.0.1

I think this is the correct version for compute-runtime v.22.43.24595.30. Anyway if this version were not supported I expect to fail the configure, don't I?

linsson, 2023-01-11:
Update: I realized I can use NEO__IGC_LIBRARY_PATH to have a better LD_LIBRARY_PATH. The software stack is:

INTEL_GMMLIB_VERSION=22.3.0
LIBVA_VERSION=2.2.0
IGC_VERSION=1.0.12504.5
VC_INTRINSICS_VERSION=0.7.1
LLVM_VERSION=11.1.0
OPENCL_CLANG_VERSION=11.0.0
SPIRV_LLVM_TRANSLATOR_VERSION=4ef524240833abfeee1c5b9fff6b1bd53f4806b3
SPIRV_HEADERS_VERSION="1.3.204.1"
SPIRV_TOOLS_VERSION="1.3.204.1"
INTEL_GRAPHICS_COMPUTE_RUNTIME_VERSION=22.43.24595.30

Build command line is now:
`cd compute-runtime-${INTEL_GRAPHICS_COMPUTE_RUNTIME_VERSION} && mkdir build && cd build && cmake -DCMAKE_INSTALL_PREFIX=${GLOBAL_PREFIX}/intel-graphics-compute-runtime/${INTEL_GRAPHICS_COMPUTE_RUNTIME_VERSION} -DCMAKE_BUILD_TYPE=Release -DNEO_SKIP_UNIT_TESTS=1 -DNEO__IGC_LIBRARY_PATH=${LD_LIBRARY_PATH} .. && make VERBOSE=1`

I am able to overcome the previous problem but a new one arises:

[mylog.txt](https://github.com/intel/compute-runtime/files/10392004/mylog.txt)

Any suggestion?

eero-t, 2023-02-27:
Searching **both** _compute-runtime_ and _intel-graphics-compiler_ project for (closed) tickets with same error you see, can often provide you with a fix for your build issue.

For example, these give some hints about the specific issues your build environment may have:
* https://github.com/intel/intel-graphics-compiler/issues/224#issuecomment-1103820725
* https://github.com/intel/compute-runtime/issues/519#issuecomment-1198068830

(This kind of LLVM version incompatibility errors appear e.g. when *all* the components the component stack, including SPIRV ones, have not been built with the same LLVM version.)

As to this:
> The output is the following for me:
> ocloc: error while loading shared libraries: libocloc.so: cannot open shared object file: No such file or directory

Like with any other project providing libraries, if you do not `sudo make install` a project to system, you need to tell dynamic linker / compiler where its libraries can be found.