intel/compute-runtime #594: Broken URLs in current release

hfp, 2022-12-12:
The currently latest release ([22.43.24595.30](https://github.com/intel/compute-runtime/releases/tag/22.43.24595.30)) refers to dead file links for the IGC component, i.e., 1.0.12504.3 is gone on IGC's side and supersede by 1.0.12504.5.

Broken URLs:
* https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12504.3/intel-igc-core_1.0.12504.3_amd64.deb
* https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12504.3/intel-igc-opencl_1.0.12504.3_amd64.deb

hazcoper, 2022-12-14:
I also tried to download the available versions
But it fails when doing Intel-igc-opencl becuase of the dependencies, as it is expecting another version

hfp, 2022-12-14:
Indeed, the missing dependencies are related to the broken URLs because IGC folks superseded own builds and dropped the older build/sub-version. Of course, the runtime package here must be rebuilt to account for the build-increment on IGC's side. If (sub-)components can drop like this (or IGC in particular), it can be advisable to copy the packages or attach release artifacts instead of URLs.

On the positive side, it's possible to force-install the newer IGC build along with the other current compute runtime components; however, this "solution" remains a source of suggested fix for `apt` and/or `dpkg`. HOWEVER, it seem this release resolved https://github.com/intel/compute-runtime/issues/576 (root cause was IGC once more).

closed 2022-12-15 by psidor99

martadinata666, 2022-12-21:
URL corrected but the package deps not corrected, should i open new issue?

```
 44.53 Setting up intel-igc-core (1.0.12504.5) ...
#7 44.54 Setting up intel-igc-opencl (1.0.12504.5) ...
#7 44.56 dpkg: dependency problems prevent configuration of intel-opencl-icd:
#7 44.56  intel-opencl-icd depends on intel-igc-opencl (= 1.0.12504.3); however:
#7 44.56   Version of intel-igc-opencl on system is 1.0.12504.5.
#7 44.56 
#7 44.56 dpkg: error processing package intel-opencl-icd (--install):
#7 44.56  dependency problems - leaving unconfigured
#7 44.56 Setting up libigdgmm12:amd64 (22.3.0) ...
#7 44.58 Setting up intel-level-zero-gpu (1.3.24595.30) ...
#7 44.59 Processing triggers for libc-bin (2.36-6) ...
#7 44.63 Errors were encountered while processing:
#7 44.63  intel-opencl-icd
```

hfp, 2022-12-23:
Why is this closed? There is now a 1-month cycle for releases which means problems can persist for a longer period. Correcting the URLs is not sufficient. Potentially all DEBs must be updated to refer to the increased version number. Really, this needs the same flow as a new release. I think stretching the lifetime of releases (to one month) may go with tightening the measures for every such release.

hfp, 2022-12-23:
> URL corrected but the package deps not corrected, should i open new issue?

I suggest doing just this and also referring this issue like #594.