intel/compute-runtime #597: Coincident platform names for integrated and discrete graphics

FreddieWitherden, 2022-12-28:
On a system with an A770M discrete GPU and Alder Lake CPU using runtime 22.43.24595.35 there are *two* OpenCL platforms both named "Intel(R) OpenCL HD Graphics" with one device a piece.  I suspect this is required on account of the underlying GPUs supporting slightly different extensions.

However, the fact that both platforms have the same name is problematic for applications which enable users to specify a platform by name rather than number.  It would therefore be helpful to rename these platforms to give some idea of what they represent.  (Similarly, it would also be helpful if the device names were not just "Intel(R) Graphics [0xxxx]", but this is somewhat less of an issue if the platform names are sensible.)

nyanmisaka, 2022-12-29:
https://github.com/intel/compute-runtime/blob/master/opencl/source/platform/platform_info.h

I think it is unlikely that the platform name will be changed since it is used to distinguish different OpenCL runtimes across vendors, such as NVIDIA CUDA and AMD APP.

The device name of Arc dGPU shall be included in the next release with this commit: ac80293

FreddieWitherden, 2022-12-29:
> https://github.com/intel/compute-runtime/blob/master/opencl/source/platform/platform_info.h
>
> I think it is unlikely that the platform name will be changed since it is used to distinguish different [...]

At least for NVIDIA and AMD they use one platform to group all devices (so an even old NVIDIA GPU and a current generation one with radically different capabilities will all appear as being part of the sole NVIDIA platform).  The approach of using multiple platforms with the same names is likely to cause issues with existing software and result in confusing user experiences (think of seeing a platform selection menu with the same name listed twice).

Is there a possibility of all Intel graphics devices being grouped under a single platform?

JablonskiMateusz, 2023-01-03:
Grouping all Intel devices under a single platform is not being considered. On the other hand, differentiating platform names is a good idea, we will follow up.