intel/compute-runtime #598: ZeroCopy buffer still seen copy usage on DG2

nyanmisaka, 2023-01-03:
https://www.intel.com/content/www/us/en/developer/articles/training/getting-the-most-from-opencl-12-how-to-increase-performance-by-minimizing-buffer-copies-on-intel-processor-graphics.html

As per the official docs linked above, I create the zero-copy buffer with `CL_MEM_ALLOC_HOST_PTR` on DG2 A380 and then use `clEnqueueMapBuffer()`, `memset()` and `clEnqueueUnmapMemObject()` to fill data to it. Resizable-bar is enabled in the BIOS.

[image]

I can still see the copy usage in task manager on Windows. But this cannot be reproduced with the same code in Intel iGPU, as well as other manufacturers such as AMD and NVIDIA GPU.

Is this the expected behavior on Intel Arc discrete GPU or is it an issue?

MichalMrozek, 2023-01-09:
On discrete GPU buffers are not created as zero-copy.
So the transfers that you see are expected.
If you want to force zero copy behavior you need to use this extension -> https://registry.khronos.org/OpenCL/extensions/intel/cl_intel_mem_force_host_memory.html

nyanmisaka, 2023-01-09:
Thanks for you help! It works just fine now.