intel/compute-runtime #599: Test failure in Skylake AUB

ArchangeGabriel, 2023-01-07:
While building latest release (22.43.24595.30) on ArchLinux with IGC 1.0.12812.4 built with a LLVM15 stack, I’m seeing one test failure:
```
========================
==  SKL AUBs FAILED   ==
========================
Tests run:      1401
Tests passed:   1053
Tests skipped:  347
Tests failed:   1
Tests disabled: 2
 Time elapsed:  10922 ms
========================
[  FAILED  ][ SKL ][ 0 ] AUBRunKernelIntegrateTest.GivenDeviceSideVmeThenExpectationsMet

make[2]: *** [target_aub_tests/gen9/skl/CMakeFiles/run_skl_9_aub_tests.dir/build.make:76: run_skl_9_aub_tests] Error 1
make[1]: *** [CMakeFiles/Makefile2:20621: target_aub_tests/gen9/skl/CMakeFiles/run_skl_9_aub_tests.dir/all] Error 2
```

jandres742, 2023-01-09:
thanks @ArchangeGabriel . Dont know if you actually need to run those tests. Could you try with this cmake rule?
`DO_NOT_RUN_AUB_TESTS=1`

https://github.com/intel/compute-runtime/blob/9cec84d3d4627a60306476c190f37e199b477764/CMakeLists.txt#L248

ArchangeGabriel, 2023-01-09:
I don’t specifically need to run tests, I just run all of them by default as part of packaging for ArchLinux. I can indeed just disable those one for now thanks to this env var you’ve pointed me at. ;)

ArchangeGabriel, 2023-01-09:
That being said I then face other errors: https://gist.github.com/ArchangeGabriel/cdb9917a8f136ca6cabd1951686ebf9d

JablonskiMateusz, 2023-01-12:
@ArchangeGabriel please try building with same IGC as mentioned in release notes: [intel/intel-graphics-compiler@igc-1.0.12504.5](https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.12504.5)

JablonskiMateusz, 2023-01-25:
Could you share the entire build log and cmake flags used to build?