intel/compute-runtime #603: Building errors: `stateless_copy_buffer_to_buffer` and `bindful_copy_buffer_to_buffer`

boberfly, 2023-01-14:
Hi all, I've encountered some build errors trying to build intel-compute-runtime, some of the SPIR-V into bins I presume.

I'm loosely following the build setup of how Arch Linux is bundling things and watching Blender:
https://developer.blender.org/D16984

**Environment:**
intel-compute-runtime - `22.49.25018.21`
LLVM - `15.0.7`
vc-intrinsics - `v0.11.0.0`
LLVM-SPIRV-Translator - `llvm_release_150` branch tip (`78ad93b9952cb78889b86e9aa390d013c49dca0b`)
intel-graphics-compiler - `igc-1.0.13064.4`
level-zero - `v1.8.12`
gmmlib - `22.3.0`
opencl-clang - `ocl-open-150` branch tip (`efdae77badff1cfe9b64802927e59f7740d671cc`)

[Full log here](https://github.com/intel/compute-runtime/files/10417337/08be772f-build.982556.log)

Interesting bits here:

`[585/2079] Generating ../../../../bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.gen, ../../../../bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.bin
FAILED: bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.gen bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.bin /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.gen /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_0_Gen8core.bin 
cd /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/ocloc -q -gen_file -file /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/stateless_copy_buffer_to_buffer_stateless_Gen8core.spv -spirv_input -device bdw -cl-intel-greater-than-4GB-buffer-required -64 -output stateless_copy_buffer_to_buffer_stateless_0 -out_dir /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8 -revision_id 0 -options -cl-kernel-arg-info
/usr/include/c++/12.2.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = llvm::SmallPtrSet<llvm::Argument*, 2>; _Dp = std::default_delete<llvm::SmallPtrSet<llvm::Argument*, 2> >; typename std::add_lvalue_reference<_Tp>::type = llvm::SmallPtrSet<llvm::Argument*, 2>&]: Assertion 'get() != pointer()' failed.`

`[592/2079] Generating ../../../../bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.gen, ../../../../bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.bin
FAILED: bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.gen bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.bin /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.gen /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_0_Gen8core.bin 
cd /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/shared/source/built_ins/kernels && LD_LIBRARY_PATH=/buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/ocloc -q -gen_file -file /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8/bindful_copy_buffer_to_buffer_Gen8core.spv -spirv_input -device bdw -force_stos_opt -64 -output bindful_copy_buffer_to_buffer_0 -out_dir /buildstream/carbonOS/pkgs/intel/compute-runtime.bst/_builddir/bin/built_ins/x64/gen8 -revision_id 0 -options -cl-kernel-arg-info
/usr/include/c++/12.2.0/bits/unique_ptr.h:445: typename std::add_lvalue_reference<_Tp>::type std::unique_ptr<_Tp, _Dp>::operator*() const [with _Tp = llvm::SmallPtrSet<llvm::Argument*, 2>; _Dp = std::default_delete<llvm::SmallPtrSet<llvm::Argument*, 2> >; typename std::add_lvalue_reference<_Tp>::type = llvm::SmallPtrSet<llvm::Argument*, 2>&]: Assertion 'get() != pointer()' failed.`

Let me know if I can help with more information.

frantisekz, 2023-01-14:
This would be https://github.com/intel/intel-graphics-compiler/issues/273 .

frantisekz, 2023-01-17:
Backtrace for this is:

~~~~
(gdb) bt
#0  0x00007ffff7cfde5c in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x00007ffff7cada76 in raise () from /lib64/libc.so.6
#2  0x00007ffff7c977fc in abort () from /lib64/libc.so.6
#3  0x00007ffff7ad7e80 in std::__glibcxx_assert_fail(char const*, int, char const*, char const*) () from /lib64/libstdc++.so.6
#4  0x00007ffff5a32375 in IGC::ScalarArgAsPointerAnalysis::findArgs(llvm::Instruction*) () from /lib64/libigc.so.1
#5  0x00007ffff5a325a4 in IGC::ScalarArgAsPointerAnalysis::findArgs(llvm::Instruction*) () from /lib64/libigc.so.1
#6  0x00007ffff5a325a4 in IGC::ScalarArgAsPointerAnalysis::findArgs(llvm::Instruction*) () from /lib64/libigc.so.1
#7  0x00007ffff5a32f87 in IGC::ScalarArgAsPointerAnalysis::analyzePointer(llvm::Value*) [clone .part.0] () from /lib64/libigc.so.1
#8  0x00007ffff5a34353 in IGC::ScalarArgAsPointerAnalysis::runOnModule(llvm::Module&) () from /lib64/libigc.so.1
#9  0x00007fffeefbc148 in llvm::legacy::PassManagerImpl::run(llvm::Module&) () from /lib64/libLLVM-15.so
#10 0x00007ffff584ecaf in IGC::CommonOCLBasedPasses(IGC::OpenCLProgramContext*, std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >) () from /lib64/libigc.so.1
#11 0x00007ffff584f803 in IGC::UnifyIRSPIR(IGC::OpenCLProgramContext*, std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >, std::unique_ptr<llvm::Module, std::default_delete<llvm::Module> >)
    () from /lib64/libigc.so.1
#12 0x00007ffff582536a in TC::TranslateBuildSPMD(TC::STB_TranslateInputArgs const*, TC::STB_TranslateOutputArgs*, TC::TB_DATA_FORMAT, IGC::CPlatform const&, float, ShaderHash const&) ()
   from /lib64/libigc.so.1
#13 0x00007ffff5fdd037 in IGC::VLD::TranslateBuildSPMDAndESIMD(TC::STB_TranslateInputArgs const*, TC::STB_TranslateOutputArgs*, TC::TB_DATA_FORMAT, IGC::CPlatform const&, float, ShaderHash const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&) [clone .constprop.0] () from /lib64/libigc.so.1
#14 0x00007ffff58275cf in TC::TranslateBuild(TC::STB_TranslateInputArgs const*, TC::STB_TranslateOutputArgs*, TC::TB_DATA_FORMAT, IGC::CPlatform const&, float) () from /lib64/libigc.so.1
#15 0x00007ffff5919093 in IGC::IgcOclTranslationCtx<0ul>::Impl::Translate(unsigned long, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int, void*) const () from /lib64/libigc.so.1
#16 0x00007ffff59197c3 in IGC::IgcOclTranslationCtx<1ul>::TranslateImpl(unsigned long, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, CIF::Builtins::Buffer<1ul>*, unsigned int) () from /lib64/libigc.so.1
#17 0x00007ffff7f36eec in IGC::IgcOclTranslationCtx<1ul>::Translate<IGC::OclTranslationOutput<1ul> > (tracingOptionsCount=0, tracingOptions=0x0, internalOptions=<optimized out>, options=<optimized out>, 
    src=<optimized out>, this=<optimized out>) at /usr/include/igc/ocl_igc_interface/igc_ocl_translation_ctx.h:33
#18 NEO::OfflineCompiler::buildSourceCode (this=0x55555556f9d0) at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/offline_compiler.cpp:367
#19 0x00007ffff7f3a215 in NEO::OfflineCompiler::build (this=0x55555556f9d0) at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/offline_compiler.cpp:398
#20 0x00007ffff7f46f91 in SafetyGuardLinux::call<int, NEO::OfflineCompiler, int (NEO::OfflineCompiler::*)()> (this=this@entry=0x7fffffffdcb0, object=object@entry=0x55555556f9d0, method=<optimized out>, 
    retValueOnCrash=retValueOnCrash@entry=-5152) at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/utilities/linux/safety_guard_linux.h:62
#21 0x00007ffff7f46cc0 in buildWithSafetyGuard (compiler=compiler@entry=0x55555556f9d0)
    at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/utilities/linux/safety_caller_linux.cpp:20
#22 0x00007ffff7f30768 in Ocloc::Commands::compile (argHelper=argHelper@entry=0x55555556c440, args=std::vector of length 17, capacity 17 = {...})
    at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/ocloc_interface.cpp:130
#23 0x00007ffff7f1b414 in oclocInvoke (numArgs=17, argv=0x7fffffffe0d8, numSources=0, dataSources=0x0, lenSources=0x0, nameSources=0x0, numInputHeaders=0, dataInputHeaders=0x0, lenInputHeaders=0x0, 
    nameInputHeaders=0x0, numOutputs=0x0, dataOutputs=0x0, lenOutputs=0x0, nameOutputs=0x0) at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/ocloc_api.cpp:52
#24 0x0000555555555097 in main (argc=<optimized out>, argv=<optimized out>) at /builddir/build/BUILD/compute-runtime-22.49.25018.21/shared/offline_compiler/source/main.cpp:11
~~~~

frantisekz, 2023-01-24:
@JablonskiMateusz @pszymich Since the igc issue ( https://github.com/intel/intel-graphics-compiler/issues/273 ) was closed (maybe because it was described as affecting just the 12812 branch, but in reality it's from that branch to master), what would the best path forward here now? Just wait a while and keep 12812.4 packaged from distro perspective? It's relevant at least for Fedora and Arch already.

Thanks!

JablonskiMateusz, 2023-01-27:
issue transferred to IGC

closed 2023-01-27 by JablonskiMateusz