intel/compute-runtime #606: 'sycl-ls' command error with recent installation

ShravanBM, 2023-01-20:
System: Ubuntu 22.04, ARC 750,
Installed driver for ARC 750.
	Installation guide: https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html
	Driver version (o/p of clinfo): 22.27.023570 (latest)
One API Toolkit
	Installation guide: https://www.intel.com/content/www/us/en/develop/documentation/installation-guide-for-intel-oneapi-toolkits-linux/top.html
	version 2023.0.0.25537 (latest)

Activated OneAPI Env and ran 'sycl-ls'. Got below error.
$ sycl-ls
Abort was called at 39 line in file:
./shared/source/gmm_helper/client_context/gmm_client_context.cpp
Aborted (core dumped)

Kindly suggest right ARC driver version for latest OneAPI Toolkit.

jandres742, 2023-01-23:
@ShravanBM : thanks. That error is commonly seen when the installed GMMLIB is not compatible with the OCL driver installed.

Please try installing the packages again, after a cleaning all existing packages. if you still see problems, please share the output when running with strace.

JablonskiMateusz, 2023-01-27:
@ShravanBM please share strace output

ShravanBM, 2023-02-01:
[clinfo_op.txt](https://github.com/intel/compute-runtime/files/10555868/clinfo_op.txt)
[strace_syclls.txt](https://github.com/intel/compute-runtime/files/10555870/strace_syclls.txt)
[syclls_op.txt](https://github.com/intel/compute-runtime/files/10555871/syclls_op.txt)

I have installed Dg2 and OneAPI after fresh installation of Ubuntu 22.04.

At end of clinfo or sycl-ls command I get "Floating point exception (core dumped)". I am using ARC 750.

JablonskiMateusz, 2023-02-01:
thanks @ShravanBM 
I see in your logs that you have driver 22.43.24595.35. Have you installed it from our release https://github.com/intel/compute-runtime/releases/tag/22.43.24595.30 ? 

Could you catch callstack of the exception?

ShravanBM, 2023-02-02:
You are right. Issue 613 seems similar.

I installed driver from the internal link that worked for other team. Other team had suggested me ARC770 but I used A750.

Will try with ARC 770 and update here. May take few days.

eero-t, 2023-02-27:
You can get callstack running `gdb sycl-ls`, and giving it the `bt` (backtrace) command.