intel/compute-runtime #607: Wrong symlinks in the latest release

AlexeySachkov, 2023-01-20:
Hi folks,

I've just installed the latest available release [22.43.24595.30](https://github.com/intel/compute-runtime/releases/tag/22.43.24595.30) and it doesn't work out-of-the box.

```
$ l neo-22.43/usr/local/lib/
igc
libiga64.so -> libiga64.so.1
libiga64.so.1 -> libiga64.so.1.0.12504.3
libiga64.so.1.0.12504.5
libigc.so -> libigc.so.1
libigc.so.1 -> libigc.so.1.0.12504.3
libigc.so.1.0.12504.5
libigdfcl.so -> libigdfcl.so.1
libigdfcl.so.1 -> libigdfcl.so.1.0.12504.3
libigdfcl.so.1.0.12504.5
libopencl-clang.so -> libopencl-clang.so.11
libopencl-clang.so.11
```

Symlinks `libiga64.so.1`, `libigc.so.1` and `libigdfcl.so.1` are simply wrong. They point to file which ends with `.3`, but the package contains libraries which end with `.5`. I can easily recreate symlinks properly locally, but that is annoying to do every time you recreate the environment.

JablonskiMateusz, 2023-01-27:
Hi @AlexeySachkov , thanks for reporting the issue. Could you please create an issue in IGC repo (and close this one) as mentioned libraries are a part of their release? https://github.com/intel/intel-graphics-compiler/releases/tag/igc-1.0.12504.5

AlexeySachkov, 2023-01-27:
@JablonskiMateusz, sure, see intel/intel-graphics-compiler#283

closed 2023-01-27 by AlexeySachkov