intel/compute-runtime #616: Conflicting packages in Ubuntu (intel-opencl-icd and intel-level-zero-gpu)

kooped, 2023-02-17:
When installing the runtime packages following the guide at [step-4-install-run-time-packages](https://dgpu-docs.intel.com/installation-guides/ubuntu/ubuntu-jammy-arc.html#step-4-install-run-time-packages). There seems to be a conflict with `intel-opencl-icd` and `intel-level-zero-gpu`, specifically with `libze_intel_gpu.so.1`

For example when installing `intel-opencl-icd` when `intel-level-zero-gpu` is installed:

``` 
$ sudo apt install intel-opencl-icd
...
Preparing to unpack .../intel-opencl-icd_22.49.25018.24-0ubuntu1~22.04.sav0_amd64.deb ...
Unpacking intel-opencl-icd (22.49.25018.24-0ubuntu1~22.04.sav0) ...
dpkg: error processing archive /var/cache/apt/archives/intel-opencl-icd_22.49.25018.24-0ubuntu1~22.04.sav0_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libze_intel_gpu.so.1', which is also in package intel-level-zero-gpu 1.3.25018.23+i554~22.04
Errors were encountered while processing:
 /var/cache/apt/archives/intel-opencl-icd_22.49.25018.24-0ubuntu1~22.04.sav0_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

eero-t, 2023-02-27:
There are some differences in how packages in Intel and Debian/Ubuntu repositories package compute stack (due to legacy reasons).

What's the output of:
`apt policy intel-opencl-icd when intel-level-zero-gpu`

I.e. do these packages come from the same repository, or from different ones?

kooped, 2023-02-28:
Thanks for your help!

There was a conflict with another PPA https://ppa.launchpadcontent.net/savoury1/display/ubuntu 

I have removed this, and it works fine now