intel/compute-runtime #617: Cannot Allocate More than 8 GB on A770 16GB

BA8F0D39, 2023-02-19:
I have a A770 16 GB and I installed intel-compute-runtime 22.43.24595.30 and intel extensions for pytorch v1.13.10+xpu

When you try to allocate more than 8 GB.
```
import torch
import torchvision

import intel_extension_for_pytorch as ipex

size = 46000
w = torch.rand(size, size, dtype=torch.bfloat16, device='xpu')
x = torch.rand(size, size, dtype=torch.bfloat16, device='xpu')
```
The python script crashes.
```
RuntimeError: Native API failed. Native API returns: -997 (The plugin has emitted a backend specific error) -997 (The plugin has emitted a backend specific error)
```
https://github.com/intel/intel-extension-for-pytorch/issues/296

However, allocating less than 8 GB, works perfectly. Why is there a memory limit on A770 16 GB?

arthuryuan1987, 2023-02-20:
@BA8F0D39 Let me confirm are you using WSL?

If yes, the possible cause is Linux subsystem might follow Windows operating system limitation. The limitation would be around half of memory in single process, ~8GB on your 16GB card. I am confirming it with our driver guys. Thanks.

BA8F0D39, 2023-02-20:
@arthuryuan1987 
I am on Arch Linux Kernel 6.2rc8 with A770 16 GB. 
Not on Windows. Not on WSL

arthuryuan1987, 2023-02-22:
> @arthuryuan1987 I am on Arch Linux Kernel 6.2rc8 with A770 16 GB. Not on Windows. Not on WSL
> 
> Allocating less than 8GB on A770 16GB works. Allocating more than 8GB on A770 16GB does not work

Thanks, I will file an internal issue and talk with our driver guys. Keep you updated.

kgibala, 2026-02-26:
Hi @BA8F0D39,

We're closing this issue due to inactivity. We haven't received the requested information or updates for an extended period.

If you encounter this or a similar problem in the future, please feel free to create a new issue. When submitting, please follow our [Issue Submission Guide](https://github.com/intel/compute-runtime/blob/master/ISSUE_SUBMISSION_GUIDE.md) to help us understand and address your concern efficiently.