intel/compute-runtime #618: GPU to GPU Memory Bandwidth Limited on A770 16 GB

BA8F0D39, 2023-02-19:
I have a A770 16 GB and I installed intel-compute-runtime 22.43.24595.30 and intel extensions for pytorch v1.13.10+xpu on Linux Kernel 6.2rc8

I made a script to test memory transfers from GPU to GPU on the A770.
```
import torch
import torchvision

import intel_extension_for_pytorch as ipex

import time

sizes = [1000, 1000,1000,1000,1000,2000, 3000, 10000, 20000,30000,40000,42000,43000,44000,45000,]

for size in sizes:

    array0 = torch.rand(size, size, dtype=torch.bfloat16, device='xpu')
    array1 = torch.rand(size, size, dtype=torch.bfloat16, device='xpu')

    torch.xpu.synchronize()
    start = time.time()
    array0 = torch.clone(array1)
    torch.xpu.synchronize()
    end = time.time()
    transferrate = (size*size*16)/(end - start)
    datasize = (size*size*16)
    print("==========")
    print("Transfering " + str(datasize/8E9) + " GB")
    print("Bandwidth " + str(transferrate/8E9) + " GB/s")
    print("==========")

    torch.xpu.empty_cache()
```

Running the script gives a maximum GPU to GPU transfer rate of around 100 GB/s
```
==========
Transfering 0.002 GB
Bandwidth 29.746836879432625 GB/s
==========
==========
Transfering 0.008 GB
Bandwidth 50.45779248120301 GB/s
==========
==========
Transfering 0.018 GB
Bandwidth 71.29128611898017 GB/s
==========
==========
Transfering 0.2 GB
Bandwidth 80.71401905128451 GB/s
==========
==========
Transfering 0.8 GB
Bandwidth 91.80419151846785 GB/s
==========
==========
Transfering 1.8 GB
Bandwidth 102.16166711772665 GB/s
==========
==========
Transfering 3.2 GB
Bandwidth 103.13888713854288 GB/s
==========
==========
Transfering 3.528 GB
Bandwidth 102.9577837521917 GB/s
==========
```

Why is the GPU to GPU transfer rate limited to 100 GB/s, when the A770 16 GB's bandwidth is 512 GB/s?

BartusW, 2023-02-20:
Thank you for your feedback. Performance numbers are collected with high level PyTorch S/W stack.
Please try to remove high level abstraction layers and use OpenCL or Level-Zero native API on your system and report score.
You can use:
- https://github.com/intel/compute-benchmarks and memory-benchmarks scenarios 
or 
- https://github.com/krrishnarraj/clpeak

BA8F0D39, 2023-02-20:
@BartusW 

clpeak has weird results, where the enqueueWriteBuffer is 2x faster than enqueueReadBuffer? Why is there a difference in reads and writes? Every other GPU has the approximately the same read and write speed 

```
Platform: Intel(R) OpenCL HD Graphics
  Device: Intel(R) Graphics [0x56a0]
    Driver version  : 22.43.30 (Linux x64)
    Compute units   : 512
    Clock frequency : 2400 MHz

    Global memory bandwidth (GBPS)
      float   : 397.87
      float2  : 403.63
      float4  : 407.18
      float8  : 416.18
      float16 : 421.80

    Single-precision compute (GFLOPS)
      float   : 13017.51
      float2  : 11136.49
      float4  : 10402.49
      float8  : 10026.09
      float16 : 9695.57

    Half-precision compute (GFLOPS)
      half   : 19543.72
      half2  : 19489.39
      half4  : 19523.66
      half8  : 19454.95
      half16 : 19336.14

    No double precision support! Skipped

    Integer compute (GIOPS)
      int   : 4380.31
      int2  : 4385.50
      int4  : 4403.38
      int8  : 4273.37
      int16 : 5004.16

    Integer compute Fast 24bit (GIOPS)
      int   : 4361.75
      int2  : 4369.68
      int4  : 4387.98
      int8  : 4265.73
      int16 : 4995.43

    Transfer bandwidth (GBPS)
      enqueueWriteBuffer              : 21.64
      enqueueReadBuffer               : 8.92
      enqueueWriteBuffer non-blocking : 22.81
      enqueueReadBuffer non-blocking  : 9.10
      enqueueMapBuffer(for read)      : 20.58
        memcpy from mapped ptr        : 22.62
      enqueueUnmap(after write)       : 23.62
        memcpy to mapped ptr          : 22.44

    Kernel launch latency : 34.76 us
```

BA8F0D39, 2023-02-21:
@BartusW 

I compiled and ran memory_benchmark_l0 and memory_benchmark_ocl
https://github.com/intel/compute-benchmarks

I attached the results as .txt files and the memory benchmarks have nonsensical outputs such as inf, negative numbers, ERROR and many others.

```

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/49
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/51
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/53
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/55
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/57
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/59
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/61
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/63
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/65
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/67
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/69
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/71
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/stream_memory_l0.cpp:173

[  FAILED  ] StreamMemoryTest/StreamMemoryTest.Test/73
[... 120 lines cut ...]
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/usm_copy_l0.cpp:101

[  FAILED  ] UsmCopyTest/UsmCopyTest.Test/124
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/usm_copy_l0.cpp:101

[  FAILED  ] UsmCopyTest/UsmCopyTest.Test/128
FAILED assertion ASSERT_ZE_RESULT_SUCCESS(zeEventQueryKernelTimestamp(event, &timestampResult))
	value: 1 (ZE_RESULT_NOT_READY)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/l0/usm_copy_l0.cpp:101
```

[api_overhead_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10788944/api_overhead_benchmark_l0.txt)
[api_overhead_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10788945/api_overhead_benchmark_ocl.txt)
[multitile_memory_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10788946/multitile_memory_benchmark_l0.txt)

[... cut ...]
[gpu_cmds_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10788948/gpu_cmds_benchmark_l0.txt)
[show_devices_l0.txt](https://github.com/intel/compute-runtime/files/10788949/show_devices_l0.txt)
[memory_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10788950/memory_benchmark_l0.txt)
[show_devices_ocl.txt](https://github.com/intel/compute-runtime/files/10788951/show_devices_ocl.txt)
[memory_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10788952/memory_benchmark_ocl.txt)

BartusW, 2023-02-21:
Thanks for follow up and additional experiments.
OpenCL API tests with clPeak, and Device-2-Device memory bandwidth scenarios looks healthy, but we can see 6.2 rc8 performance discrepancies against kernel launch latency. This could be one of the reason of low end-to-end efficiency observed above PyTorch layer.
We are queueing this github issue as our internal task for reproduction and analyzes. 
We keep the issue as 'open' until confirm and deliver potential fix. 

We are also working to confirm Level-Zero failures, which are not visible in Intel CI environment.

FreddieWitherden, 2023-02-21:
> Thanks for follow up and additional experiments. OpenCL API tests with clPeak, and Device-2-Device memory bandwidth scenarios looks healthy.

For FP32 clPeak maxes out at ~13 TFLOP/s.  But 512 (EUs) * 2400 (Mhz) * 8 (SIMD length) * 2 (FMA) = ~20 TFLOP/s. For comparison the same clPeak FP32 kernels on an iGPU get the expected peak.  Similarly, FP16 is exactly a factor of two off of the theoretical peak.

BA8F0D39, 2023-02-21:
@BartusW 

memory_benchmark_ocl has a few errors and crashed my system.

```
[  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/1
FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
	value: -59 (CL_INVALID_OPERATION)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48

[  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/3
FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
	value: -59 (CL_INVALID_OPERATION)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48

[  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/5
FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
	value: -59 (CL_INVALID_OPERATION)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48

[  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/7
FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
	value: -59 (CL_INVALID_OPERATION)
	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48

```

BartusW, 2023-02-22:
> 
> For FP32 clPeak maxes out at ~13 TFLOP/s. But 512 (EUs) * 2400 (Mhz) * 8 (SIMD length) * 2 (FMA) = ~20 TFLOP/s. For comparison the same clPeak FP32 kernels on an iGPU get the expected peak. Similarly, FP16 is exactly a factor of two off of the theoretical peak.

clPeak SP32 FMA (a.k.a MAD multiply+add) synthetic test observations are correct. In case of Arc family, MAD instruction is split into two ticks, while in mentioned integrated GFX there are fused two ops per single instruction in one tick in this test. 
Observed MAD throughput is exposed only in synthetic cases while there is no penalty in real scenarios.

FreddieWitherden, 2023-02-22:
> clPeak SP32 FMA (a.k.a MAD multiply+add) synthetic test observations are correct. In case of Arc family, MAD instruction is split into two ticks, while in mentioned integrated GFX there are fused [...]

So as a point of reference on a 96 EU iGPU my SGEMM kernels (derived from https://github.com/intel/intel-graphics-compiler/issues/254) can get 1.5 TFLOP/s or ~75% of peak.  On my A770M (~16 TFLOP/s peak) and an appropriately scaled up problem the same kernels strike out at ~6.3 TFLOP/s, or ~37.5% of peak.  *Precisely* a factor of two off what an iGPU can benchmark, and inline with the synthetics.

BartusW, 2023-02-28:
> @BartusW
> 
> memory_benchmark_ocl has a few errors and crashed my system. I think the Level Zero/OpenCL API returns invalid values too, which causes many applications like pytorch/tensorflow to behave weirdly.
> 
> ```
> [  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/1
> FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
> 	value: -59 (CL_INVALID_OPERATION)
> 	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48
> 
> [  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/3
> FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
> 	value: -59 (CL_INVALID_OPERATION)
> 	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48
> 
> [  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/5
> FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
> 	value: -59 (CL_INVALID_OPERATION)
> 	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48
> 
> [  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/7
> FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
> 	value: -59 (CL_INVALID_OPERATION)
> 	Location: /opt/test/compute-benchmarks/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48
> ```

Hello, 
Intel team quickly checked latest UMD compute driver release with generic 6.2.RC6 kernel and we can't confirm mentioned failures. However, we focused on latest release rather that visiting back previous driver package mentioned in the issue: package 22.43.24595.30 and RC8.

Please update driver to the latest release: 22.49.25018.24 https://github.com/intel/compute-runtime/releases
Mentioned failures highlighted above manifested in generic path, driver refresh could be a remedy here.

BA8F0D39, 2023-03-01:
@BartusW 

I upgrade to 22.49.25018.24 and kernel 6.2.1 but the errors still persists. I have an i5-13600K raptor lake CPU and 64 GB of RAM. Resizable bar is enabled and Arc A770 is the only OpenCL device available on the system.

```
03:00.0 VGA compatible controller: Intel Corporation DG2 [Arc A770] (rev 08) (prog-if 00 [VGA controller])
	Subsystem: Intel Corporation Device 1020
	Flags: bus master, fast devsel, latency 0, IRQ 172
	Memory at 84000000 (64-bit, non-prefetchable) [size=16M]
	Memory at 4000000000 (64-bit, prefetchable) [size=16G]
	Expansion ROM at 85000000 [disabled] [size=2M]
	Capabilities: [40] Vendor Specific Information: Len=0c <?>
	Capabilities: [70] Express Endpoint, MSI 00
	Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable+ 64bit+

sudo dpkg -i *.deb
(Reading database ... 22170 files and directories currently installed.)
Preparing to unpack intel-igc-core_1.0.12812.24_amd64.deb ...
Unpacking intel-igc-core (1.0.12812.24) over (1.0.12812.24) ...
Preparing to unpack intel-igc-opencl_1.0.12812.24_amd64.deb ...
Unpacking intel-igc-opencl (1.0.12812.24) over (1.0.12812.24) ...
Preparing to unpack intel-level-zero-gpu_1.3.25018.24_amd64.deb ...
Unpacking intel-level-zero-gpu (1.3.25018.24) over (1.3.25018.24) ...
Preparing to unpack intel-opencl-icd_22.49.25018.24_amd64.deb ...
Unpacking intel-opencl-icd (22.49.25018.24) over (22.49.25018.24) ...
dpkg: warning: downgrading libigdgmm12:amd64 from 22.3.3+i550~22.04 to 22.3.0
Preparing to unpack libigdgmm12_22.3.0_amd64.deb ...
Unpacking libigdgmm12:amd64 (22.3.0) over (22.3.3+i550~22.04) ...
Setting up intel-igc-core (1.0.12812.24) ...
Setting up intel-igc-opencl (1.0.12812.24) ...
Setting up libigdgmm12:amd64 (22.3.0) ...
Setting up intel-level-zero-gpu (1.3.25018.24) ...
Setting up intel-opencl-icd (22.49.25018.24) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1) ...
```

clinfo
```
Number of platforms                               1
  Platform Name                                   Intel(R) OpenCL HD Graphics
  Platform Vendor                                 Intel(R) Corporation
  Platform Version                                OpenCL 3.0 
  Platform Profile                                FULL_PROFILE
  Platform Extensions                             cl_khr_byte_addressable_store cl_khr_device_uuid cl_khr_fp16 cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_icd cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_intel_command_queue_families cl_intel_subgroups cl_intel_required_subgroup_size cl_intel_subgroups_short cl_khr_spir cl_intel_accelerator cl_intel_driver_diagnostics cl_khr_priority_hints cl_khr_throttle_hints cl_khr_create_command_queue cl_intel_subgroups_char cl_intel_subgroups_long cl_khr_il_program cl_intel_mem_force_host_memory cl_khr_subgroup_extended_types cl_khr_subgroup_non_uniform_vote cl_khr_subgroup_ballot cl_khr_subgroup_non_uniform_arithmetic cl_khr_subgroup_shuffle cl_khr_subgroup_shuffle_relative cl_khr_subgroup_clustered_reduce cl_intel_device_attribute_query cl_khr_suggested_local_work_size cl_intel_split_work_group_barrier cl_intel_spirv_media_block_io cl_intel_spirv_subgroups cl_khr_spirv_no_integer_wrap_decoration cl_intel_unified_shared_memory cl_khr_mipmap_image cl_khr_mipmap_image_writes cl_intel_planar_yuv cl_intel_packed_yuv cl_khr_int64_base_atomics cl_khr_int64_extended_atomics cl_khr_image2d_from_buffer cl_khr_depth_images cl_khr_3d_image_writes cl_intel_media_block_io cl_intel_bfloat16_conversions cl_intel_va_api_media_sharing cl_intel_sharing_format_query cl_khr_pci_bus_info cl_intel_create_buffer_with_properties cl_intel_dot_accumulate cl_intel_subgroup_local_block_io cl_intel_subgroup_matrix_multiply_accumulate cl_intel_subgroup_split_matrix_multiply_accumulate 
  Platform Extensions with Version                cl_khr_byte_addressable_store                                    0x400000 (1.0.0)
                                                  cl_khr_device_uuid                                               0x400000 (1.0.0)
                                                  cl_khr_fp16                                                      0x400000 (1.0.0)
                                                  cl_khr_global_int32_base_atomics                                 0x400000 (1.0.0)
                                                  cl_khr_global_int32_extended_atomics                             0x400000 (1.0.0)
                                                  cl_khr_icd                                                       0x400000 (1.0.0)
                                                  cl_khr_local_int32_base_atomics                                  0x400000 (1.0.0)
                                                  cl_khr_local_int32_extended_atomics                              0x400000 (1.0.0)
                                                  cl_intel_command_queue_families                                  0x400000 (1.0.0)
                                                  cl_intel_subgroups                                               0x400000 (1.0.0)
                                                  cl_intel_required_subgroup_size                                  0x400000 (1.0.0)
                                                  cl_intel_subgroups_short                                         0x400000 (1.0.0)
[... cut ...]
	Location: /workspace/neo/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48

[  FAILED  ] UsmSharedMigrateGpuForFillTest/UsmSharedMigrateGpuForFillTest.Test/7
FAILED assertion ASSERT_CL_SUCCESS(clEnqueueMemFillINTEL(opencl.commandQueue, buffer, &pattern, 1, arguments.bufferSize, 0, nullptr, nullptr))
	value: -59 (CL_INVALID_OPERATION)
	Location: /workspace/neo/compute-benchmarks/source/benchmarks/memory_benchmark/implementations/ocl/usm_shared_migrate_gpu_for_fill_ocl.cpp:48
```

al42and, 2023-03-03:
I'm also observing some failures on 6.2.1 kernel and A770, logs attached.

The performance drop for non-D2D copies is even worse than reported by others. Here's excerpt from `memory_benchmark_l0.txt` for `UsmCopy(api=l0 src=* dst=* size=512MB contents=Zeros forceBlitter=0 useEvents=1 reuseCmdList=0)`:

```
                      TestCase       Mean     Median   StdDev      Min        Max   Type   Label [unit]
UsmCopy(src=Device dst=Device)    225.207    224.840    0.34%  224.298    226.493  [GPU]         [GB/s]
  UsmCopy(src=Device dst=Host)      2.845      2.848    0.28%    2.822      2.849  [GPU]         [GB/s]
  UsmCopy(src=Host dst=Device)      2.920      2.921    0.05%    2.916      2.922  [GPU]         [GB/s]
    UsmCopy(src=Host dst=Host)      2.071      2.071    0.03%    2.070      2.072  [GPU]         [GB/s]
```

- :red_circle: [api_overhead_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881461/api_overhead_benchmark_l0.txt)
- :green_circle:  [api_overhead_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10881462/api_overhead_benchmark_ocl.txt)
- :yellow_circle: [memory_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881467/memory_benchmark_l0.txt)
- :red_circle:  [memory_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10881468/memory_benchmark_ocl.txt)
- :red_circle:  [multithread_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881470/multithread_benchmark_l0.txt)
- :green_circle: [multithread_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10881471/multithread_benchmark_ocl.txt)
- :yellow_circle: [multitile_memory_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881472/multitile_memory_benchmark_l0.txt)
- :yellow_circle: [multitile_memory_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10881473/multitile_memory_benchmark_ocl.txt)
- :red_circle: [overlap_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881474/overlap_benchmark_l0.txt)
- :green_apple: [overlap_benchmark_ocl.txt](https://github.com/intel/compute-runtime/files/10881475/overlap_benchmark_ocl.txt)
- :red_circle: [p2p_benchmark_l0.txt](https://github.com/intel/compute-runtime/files/10881476/p2p_benchmark_l0.txt)

- :information_source: [lspci.txt](https://github.com/intel/compute-runtime/files/10881466/lspci.txt)

JablonskiMateusz, 2023-07-25:
With recent release it works fine on our side, @BA8F0D39  could you confirm it is working for you?

JablonskiMateusz, 2023-07-28:
@BA8F0D39  Please re-open the issue or create a new one if problem still exists

closed 2023-07-28 by JablonskiMateusz