intel/compute-runtime #619: enqueueMapBuffer hangs on A770 on linux

knzivid, 2023-02-24:
The sample attached at the end (main.cpp) is a vector addition kernel. This sample works fine on `Intel(R) UHD Graphics 630 [0x3e98]` using the same runtime, and `NVIDIA GeForce GTX 1660`. I do not have the debug symbols for this package on ArchLinux. Let me know if you'd like them and I'll build the runtime from source.

**Environment**

- Arch Linux, kernel 6.1.12-arch1-1
- Intel Corporation DG2 [Arc A770] (rev 08)

**Packages**

```
intel-compute-runtime 22.43.24595.30-1
intel-gmmlib 22.3.3-1
intel-gpu-tools 1.26-1
intel-graphics-compiler 1:1.0.12812.4-1
intel-media-driver 23.1.0-1
intel-media-sdk 23.1.0-1
intel-opencl-clang 15.0.0-1
intel-opencl-runtime 1:18.1.0.015-3
```

**Backtrace**

```
#0  0x00007ffff770545b in sched_yield () from /usr/lib/libc.so.6
#1  0x00007fffdd6adf01 in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#2  0x00007fffdd5b619f in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#3  0x00007fffdd0e388f in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#4  0x00007fffdd0e4c36 in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#5  0x00007fffdd4141aa in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#6  0x00007fffdd418527 in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#7  0x00007fffdd0e0ab5 in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#8  0x00007fffdd0e0dfd in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#9  0x00007fffdd0b36e6 in ?? () from /usr/lib/intel-opencl/libigdrcl.so
#10 0x0000555555559aee in cl::CommandQueue::enqueueMapBuffer (this=0x7fffffffe1a0, buffer=..., blocking=1, flags=2, offset=0, size=40, events=0x0, event=0x0, err=0x7fffffffe0b4) at /usr/include/CL/opencl.hpp:8132
#11 0x000055555555d415 in cl::copy<__gnu_cxx::__normal_iterator<float*, std::vector<float, std::allocator<float> > > > (queue=..., startIterator=0, endIterator=9.10844002e-44, buffer=...) at /usr/include/CL/opencl.hpp:9549
#12 0x000055555555b828 in cl::Buffer::Buffer<__gnu_cxx::__normal_iterator<float*, std::vector<float, std::allocator<float> > > > (this=0x7fffffffe1b0, queue=..., startIterator=0, endIterator=9.10844002e-44, readOnly=true, useHostPtr=false, err=0x0) at /usr/include/CL/opencl.hpp:9236
#13 0x0000555555557b75 in main () at /home/user/source/main/opencl-test.cpp:59
#14 0x00007ffff763c790 in ?? () from /usr/lib/libc.so.6
#15 0x00007ffff763c84a in __libc_start_main () from /usr/lib/libc.so.6
#16 0x00005555555574f5 in _start ()
```

<details>
<summary>main.cpp</summary>

```cpp
#define CL_HPP_TARGET_OPENCL_VERSION 120
#define CL_HPP_MINIMUM_OPENCL_VERSION 120
#define CL_HPP_ENABLE_EXCEPTIONS
#include <CL/cl2.hpp>

#include <cassert>
#include <iostream>
#include <numeric>
#include <vector>

std::string kernel()
{
    return R"ocl(
    kernel void vec_add(global const float *a, global const float *b, global float *c)
    {
        size_t i = get_global_id(0);
        c[i] = a[i] + b[i];
    }
    )ocl";
}

int main()
{
    std::vector<cl::Platform> platforms;
    cl::Platform::get(&platforms);
    cl::Device device;
    for (const auto &platform : platforms) {
        std::vector<cl::Device> devices;
        platform.getDevices(CL_DEVICE_TYPE_GPU, &devices);
        for (const auto &it : devices) {
            if (it.getInfo<CL_DEVICE_NAME>() == "Intel(R) Graphics [0x56a0]") {
                std::cout << "Using " << it.getInfo<CL_DEVICE_NAME>() << std::endl;
                device = it;
                break;
            }
        }
        if (device != cl::Device {})
            break;
    }

    cl::Context context { device };
    cl::CommandQueue queue { context };
    cl::Program program;

    try {
        program = cl::Program { context, kernel(), true };
    } catch (const cl::BuildError &error) {
        std::cerr << error.getBuildLog()[0].second << std::endl;
        throw;
    }

    constexpr auto n = 10;
    std::vector<float> vecA(n);
    std::iota(vecA.begin(), vecA.end(), 0);

    std::vector<float> vecB(n);
    std::iota(vecB.rbegin(), vecB.rend(), 0);

    cl::Buffer bufA = cl::Buffer { queue, vecA.begin(), vecA.end(), true };
    cl::Buffer bufB = cl::Buffer { queue, vecB.begin(), vecB.end(), true };

    std::vector<float> vecC(n);
    cl::Buffer bufC { context, CL_MEM_WRITE_ONLY, sizeof(float) * n };

    cl::KernelFunctor<cl::Buffer, cl::Buffer, cl::Buffer> vecAdd { program, "vec_add" };
    vecAdd(cl::EnqueueArgs(queue, cl::NDRange(n)), bufA, bufB, bufC);

    cl::copy(queue, bufC, vecC.begin(), vecC.end());

    for (auto it : vecC) {
        assert(it == n - 1);
    }
}
```

</details>

nyanmisaka, 2023-02-24:
You may need to upgrade the kernel to 6.2 and try again, which is the first version that supports Arc GPU out of the box.

JablonskiMateusz, 2023-04-04:
@knzivid  I see it moved to 6.2.9

knzivid, 2023-04-04:
Thanks for the reminder, I will upgrade and test soon

knzivid, 2023-04-24:
Sorry for the delay. I have moved to developing on windows at the moment and I am not sure when I'll get back to linux. I will close the issue until I can reproduce this issue with kernel 6.2 or higher.

closed 2023-04-24 by knzivid