intel/compute-runtime #621: Build failed with error code: -44

wangzy0327, 2023-03-02:
I use oloc compile --help to find the list of avaliable device types . But I get " Build failed with error code: -44"
how to solve the problem?

[image]

JablonskiMateusz, 2023-03-06:
Fix merged: https://github.com/intel/compute-runtime/commit/2e872c18710c9731a00b10917fdf9a4f4a6600da

wangzy0327, 2023-03-07:
@JablonskiMateusz how to build oloc from the source dep on [Github](https://github.com/intel/compute-runtime)

JablonskiMateusz, 2023-03-07:
@wangzy0327  you may go to release you want, download and install igc and gmmlib packages (both regular and devel), call cmake with `-DBUILD_WITH_L0=0` and then make `ocloc`

wangzy0327, 2023-03-07:
@JablonskiMateusz  Can you point to me which is requested package for igc and gmmlib? 
Is there a build and install ocloc guide?

```
wzy@intel12732:~$ sudo apt-cache search igc

libsigc++-2.0-0v5 - type-safe Signal Framework for C++ - runtime
libsigc++-2.0-dev - type-safe Signal Framework for C++ - development files
libsigc++-2.0-doc - type-safe Signal Framework for C++ - reference documentation
android-libart - Android Runtime
daemonize - tool to run a command as a daemon
fonts-sambhota-yigchung - sambhota yigchung free OpenType Tibetan font
gpredict - Satellite tracking program
gpsbabel - GPS file conversion plus transfer to/from GPS units
hamradio-rigcontrol - Debian Hamradio Rig Control Packages
libapp-control-perl - Perl module for apachectl style control of another executable
liberis-1.3-21 - WorldForge client entity library
liberis-1.3-dev - WorldForge client entity library - development files
liberis-doc - WorldForge client entity library - API documentation
libigc-dev - Intel graphics compiler for OpenCL -- core development files
libigc-tools - Intel graphics compiler for OpenCL -- media tools
libigc1 - Intel graphics compiler for OpenCL -- core libs
libigdfcl-dev - Intel graphics compiler for OpenCL -- OpenCL development files
libkf5config-bin - configuration settings framework for Qt
libkf5config-data - configuration settings framework for Qt
libkf5config-dev - configuration settings framework for Qt
libkf5config-dev-bin - configuration settings framework for Qt -- binary package
libkf5config-doc - configuration settings framework for Qt (documentation)
libkf5configcore5 - configuration settings framework for Qt
libkf5configgui5 - configuration settings framework for Qt
libpam-pwdfile - PAM module allowing authentication via an /etc/passwd-like file
librandom123-dev - parallel random numbers library
librandom123-doc - documentation and examples of parallel random numbers library
libsigx-2.0-2v5 - interthread communication library for C++ - runtime
libsigx-2.0-dev - interthread communication for C++ - development files
libsigx-2.0-doc - interthread communication for C++ - reference documentation
r-cran-zeligchoice - GNU R zelig choice models
ruby-sigdump - SIGQUIT of Java VM for Ruby
texlive-lang-japanese - TeX Live: Japanese
xdx - DX-cluster tcp/ip client for amateur radio
tiemu - Texas Instruments calculators emulator (without GDB)
intel-igc-cm - Intel(R) C for Metal Compiler -- CM Frontend lib
intel-igc-cm-devel - Intel(R) C for Metal Compiler -- CM Frontend development package

wzy@intel12732:~$ sudo apt-cache search gmm

libgmm++-dev - Generic C++ template library for sparse, dense and skyline matrices
libigdgmm-dev - Intel Graphics Memory Management Library -- development files
libigdgmm11 - Intel Graphics Memory Management Library -- shared library
octave-econometrics - econometrics functions for Octave
ogmtools - Tools for manipulating Ogg multimedia streams
r-cran-gmm - GNU R generalized method of moments and generalized empirical likelihood
r-cran-plm - GNU R estimators and tests for panel data econometrics

```

closed 2023-05-18 by JablonskiMateusz