intel/compute-runtime #622: Regression in OpenCL runtime 22.53 on Arc A770

al42and, 2023-03-02:
Hello!

After installing compute-runtime 22.53, I'm having my SYCL application fail on Arc A770. It works fine on the same machine on integrated UHD 770, or if I revert to 22.49.

I don't have a short repro, as the problem looks like a missing synchronization or another such issue, not a kernel miscompilation. Hopefully, the reproducer below could still be helpful:

1. Build the app in the docker container:

```dockerfile
FROM intel/oneapi:2023.0.0-devel-ubuntu22.04

RUN cd / && wget https://ftp.gromacs.org/gromacs/gromacs-2023.tar.gz && tar xzf gromacs-2023.tar.gz && rm -rf gromacs-2023.tar.gz && \
    mkdir /gromacs-2023/build && cd /gromacs-2023/build && \
    cmake .. -DCMAKE_C_COMPILER=icx -DCMAKE_CXX_COMPILER=icpx -DGMX_GPU=SYCL -DGMX_SIMD=None -DCMAKE_BUILD_TYPE=Debug && \
    cmake --build . --target tests -- -j $(nproc)

RUN apt update && apt install -y ocl-icd-libopencl1 cmake && apt clean
```

2. In the container, launched with `--devices /dev/dri`, install compute-runtime and run tests:

```console
$ uname -r 
6.2.1-060201-generic

# Install 22.53
$ wget https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12812.26/intel-igc-core_1.0.12812.26_amd64.deb && \
    wget https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12812.26/intel-igc-opencl_1.0.12812.26_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.53.25242.13/intel-level-zero-gpu_1.3.25242.13_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.53.25242.13/intel-opencl-icd_22.53.25242.13_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.53.25242.13/libigdgmm12_22.3.0_amd64.deb && \
    dpkg -i intel-igc-core_1.0.12812.26_amd64.deb intel-igc-opencl_1.0.12812.26_amd64.deb intel-level-zero-gpu_1.3.25242.13_amd64.deb intel-opencl-icd_22.53.25242.13_amd64.deb libigdgmm12_22.3.0_amd64.deb && \
    rm -rf *.deb
...

# Fails on Arc
$ export SYCL_DEVICE_FILTER=opencl:gpu:2; sycl-ls && cd /gromacs-2023/build && SYCL_CACHE_PERSISTENT=1 ctest -R MdrunIOTests
Warning: SYCL_DEVICE_FILTER environment variable is set to opencl:gpu:2.
To see the correct device id, please unset SYCL_DEVICE_FILTER.

[opencl:gpu:0] Intel(R) OpenCL HD Graphics, Intel(R) Arc(TM) A770 Graphics [0x56a0] 3.0 [22.53.25242.13]
Test project /gromacs-2023/build
    Start 62: MdrunIOTests
1/1 Test #62: MdrunIOTests .....................***Failed    5.45 sec

0% tests passed, 1 tests failed out of 1

Label Time Summary:
GTest              =   5.45 sec*proc (1 test)
IntegrationTest    =   5.45 sec*proc (1 test)
SlowGpuTest        =   5.45 sec*proc (1 test)

Total Test time (real) =   5.45 sec

The following tests FAILED:
         62 - MdrunIOTests (Failed)
Errors while running CTest
Output from these tests are in: /gromacs-2023/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.

# Passes with LevelZero
$ export SYCL_DEVICE_FILTER=level_zero:gpu:0; sycl-ls && cd /gromacs-2023/build && SYCL_CACHE_PERSISTENT=1 ctest -R MdrunIOTests
Warning: SYCL_DEVICE_FILTER environment variable is set to level_zero:gpu:0.
To see the correct device id, please unset SYCL_DEVICE_FILTER.

[ext_oneapi_level_zero:gpu:0] Intel(R) Level-Zero, Intel(R) Arc(TM) A770 Graphics [0x56a0] 1.3 [1.3.25242]
Test project /gromacs-2023/build
    Start 62: MdrunIOTests
1/1 Test #62: MdrunIOTests .....................   Passed   10.21 sec

100% tests passed, 0 tests failed out of 1

Label Time Summary:
GTest              =  10.21 sec*proc (1 test)
IntegrationTest    =  10.21 sec*proc (1 test)
SlowGpuTest        =  10.21 sec*proc (1 test)

Total Test time (real) =  10.21 sec

# Works on iGPU
$ export SYCL_DEVICE_FILTER=opencl:gpu:3; sycl-ls && cd /gromacs-2023/build && SYCL_CACHE_PERSISTENT=1 ctest -R MdrunIOTests
Warning: SYCL_DEVICE_FILTER environment variable is set to opencl:gpu:3.
To see the correct device id, please unset SYCL_DEVICE_FILTER.

[opencl:gpu:0] Intel(R) OpenCL HD Graphics, Intel(R) UHD Graphics 770 [0x4680] 3.0 [22.53.25242.13]
Test project /gromacs-2023/build
    Start 62: MdrunIOTests
1/1 Test #62: MdrunIOTests .....................   Passed    8.61 sec

100% tests passed, 0 tests failed out of 1

Label Time Summary:
GTest              =   8.61 sec*proc (1 test)
IntegrationTest    =   8.61 sec*proc (1 test)
SlowGpuTest        =   8.61 sec*proc (1 test)

Total Test time (real) =   8.61 sec

# Installing 22.49
$ wget https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12812.24/intel-igc-core_1.0.12812.24_amd64.deb && \
    wget https://github.com/intel/intel-graphics-compiler/releases/download/igc-1.0.12812.24/intel-igc-opencl_1.0.12812.24_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.49.25018.24/intel-level-zero-gpu_1.3.25018.24_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.49.25018.24/intel-opencl-icd_22.49.25018.24_amd64.deb && \
    wget https://github.com/intel/compute-runtime/releases/download/22.49.25018.24/libigdgmm12_22.3.0_amd64.deb && \
    dpkg -i intel-igc-core_1.0.12812.24_amd64.deb intel-igc-opencl_1.0.12812.24_amd64.deb intel-level-zero-gpu_1.3.25018.24_amd64.deb intel-opencl-icd_22.49.25018.24_amd64.deb libigdgmm12_22.3.0_amd64.deb && \
    rm -rf *.deb

# Arc device name is hidden now, but the test passes!
$ export SYCL_DEVICE_FILTER=opencl:gpu:2; sycl-ls && cd /gromacs-2023/build && SYCL_CACHE_PERSISTENT=1 ctest -R MdrunIOTests\Warning: SYCL_DEVICE_FILTER environment variable is set to opencl:gpu:2.
To see the correct device id, please unset SYCL_DEVICE_FILTER.

[opencl:gpu:0] Intel(R) OpenCL HD Graphics, Intel(R) Graphics [0x56a0] 3.0 [22.49.25018.24]
Test project /gromacs-2023/build
    Start 62: MdrunIOTests
1/1 Test #62: MdrunIOTests .....................   Passed    5.56 sec

100% tests passed, 0 tests failed out of 1

Label Time Summary:
GTest              =   5.56 sec*proc (1 test)
IntegrationTest    =   5.56 sec*proc (1 test)
SlowGpuTest        =   5.56 sec*proc (1 test)

Total Test time (real) =   5.56 sec
```

Note 1: `SYCL_CACHE_PERSISTENT=1` is not strictly necessary, but the test takes much longer without it, ~150 seconds vs ~5 seconds, because kernels are JITted over and over.

Note 2: After 22.53.25242.13 is installed, reverting `intel-igc-core` and `intel-igc-opencl` to 1.0.12812.24 (from [22.49.25018.24](https://github.com/intel/compute-runtime/releases/tag/22.49.25018.24)) **does not** help. Further reverting `intel-opencl-icd` to 22.49.25018.24 **helps**.

Note 3: The overall failure reproduces rather reliably for me, but the number of test cases that failed varies from run to run. To be on the safe side, it could be better to use `ctest --repeat-until-fail 20 -R MdrunIOTests`

Note 4: the behavior is also reproduced with a recent build of open-source IntelLLVM on the same machine 3250db55b1d1b2bf7f9cde9ffad2106b29b1f864

BA8F0D39, 2023-03-02:
@al42and 

I am not an Intel employee but I have similar errors due to the linux kernel drivers for Arc A770 not having complete OpenCL/SYSCL support.

If you run the OpenCL intel compute benchmarks, some of the memory, IPC, multithreading, and fabric benchmarks failed for me too.
https://github.com/intel/compute-benchmarks
Also, Level Zero API has even more failures if you run their unit tests.
https://github.com/oneapi-src/level-zero-tests

al42and, 2023-03-03:
> If you run the OpenCL intel compute benchmarks, some of the memory, IPC, multithreading, and fabric benchmarks failed for me too.
> https://github.com/intel/compute-benchmarks

Good point; I'm also observing some failures and weird behavior, even with 22.49.25018.24.

Despite that, our application works fine with compute-runtime 22.49, so in 22.53 even more things got broken  :)

PiotrRozenfeld, 2023-03-03:
Thank you for detailed description.

We have managed to reproduce the reported gromacs issue. It fails sporadically in our testing, but it was enough to rootcause the problem to optimization in GPU synchronization logic added as 4b3e4bdd88a25a3e14e1d2bf3afecebdfb985b97. The failures are no longer reproducible at the top of the tree, commit 0a22230723ee351d475a19cd65318a2d30943fd4 fixes the issue and will be cherry-picked to upcoming releases. We'll also post a hotfix release for 22.53.

Please note that there is a WA that works with existing release -  please use following environment variables NEOReadDebugKeys=1 ResolveDependenciesViaPipeControls=0.
e.g. `export SYCL_DEVICE_FILTER=opencl:gpu:2; sycl-ls && cd /gromacs-2023/build && NEOReadDebugKeys=1 ResolveDependenciesViaPipeControls=0 SYCL_CACHE_PERSISTENT=1 ctest -R MdrunIOTests`

We are still looking into the other issues reported against compute-benchmarks.

al42and, 2023-03-03:
@PiotrRozenfeld, thank you for the quick reply.

As far as GROMACS is concerned, there is no urgent need for the fix.

al42and, 2023-03-20:
Note: the problem persists with 23.05; setting `NEOReadDebugKeys=1 ResolveDependenciesViaPipeControls=0` helps.

closed 2023-05-23 by al42and