intel/compute-runtime #623: How to show the Intel Device detail message?

wangzy0327, 2023-03-03:
Do you know how to show the Intel GPU detail message using command line ? like cuda nvidia-smi

nyanmisaka, 2023-03-04:
intel_gpu_top.

wangzy0327, 2023-03-05:
@nyanmisaka 
[image]
cannot find the command line

nyanmisaka, 2023-03-05:
`sudo apt install intel-gpu-tools`

wangzy0327, 2023-03-05:
@nyanmisaka 
[image]
Failed to detect engines,but my system kernel is 5,

nyanmisaka, 2023-03-05:
https://github.com/intel/compute-runtime/blob/58d3f892a9a72a163324f7db630a9634d3f29f6b/shared/source/dll/devices/devices_base.inl#L19

It seems you are using a custom kernel build with Intel DKMS i915 driver installed to enable XehpSdv support.

I'm afraid this driver is not supported by the intel-gpu-tools version that comes with the distro.

Build from source may helps this issue. https://gitlab.freedesktop.org/drm/igt-gpu-tools

wangzy0327, 2023-03-05:
@nyanmisaka  I tried to build from source [igt-gpu-tools](https://gitlab.freedesktop.org/drm/igt-gpu-tools)
But I get the result when I run `tests/core_auth` .
How can I use the igt-gpu-tools? It seems no igt-gpu-tools install rather build. I don't know how to use it.

[image]