intel/compute-runtime #624: global memory size issue

jinz2014, 2023-03-08:
"clinfo" shows that the global memory size is about 243 MiB. Is there anything wrong with the installation ? Thanks.

OS: Ubuntu 22.04.2
Kernel version: 5.17.0-1020-oem

```
  Device Name                                     Intel(R) Graphics [0x56a0]
  Driver Version                                  22.49.25018.23
 Global memory size                              255012864 (243.2MiB)
  Max memory allocation                           127506432 (121.6MiB)
```

jinz2014, 2023-03-09:
Any suggestions on troubleshooting ? Thanks.

BA8F0D39, 2023-03-09:
@jinz2014 
Enable resizable bar on BIOS or try latest stable kernel 6.2.2

jinz2014, 2023-03-10:
Thank you for your suggestions.

closed 2023-03-10 by jinz2014