intel/compute-runtime #625: iGPU drivers does not work under WSL under Docker

dm-vodopyanov, 2023-03-08:
_**Originally posted by @ZzEeKkAa in https://github.com/intel/llvm/issues/8571:**_

I'm trying to get intel GPU available in docker that is run in on WSL backend. Unfortunately it does not show GPUs:

**To Reproduce**
1. Install Ubuntu 22.04 on WSL2.
2. Install Docker Desktop on Windows and make sure it uses WSL as a backend
3. In WSL Ubuntu terminal execute: ```sudo docker run --rm -it --device=/dev/dri https://ghcr.io/intel/llvm/sycl_ubuntu2004_nightly -- sycl-ls```
Response:
```
[opencl:cpu:0] Intel(R) OpenCL, 12th Gen Intel(R) Core(TM) i7-1270P 3.0 [2022.14.8.0.04_160000]
[opencl:acc:1] Intel(R) FPGA Emulation Platform for OpenCL(TM), Intel(R) FPGA Emulation Device 1.2 [2022.14.8.0.04_160000]
```

Expected response (got from Ubuntu Desktop 22.04 livecd with just docker installed):
```
[opencl:gpu:0] Intel(R) OpenCL HD Graphics, Intel(R) Graphics [0x46a6] 3.0 [22.43.24595.30]
[opencl:cpu:1] Intel(R) OpenCL, 12th Gen Intel(R) Core(TM) i7-1270P 3.0 [2022.14.8.0.04_160000]
[opencl:acc:2] Intel(R) FPGA Emulation Platform for OpenCL(TM), Intel(R) FPGA Emulation Device 1.2 [2022.14.8.0.04_160000]
[ext_oneapi_unified_runtime:gpu:0] Intel(R) oneAPI Unified Runtime over Level-Zero, Intel(R) Graphics [0x46a6] 1.3 [1.3.24595]
[ext_oneapi_level_zero:gpu:0] Intel(R) Level-Zero, Intel(R) Graphics [0x46a6] 1.3 [1.3.24595]
```

**Environment (please complete the following information):**

- OS: 
Edition	Windows 11 Enterprise
Version	21H2
Installed on	‎2/‎13/‎2023
OS build	22000.1574
Experience	Windows Feature Experience Pack 1000.22000.1574.0

- Target device and vendor: Integrated GPU 12th Gen Intel(R) Core(TM) i7-1270P   2.20 GHz

Output for sycl-ls from WSL:
```
[opencl:acc:0] Intel(R) FPGA Emulation Platform for OpenCL(TM), Intel(R) FPGA Emulation Device 1.2 [2022.15.12.0.01_081451]
[opencl:cpu:1] Intel(R) OpenCL, 12th Gen Intel(R) Core(TM) i7-1270P 3.0 [2022.15.12.0.01_081451]
[opencl:gpu:2] Intel(R) OpenCL HD Graphics, Intel(R) Graphics [0x46a6] 3.0 [22.53.25242.13]
[ext_oneapi_level_zero:gpu:0] Intel(R) Level-Zero, Intel(R) Graphics [0x46a6] 1.3 [1.3.25242]
```

JablonskiMateusz, 2023-03-08:
@dm-vodopyanov, for WSL scenario devices are not taken from `/dev/dri` path. Adapters are queried from dxcore library. Please ensure you pass it to docker container

AlexeySachkov, 2023-03-08:
Tagging @ZzEeKkAa as original reporter to make sure he is aware of the reply provided above

ZzEeKkAa, 2023-03-08:
Thank you @JablonskiMateusz and @AlexeySachkov. Now it makes sense, why `sycl-ls` does not see gpus. Is there any tutorial how to do so?
I've tried following, but none of them worked:
```
 sudo docker run --rm -it --device=/dev/dri --device=/dev/dxg ghcr.io/intel/llvm/sycl_ubuntu2004_nightly -- sycl-ls
```

```
sudo docker run --rm -it --device=/dev/dri --device=/dev/dxg -v /usr/lib/wsl:/urs/lib/wsl ghcr.io/intel/llvm/sycl_ubuntu2004_nightly -- sycl-ls
```

Adding `/usr/lib/wsl/lib` to PATH did not help.

closed 2023-03-08 by JablonskiMateusz

fedecompa, 2024-10-01:
I have a similar problem. I installed all the packages from Intel OneAPI link. On Windows 11 all works fine and the VideoProcessor Intel(R) UHD Graphics can be used from OpenVINO libraries. On the contrary on WSL2 with the Ubuntu 22.04 distro I see no GPU:

  clinfo
  clCreateContextFromType(NULL, CL_DEVICE_TYPE_GPU)  No devices found in platform
  
   sycl-ls
   [opencl:cpu][opencl:0] Intel(R) OpenCL, Intel(R) Core(TM) i7-10850H CPU @ 2.70GHz OpenCL 3.0 (Build 0) [2024.18.7.0.11_160000]
   
   
   https://www.intel.com/content/www/us/en/docs/oneapi/installation-guide-linux/2023-0/configure-wsl-2-for-gpu-workflows.html

ZzEeKkAa, 2024-10-01:
Looks like missing drivers/user permission. Make sure to add your user into `renderer` group and restart shell (as described in the instruction). Also install all the required packages from the doc. I have it working under WSL. And your issue is different from the one I had - so please open separate issue if you still have a problem.

fedecompa, 2024-10-01:
My user is in the render group by I still get "Number of platforms  0"

Are there particular drivers to install for the Intel(R) UHD Graphics Family? 

https://askubuntu.com/questions/1507167/no-driver-for-intel-integrated-graphics-ubuntu-22-04-4-lts

I tried this but it still doesn't work.

fedecompa, 2024-10-09:
I solved updating the Intel graphics drivers to latest version for WSL2:

https://docs.openvino.ai/archive/2023.1/openvino_docs_install_guides_configurations_for_intel_gpu.html#additional-resources