intel/compute-runtime #626: Building Git head version of "zello_sysman.cpp" fails

eero-t, 2023-03-10:
While latest release version (23.05.25593.9) works fine, building Git head version of `zello_sysman`:
`g++ -O2 -Wall -o zello_sysman  zello_sysman.cpp -lze_loader -locloc`

Fails to following errors:
```
zello_sysman.cpp: In function 'void getSysmanDeviceHandles(_ze_driver_handle_t*&, std::vector<_ze_device_handle_t*>&)':
zello_sysman.cpp:166:18: error: 'zesInit' was not declared in this scope; did you mean 'zeInit'?
  166 |     VALIDATECALL(zesInit(0));
      |                  ^~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:169:18: error: 'zesDriverGet' was not declared in this scope; did you mean 'zeDriverGet'?
  169 |     VALIDATECALL(zesDriverGet(&driverCount, nullptr));
      |                  ^~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:174:18: error: 'zesDriverGet' was not declared in this scope; did you mean 'zeDriverGet'?
  174 |     VALIDATECALL(zesDriverGet(&driverCount, &sysmanDriverHandle));
      |                  ^~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:177:18: error: 'zesDeviceGet' was not declared in this scope; did you mean 'zeDeviceGet'?
  177 |     VALIDATECALL(zesDeviceGet(sysmanDriverHandle, &deviceCount, nullptr));
      |                  ^~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:183:18: error: 'zesDeviceGet' was not declared in this scope; did you mean 'zeDeviceGet'?
  183 |     VALIDATECALL(zesDeviceGet(sysmanDriverHandle, &deviceCount, sysmanDevices.data()));
      |                  ^~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp: In function 'void testSysmanFabricPort(_ze_device_handle_t*&)':
zello_sysman.cpp:1147:9: error: 'zes_fabric_port_error_counters_t' was not declared in this scope; did you mean 'zes_fabric_port_throughput_t'?
 1147 |         zes_fabric_port_error_counters_t fabricPortErrorCounters = {};
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |         zes_fabric_port_throughput_t
zello_sysman.cpp:1197:67: error: 'fabricPortErrorCounters' was not declared in this scope
 1197 |         VALIDATECALL(zesFabricPortGetFabricErrorCounters(handle, &fabricPortErrorCounters));
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:1197:22: error: 'zesFabricPortGetFabricErrorCounters' was not declared in this scope
 1197 |         VALIDATECALL(zesFabricPortGetFabricErrorCounters(handle, &fabricPortErrorCounters));
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
zello_sysman.cpp:94:25: note: in definition of macro 'VALIDATECALL'
   94 |         ze_result_t r = myZeCall;             \
      |                         ^~~~~~~~
zello_sysman.cpp:1199:48: error: 'fabricPortErrorCounters' was not declared in this scope
 1199 |             std::cout << "Link Failures = " << fabricPortErrorCounters.linkFailureCount << std::endl;
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
```

Note: as this is very useful tool for testing Sysman implementation, I'm building it against release version of L0, not against Git head version of compute-runtime project.  Others can do the same, as this tool has been mentioned in different public Sysman related tickets (not in just this project).

JablonskiMateusz, 2023-03-13:
Hi @eero-t 
Please ensure you have level zero headers in version that is pointed in manifest of neo repo as of commit you are building on https://github.com/intel/compute-runtime/blob/master/manifests/manifest.yml#L51

eero-t, 2023-03-13:
After such defines are provided by L0, `zello_sysman` could also use suitable ifdefs to build support for newer L0 features only when system L0 version (headers) support them.

This would require occasionally removing older ifdefs [1] to keep code maintainable, so it might be too much effort compared to just updating required `L0_HEADER_VERSION` error check.

[1] E.g. when distros upgrade their L0 versions against which `zello_sysman` might get compiled against.

eero-t, 2023-03-13:
> Ok, so it requires the latest L0 v1.9.9 release, released 10 days ago. I assume it should work also with newer versions?

When checking the L0 APIs more carefully, spec version 1.5.0, added in L0 frontend v1.9, should actually suffice.

Based on that, and info from https://github.com/oneapi-src/level-zero/issues/111, adding something like this (untested):
```
#if ZE_API_VERSION_CURRENT < ZE_MAKE_VERSION( 1, 5 )
# error "zello_sysman.cpp requires L0 API >= v1.5"
#endif
```

to `zello_systman.cpp` would be enough to fix this ticket (no additional L0 API extension version checks are needed).

PS.  Intel repos apparently have still only v1.8.x frontend version, as does Debian: https://salsa.debian.org/debian/level-zero/-/blob/master/debian/changelog

Whereas latest Fedora has already v1.9.x L0 frontend in F37 and newer: https://src.fedoraproject.org/rpms/oneapi-level-zero

eero-t, 2025-11-05:
>> adding something like this (untested):
>> ```
>> #if ZE_API_VERSION_CURRENT < ZE_MAKE_VERSION( 1, 5 )
>> # error "zello_sysman.cpp requires L0 API >= v1.5"
>> #endif
>> ```
>> to `zello_systman.cpp` would be enough to fix this ticket (no additional L0 API extension version checks are needed).

> We’d like to know if this issue is still affecting you, we believe that the issue should be already resolved on the latest stack. If not, please provide an update or any additional information. Otherwise, we’ll close this issue after 30 days of inactivity. Your feedback is appreciated!

@kgibala 
```
$ grep -e ZE_API_VERSION_CURRENT -e ZE_MAKE_VERSION  ./level_zero/tools/test/black_box_tests/zello_sysman.cpp
$
```

=> Not fixed.