intel/compute-runtime #627: Too low 4GB allocation limit on Intel Arc GPUs (CL_DEVICE_MAX_MEM_ALLOC_SIZE)

ProjectPhysX, 2023-03-14:
The `CL_DEVICE_MAX_MEM_ALLOC_SIZE` on Intel Arc GPUs is currently set to 4GB (A770 16GB) and 3.86GB (A750). Trying to allocate larger buffers makes the `cl::Buffer` constructor return error `-61`. Disabling the error by setting buffer flag [`(1<<23)`](https://github.com/intel/opencl-intercept-layer/blob/main/intercept/src/intercept.cpp#L8206) during allocation turns compute results into nonsense when the buffer size is larger than 4GB.

This is likely related to 32-bit integer overflow in index calculation.

A 4GB limit on buffer allocation is not contemporary in 2023, especially on a 16GB GPU; it's not 2003 anymore where computers were limited to 32-bit. A lot of software needs to be able to allocate larger buffers in order to fully use the available VRAM capacity. [FluidX3D](https://github.com/ProjectPhysX/FluidX3D) for example needs up to 82% of VRAM in a single buffer; if the allocation limit is 25% on the A770 16GB, only 4.9GB of the 16GB can be used by the software.

The limit should be removed altogether, by setting `CL_DEVICE_MAX_MEM_ALLOC_SIZE` = `CL_DEVICE_GLOBAL_MEM_SIZE` = [100% of physical VRAM capacity](https://github.com/intel/compute-runtime/issues/586#issuecomment-1467366526), and making sure that array indices are computed with 64-bit integers.

ProjectPhysX, 2023-03-14:
@jinz2014 kernel 6.2.0-060200-generic, on Ubuntu 22.04.

jinz2014, 2023-03-14:
https://github.com/intel/compute-runtime/issues/617 
I think developers are aware of the issue. Look forward to the solution in the future.

BA8F0D39, 2023-03-15:
Not an intel employee but I found many bugs related to memory transfer

1. Linux kernel driver causes random memory corruption even if you allocate memory less than 4GB in A770
2. The blocking memory transfer functions in compute runtime unblocks prematurely and returns invalid data because the memory transfer hasn't completed yet
3. Segment fault if you transfer GPU data from one CPU thread to another CPU thread.
4. Compute runtime preventing you from allocating all of the VRAM. Vulkan and OpenGL allows you to allocate all of the VRAM for some reason.

jinz2014, 2023-03-15:
@BA8F0D39  Thank you for your summary. Are there links/reproducers available for each item in your list ?

FreddieWitherden, 2023-04-03:
> The limit should be removed altogether, by setting `CL_DEVICE_MAX_MEM_ALLOC_SIZE` = `CL_DEVICE_GLOBAL_MEM_SIZE` = [100% of physical VRAM capacity](https://github.com/intel/compute-runtime/issues/586#issuecomment-1467366526), and making sure that array indices are computed with 64-bit integers. Nvidia and AMD both allow full VRAM allocation in a single buffer for a long time already.

I am unsure how practical this is.  A lot of optimizations in the compiler are based around the buffer size being limited to 4 GiB.  From what I can gather from the ISA a lot of the memory instructions only support `base + 32-bit offset` calculations.  If larger allocations are permitted the compiler will not be able to emit these instructions and will have to use those which accept 64-bit addresses.  This requires extra registers in the kernel and emulated 64-bit arithmetic.

iamhumanipromise, 2023-04-12:
I have a laptop with 9th Gen Core i9 (9th Gen gfx also), 32GB RAM. Random Discord convos have talked about using it for Stable Diffusion. That being said, the 4GB limit seems to apply there as well. 

So is this some sort of carryover from GFX8/GFX9 cards? Change the allocation size when using dedicated GPU vs. iGPU?

MaciejPlewka, 2023-04-13:
It's possible to use allocations greater than 4GB. Please take a look at this guide https://github.com/intel/compute-runtime/tree/master/programmers-guide

closed 2023-04-14 by JablonskiMateusz

iamhumanipromise, 2023-04-14:
That's the programmer's guide. I'm specifically talking about after the python environment has been launched and I'm executing already-generated code without this flag enabled. 

Looks like I have to file an issue with that project, in this case.

ProjectPhysX, 2023-04-15:
> It's possible to use allocations greater than 4GB. Please take a look at this guide https://github.com/intel/compute-runtime/blob/master/programmers-guide/ALLOCATIONS_GREATER_THAN_4GB.md

I just tried it again with my A750 and the latest driver `22.49.25018.23` and kernel `6.2.11-060211-generic` on `Ubuntu 22.04.2 LTS`, and it is still broken. Passing the `CL_MEM_ALLOW_UNRESTRICTED_SIZE_INTEL = (1 << 23)` to `cl::Buffer` constructor / `clCreateBuffer` disables the buffer allocation error `-61`, but simulation results then become nonsense.

To reproduce:
```
git clone https://github.com/ProjectPhysX/FluidX3D.git
cd FluidX3D
chmod +x make.sh
```

Change [`src/opencl.hpp` line 213](https://github.com/ProjectPhysX/FluidX3D/blob/master/src/opencl.hpp#L213) to
```
device_buffer = cl::Buffer(device.get_cl_context(), CL_MEM_READ_WRITE|(1<<23), capacity(), nullptr, &error);
```

Set the benchmark grid resolution in `src/setup.cpp` from 256³ to 384³ by commenting [line 1137](https://github.com/ProjectPhysX/FluidX3D/blob/master/src/setup.cpp#L1137) and uncommenting [line 1138](https://github.com/ProjectPhysX/FluidX3D/blob/master/src/setup.cpp#L1138).

Compile and run with:
```
./make.sh
```
If it's broken, it will show impossibly large performance/bandwidth, like
```
| MLUPs   | Bandwidth   | Steps/s   | Current Step      | Time Remaining      |
|   28144 |   4306 GB/s |       497 |         6281  10% |                  0s |
```
If it works, reported bandwidth will be realistic, like:
```
| MLUPs   | Bandwidth   | Steps/s   | Current Step      | Time Remaining      |
|    2500 |    382 GB/s |       149 |         9994  40% |                  0s |
```

bashbaug, 2023-04-16:
Hi @ProjectPhysX, please note that you fundamentally need two changes to make >4GB allocations "work":

1. You need to relax allocation limits using `CL_MEM_ALLOW_UNRESTRICTED_SIZE_INTEL`.  It looks like you are doing this in the steps above and it seems like this is working because you are no longer seeing the `CL_INVALID_BUFFER_SIZE` allocation error.
2. You need to tell the compiler that you are using >4GB allocations by passing the `-cl-intel-greater-than-4GB-buffer-required` program build option.  This isn't mentioned in the steps above, so I'm guessing it's missing and this is what is causing the nonsense simulation results.  Note, this isn't free, which is why this option is not enabled by default.

If you want to play around with >4GB allocations without hacking around in your code, please consider trying the OpenCL Intercept layer and specifically the [RelaxAllocationLimits](https://github.com/intel/opencl-intercept-layer/blob/main/docs/controls.md#relaxallocationlimits-bool) control, which will automatically do both of these steps for you.

ProjectPhysX, 2023-04-16:
> 2. You need to tell the compiler that you are using >4GB allocations by passing the `-cl-intel-greater-than-4GB-buffer-required` program build option.  This isn't mentioned in the steps above, so [...]

Thanks, it works! I totally missed that build flag. Would be better to enable above 4GB allocations by default though.

ElliottDyson, 2024-01-12:
Any updates on enabling this by default?

I understand there seems to be some performance compromises if this is done? Just out of curiosity, are these being worked on?

If this has found to no be possible, could a method be implemented so that instead of erroring it gets the CPU to send over 4GB chunks at a time, then the remaining amount, so that greater than 4GB can be sent in one go still, whether it be for PyTorch, or any other similar applications that need to make use of greater than 4GB of allocation?

I understand that custom compiling of say, PyTorch, is something that can be done as you have suggested with custom compiler flags. However, some of us aren't developers which makes that quite a hurdle.

Thank you

dbenedb, 2024-01-21:
I'm seconding @ElliottDyson request. 4GB allocation limit makes Intel's GPUs like ARC A770 useless for any Stable Diffusion work.

ElliottDyson, 2024-01-21:
> I'm seconding @ElliottDyson request. 4GB allocation limit makes Intel's GPUs like ARC A770 useless for any Stable Diffusion work.

I'm not sure how visible this thread is since it's been closed. Perhaps we should be opening a new one that references this? Not sure what typical GitHub etiquette is about something like this though, which is why I haven't done so yet.

lorenzeszz, 2024-09-24:
With Intel UHD it is only 1.86GB on a 32GB RAM system. OpenVINO GPU Plugin is useless for me.

simonlui, 2024-10-25:
I'm going to add another usecase that I have hit the wall with that indisputably needs a larger buffer size than this 4GB limit that has been enforced. Video diffusion has now started to hit its stride with bigger models coming out like [Mochi](https://github.com/genmoai/models) but even when quantized to 8 bit, I can not generate more than 7 frames with ComfyUI and a corresponding wrapper plugin because the generated data is overflowing some memory somewhere and I either get an out of memory or unknown PI error when trying to run settings that a 4060 Ti with 16 GB of VRAM is able to handle.

koitsu, 2025-05-12:
> Hi [@ProjectPhysX](https://github.com/ProjectPhysX), please note that you fundamentally need two changes to make >4GB allocations "work":
>
> 1. You need to relax allocation limits using [...]

Why aren't these the default in the first place?  Why does Intel impose 4GB limit?  What are the advantages to having this 4GB limit enforced vs. unenforced?  Why did Intel implement this safety mechanism in the first place?  Please shed some light on that so developers can understand any potential ramifications (particularly performance but not limited to).  Thanks.

MichalMrozek, 2025-05-12:
They are defaults on XE2+ and PVC Hardware which improves handling of stateless 64 bit messages.
On older Hardware the impact of stateless messages over stateful messages is so significant that we cannot enable those by default and we require this flags to be passed.
Overhead is workload specific, with stateless messages kernels are getting more register pressure (so highly optimized kernels may start to spill fill), the address calculation is also more complex, so shorter kernels where address calculation is significant part will take more impact.
We have no plans to change the defaults on older Hardware as that would cause the performance regressions there which is not acceptable. On new Hardware (XE2+) problem is solved and big buffers are available out of box.