intel/compute-runtime #628: Clinfo returns "Number of platforms 0"

JoeRock1298, 2023-03-14:
Hi there, I'm kind of a newbie in the subject but I need to set up the opencl environment for a master degree project.
I have followed all the steps and instilled all the packages for the Intel runtime but when I run clinfo no platforms seems to appear. Not even my CPU.
Processor: 12th Gen Intel® Core™ i7-12650H 
SO: 
- Centos 7 installed in Virtual Box (Windows)
- uname -r: 3.10.0-1160.88.1.el7.x86_64
- gcc --version: gcc (GCC) 9.2.0
Steps followed to instal "libOpenCL.so":
- sudo yum install opencl-headers
- sudo yum provides '*/libOpenCL.so'
- sudo yum install ocl-icd
- sudo ln -s /usr/lib64/libOpenCL.so.1 /usr/lib/libOpenCL.so
After that, I installed the runtime through #193:
> @tomdeakin First packages for Centos 7 are available in copr. To use them run below commands:> 
> ```
> yum install yum-plugin-copr
> yum copr enable jdanecki/intel-opencl
> yum install intel-opencl
> ```
> 
> You can check them with clinfo:
> 
> ```
> yum install epel-release
> yum install clinfo
> clinfo
> ```
When running clinfo, the result is "Number of platforms 0" so I followed the "Troubleshooting OpenCL on Linux" section in https://github.com/bashbaug/OpenCLPapers/blob/e5871c927417bad69d8537a796c0309a62e23df1/OpenCLOnLinux.md.
After the strace troubleshoot I confirmed that the system was able to find the ICD loader and the ICD loader was able to enumerate and load the vendor implementation but still not having clinfo to detect the CPU.
What would be the next steps?
The clinfo log file is:
[log.txt](https://github.com/intel/compute-runtime/files/10969633/log.txt)

Many thanks in advance.

JablonskiMateusz, 2023-03-15:
@JoeRock1298 could you provide version of intel-opencl package? I'm afraid that this driver is too old to support this GPU

JablonskiMateusz, 2023-03-15:
I just realized you are using Linux in virtual box on Windows, we have never supported such configuration. If you need to use Linux on Windows you may try with WSL. Recently @ZzEeKkAa confirmed that docker inside WSL on Windows works fine. https://github.com/intel/compute-runtime/issues/625

JoeRock1298, 2023-03-15:
@JablonskiMateusz the version of intel-opencl package I have installed is:
intel-opencl.x86_64                   21.38.21026-1.el7          @copr:copr.fedorainfracloud.org:jdanecki:intel-opencl

> I just realized you are using Linux in virtual box on Windows, we have never supported such configuration. If you need to use Linux on Windows you may try with WSL. Recently @ZzEeKkAa confirmed that docker inside WSL on Windows works fine. #625
Great. thanks for the help. Anyways, even if the GPU is not supported, clinfo should recognize CPU. Should't it?
Again, many thanks.

JablonskiMateusz, 2023-03-15:
> Anyways, even if the GPU is not supported, clinfo should recognize CPU. Should't it?

`intel-opencl` package supports only GPU

JablonskiMateusz, 2023-03-17:
@JoeRock1298 can we close this issue?

closed 2023-03-28 by JablonskiMateusz