intel/compute-runtime #629: Documentation clarification

pvelesko, 2023-03-16:
Haven't touched the tool in a while and recently noticed that architecture names changed. 
Now, if you provide "gen11" as the device, instead of creating myKernel.spv file, the kernel gets compiled for multiple architectures and myKernel.ar file gets created instead. 

Could someone please clarify what to do with this file in terms of compiling a module in Level Zero?

daariahinz, 2023-04-13:
There are several options for building binaries in ocloc. One of the supported options is fat binary. This allows you to generate binaries for several targets in the form of a single file, which is the .ar file.

If gen11 is given to ocloc as a device, a fat binary will be created with binaries for platforms included in gen11.
Both .spv and .ar files are supported and on the L0 side there is validation and correct selection of the binary 
from the fat binary file.

If you want to create a single binary, you can use a specific platform name (e.g. icllp). 
You can check the complete list of supported target device types for your installed version of ocloc with:
ocloc compile --help

JablonskiMateusz, 2023-07-07:
@pvelesko does it  answer your question?

closed 2024-01-30 by pvelesko